fwupd/fwupd #5575: segmentation fault on fwupdtool get-devices

wofwofwof, 2023-02-22:
**Describe the bug**
After running fwupdtool get-devices the program crashes with a segmentation fault. The logfile is attached. 

Interesting is the message before the segfault: fu_strsplit: assertion 'sz > 0' failed

I also gets an segfault when running the service /etc/init.d/fwupd start

In the syslog I see following entries for both tries:

Feb 21 12:24:05 localhost kernel: fwupdtool[22982]: segfault at 0 ip 00007f16bd9d0fbd sp 00007ffe6be37380 error 4 in libfwupdplugin.so[7f16bd9be000+60000] likely on CPU 5 (core 2, socket 0)
Feb 21 12:24:05 localhost kernel: Code: 69 31 ff ff 48 8b 74 24 08 49 89 c4 48 85 f6 74 8b 48 8b 7c 24 10 48 ff ce b9 ff ff ff ff 48 8d 15 a5 ed 04 00 e8 13 18 ff ff <48> 8b 38 49 89 c7 48 85 ff 74 67 31 db 4c 8d 2d ff 75 05 00 66 66

I'm not aware if I need any special kernel configuration for using fwupd. I've attached my kernel-config, too.

**Steps to Reproduce**
run fwupdtool get-devices

**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
runs to the end

```shell
fwupdmgr --version
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Process org.freedesktop.fwupd received signal 11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Package installed from Gentoo sys-apps/fwupd-1.8.10

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Process org.freedesktop.fwupd received signal 11
```

</details>

**Additional questions**

- Operating system and version: Gentoo Stable
- Have you tried rebooting? yes
- Is this a regression? no, has not worked with this laptop since the start (Tuxedo Stellaris 17 Gen 3)

wofwofwof, 2023-02-22:
fwupdtool get-devices --verbose:
[fwupdtool-get-devices.txt](https://github.com/fwupd/fwupd/files/10807439/fwupdtool-get-devices.txt)

G_DEBUG=fatal-criticals gdb --args fwupdtool get-devices --verbose
[fwupdtool-get-devices-gdb.txt](https://github.com/fwupd/fwupd/files/10807440/fwupdtool-get-devices-gdb.txt)

Kernel Config
[config.txt](https://github.com/fwupd/fwupd/files/10807441/config.txt)

hughsie, 2023-02-22:
can you share your /proc/cmdline please.

wofwofwof, 2023-02-22:
cat /proc/cmdline  is empty.

hughsie, 2023-02-22:
Literally a zero length file? Can you hexdump it pls.

wofwofwof, 2023-02-22:
sorry, my fault:

`cat /proc/cmdline | hd
00000000  0a                                                |.|
00000001
`

hughsie, 2023-02-22:
Can you attach it pls?

wofwofwof, 2023-03-06:
I think its not fixed completely because I still get a segmentation fault. Could it be that there is another place where the string is not correctly checked?

# gdb --args fwupdtool get-devices --verbose 
GNU gdb (Gentoo 12.1 vanilla) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.gentoo.org/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from fwupdtool...
(No debugging symbols found in fwupdtool)
(gdb) r
Starting program: /usr/bin/fwupdtool get-devices --verbose
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7ffff59ff6c0 (LWP 8770)]
[New Thread 0x7ffff51fe6c0 (LWP 8771)]
[Detaching after fork from child process 8772]
[New Thread 0x7ffff49fd6c0 (LWP 8773)]
16:15:04.826 FuDebug              Verbose debugging enabled (on console 1)
16:15:04.827 FuMain               locked /run/lock/fwupdtool
Loading…                 [ -                                     ]16:15:04.827 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
16:15:04.827 FuCommon             reading /etc/fwupd/daemon.conf with 2560 bytes
16:15:04.827 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
16:15:04.827 FuConfig             using autodetected max archive size 4.3 GB
16:15:04.832 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
16:15:04.832 FuRemoteList         loading /usr/share/fwupd/metainfo
16:15:04.832 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
16:15:04.832 XbSilo               file: cd7a6305-bb95-95a4-be15-7d54c9c87b63, current:cd7a6305-bb95-95a4-be15-7d54c9c87b63, cached: (null)
16:15:04.832 XbSilo               loading silo with file contents
16:15:04.832 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
16:15:04.832 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
16:15:04.832 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
16:15:04.832 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
16:15:04.832 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
16:15:04.832 FuRemoteList         ordering lvfs-testing=lvfs+1
16:15:04.832 FuRemoteList         ignoring unfound remote fwupd
Loading…                 [                                       ]16:15:04.833 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
16:15:04.833 Jcat                 loaded 1 certificates
16:15:04.833 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
16:15:04.833 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:15:04.833 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
16:15:04.833 Jcat                 loaded 1 certificates
16:15:04.833 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:15:04.833 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
16:15:04.833 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
16:15:04.833 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
16:15:04.833 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]16:15:04.833 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
16:15:04.833 FuHistory            got schema version of 8
Loading…                 [*                                      ]16:15:04.851 FuPlugin             load(/usr/lib64/fwupd-1.8.11/libfu_plugin_modem_manager.so)
16:15:04.856 FuPluginUefiCapsule  failed to get fwupd-efi runtime version: /usr/libexec/fwupd/efi/fwupdx64.efi.signed cannot be found
16:15:04.856 FuContext            added udev subsystem watch of platform
16:15:04.856 FuContext            added udev subsystem watch of pci
16:15:04.856 FuContext            added udev subsystem watch of block
16:15:04.856 FuContext            added udev subsystem watch of i2c
16:15:04.856 FuContext            added udev subsystem watch of i2c-dev
16:15:04.856 FuContext            added udev subsystem watch of hidraw
16:15:04.856 FuContext            added udev subsystem watch of gpio
16:15:04.856 FuContext            added udev subsystem watch of mei
16:15:04.856 FuContext            added udev subsystem watch of iommu
[... 420 lines cut ...]
  Created:              2023-03-06
  Guid[quirks]:         48508233-40ae-5535-a84d-d0a998285564 ← NVME
  Guid[quirks]:         8fd4ca73-d0ae-52e8-8977-461435c6f4cf ← NVME\VEN_144D
  Guid[quirks]:         61e092a7-5dee-5403-ad29-62c76dd4a770 ← NVME\VEN_144D&DEV_A80A&VER_3B2QGXA7
  PhysicalId:           PCI_SLOT_NAME=0000:03:00.0
[... cut ...]
  Class:                0x10802
  Subsystem:            nvme
  DeviceFile:           /dev/nvme0
  SysfsPath:            /sys/devices/pci0000:00/0000:00:02.1/0000:03:00.0/nvme/nvme0
  PciDepth:             2

16:15:05.418 FuDevice             changing verfmt for (null): plain->triplet
16:15:05.418 FuCommon             fu_strsplit: assertion 'sz > 0' failed

Thread 1 "fwupdtool" received signal SIGSEGV, Segmentation fault.
0x00007ffff7ca584d in fu_kernel_get_cmdline () from /usr/lib64/fwupd-1.8.11/libfwupdplugin.so

hughsie, 2023-03-06:
Indeed, I've pushed another fix to the main branch -- thanks.