fwupd/fwupd #5582: Dell Precision 3540: Available uppdate not recognized and applied

paulmenzel, 2023-02-24:
This is a Dell Precision 3540 with Debian sid/unstable and *fwupd* 1.8.11-1.

Although firmware update 1.23.0 for the Dell Precision 3540 is [available on fwupd.org](https://fwupd.org/lvfs/devices/com.dell.uefi388ff1f2.firmware), released 2023-02-18 02:35:27, it is not picked up.

```
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.11
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.1.0-5-amd64
runtime   org.freedesktop.fwupd         1.8.11

$ fwupdmgr get-updates
Devices with no available firmware updates: 
 • Thunderbolt host controller
Devices with the latest available firmware version:
 • KXG60ZNV1T02 NVMe KIOXIA 1024GB
 • System Firmware
 • UEFI dbx
No updates available
```

PS:

```
$ fwupdmgr get-devices --show-all-devices
Dell Inc. Precision 3540
│
├─Unknown Device:
│     Device ID:          59a11bc1795b6b83caa7eface68d0ad302cf5f86
│     GUID:               0422c218-5b8a-504d-8fba-43f8a29f4312 ← GPIO\ID_INT34BB:00
│   
├─Core™ i7-8665U CPU @ 1.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         d5092916-1868-5182-8c70-12f83dab6372 ← CPUID\PRO_0&FAM_06&MOD_8E&STP_C
│     Device Flags:       • Internal device
│   
├─KXG60ZNV1T02 NVMe KIOXIA 1024GB:
│     Device ID:          e11623b2caa18fee292058a5c09ca4e6152f7ecf
│     Summary:            NVM Express solid state drive
│     Current version:    10604107
│     Vendor:             Toshiba Corporation (NVME:0x1179)
│     Serial Number:      30JA82ILK84L
│     GUIDs:              8a412b95-5c07-5ce4-b63a-f9cbf254dd5d ← STORAGE-DELL-107443
│                         344c2804-4948-e811-842f-0ed5f89f718b
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─Lexa XT [Radeon PRO WX 3100]:
│     Device ID:          3b22da9de44c1641a94f92a720b14575aa71dd80
│     Current version:    BR46800.001
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002, PCI:0x8086)
│     GUIDs:              53f9e9fe-ae48-5ecd-ba8e-fa35b7aae070 ← PCI\VEN_1002&DEV_6985
│                         d6aa943b-ff91-5448-9a02-c413e12d8b45 ← PCI\VEN_1002&DEV_6985&REV_00
│                         9f45975f-1b68-5140-bd3b-b67886a5df46 ← PCI\VEN_1002&DEV_6985&SUBSYS_102808BA
│                         b353dc5b-d50a-5cdf-ba45-e8e710d1de35 ← PCI\VEN_1002&DEV_6985&SUBSYS_102808BA&REV_00
[... 640 lines cut ...]
12:03:00.613 FuMain               FuOptionromDevice:
  DeviceId:             5792b48846ce271fab11c4a545f7a3df0d36e00a
  Name:                 WhiskeyLake-U GT2 [UHD Graphics 620]
  Guid:                 52aa8ac7-c837-561f-af32-b7979cd2f6ed ← PCI\VEN_8086&DEV_3EA0
  Guid:                 6986c21a-0f3f-5538-bcef-fe4e8cda7ab7 ← PCI\VEN_8086&DEV_3EA0&REV_02
  Guid:                 ca341255-08af-58e6-897e-69a2b3ea79d1 ← PCI\VEN_8086&DEV_3EA0&SUBSYS_102808BA
  Guid:                 8981f92b-6a62-58ab-88d0-cf499dc94b3a ← PCI\VEN_8086&DEV_3EA0&SUBSYS_102808BA&REV_02
  Guid:                 1bb467a3-4af9-53a9-ab48-f588ee6e8066 ← PCI\VEN_8086&DEV_3EA0&REV_00
  Guid:                 ab7957d5-e7ca-5326-842a-90c8800b1720 ← PCI\VEN_8086&DEV_3EA0&SUBSYS_102808BA&REV_00
  Plugin:               optionrom
  Flags:                internal|registered|can-verify|can-verify-image
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Version:              02
  VersionFormat:        plain
  Created:              2023-02-24

12:03:00.615 FuPlugin             finalize(modem_manager)
12:03:00.615 FuPlugin             finalize(flashrom)
12:03:00.615 FuPluginTest         destroy

12:03:00.615 FuProgress           raw timing data was { 0.000, 0.000, 0.003, 0.000 } -- steps were set as [ 1 1 3 95 ] but should have been [ 7 2 87 3 ] at ../libfwupdplugin/fu-context.c:810
12:03:00.616 FuProgress           raw timing data was { 0.022, 0.002, 0.006, 0.000, 0.000, 0.002 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 68 5 19 1 0 7 ] at ../plugins/uefi-capsule/fu-uefi-capsule-plugin.c:906
```
</details>

hughsie, 2023-02-24:
@paulmenzel I'm seeing lots of:

    Failed to create file “/var/cache/fwupd/devices.json.GZZL01”: Permission denied

can you add `sudo fwupdtool get-devices --verbose` and then grab the output again please. I can see the update in the metadata just fine so something odd is going on.

superm1, 2023-02-24:
Also weird:
> 12:03:00.536 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:1d.4/0000:3c:00.0/nvme/nvme0: failed to open /dev/nvme0: Permission denied

Is the daemon not running as root?

superm1, 2023-02-24:
> 12:02:57.397 FuPluginUefiCapsule  not using CapsuleOnDisk support: Capsule-on-Disk is not supported
> 12:02:57.403 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
> 12:02:57.405 FuPluginUefiCapsule  UX Capsule support : Enabled

You must not be running as root I conclude.

> Although firmware update 1.23.0 for the Dell Precision 3540 is [available on fwupd.org](https://fwupd.org/lvfs/devices/com.dell.uefi388ff1f2.firmware), released 2023-02-18 02:35:27, it is not picked up.

I don't see any special requirements on this firmware for a certain fwupd version.

paulmenzel, 2023-02-27:
> can you add `sudo fwupdtool get-devices --verbose` and then grab the output again please. I can see the update in the metadata just fine so something odd is going on.

```
$ sudo fwupdtool get-devices --verbose
09:35:11.055 FuDebug              Verbose debugging enabled (on console 1)
09:35:11.055 FuMain               locked /run/lock/fwupdtool
09:35:11.056 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
09:35:11.059 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [ -                                     ]09:35:11.059 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
09:35:11.059 FuCommon             reading /etc/fwupd/daemon.conf with 2563 bytes
09:35:11.059 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
09:35:11.059 FuConfig             using autodetected max archive size 4.3 GB
09:35:11.060 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
09:35:11.060 FuRemoteList         loading /usr/share/fwupd/metainfo
09:35:11.060 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
09:35:11.060 XbSilo               file: e01a7282-38ec-1e09-6f23-006384f115b1, current:e01a7282-38ec-1e09-6f23-006384f115b1, cached: (null)
09:35:11.060 XbSilo               loading silo with file contents
09:35:11.060 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs.conf.dpkg-old
09:35:11.060 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
09:35:11.060 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
09:35:11.060 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
09:35:11.060 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs-testing.conf.dpkg-old
09:35:11.060 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
09:35:11.060 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
09:35:11.060 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
09:35:11.060 FuRemoteList         ordering lvfs-testing=lvfs+1
09:35:11.060 FuRemoteList         ignoring unfound remote fwupd
Loading…                 [                                       ]09:35:11.060 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
09:35:11.060 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
09:35:11.060 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
09:35:11.060 Jcat                 loaded 1 certificates
09:35:11.061 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
09:35:11.061 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
09:35:11.061 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
09:35:11.061 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
09:35:11.061 Jcat                 loaded 1 certificates
09:35:11.061 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
09:35:11.061 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
09:35:11.061 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
09:35:11.061 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
09:35:11.061 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]09:35:11.061 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
09:35:11.061 FuHistory            got schema version of 8
Loading…                 [*                                      ]09:35:11.065 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-1.8.12/libfu_plugin_modem_manager.so)
09:35:11.068 FuContext            added udev subsystem watch of platform
09:35:11.068 FuContext            added udev subsystem watch of pci
09:35:11.068 FuContext            added udev subsystem watch of block
09:35:11.068 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
09:35:11.068 FuPluginDell         Using libsmbios 2.4
09:35:11.068 FuContext            added udev subsystem watch of i2c
09:35:11.068 FuContext            added udev subsystem watch of i2c-dev
09:35:11.068 FuContext            added udev subsystem watch of hidraw
09:35:11.068 FuContext            added udev subsystem watch of gpio
09:35:11.068 FuContext            added udev subsystem watch of mei
09:35:11.068 FuContext            added udev subsystem watch of iommu
09:35:11.068 FuContext            added udev subsystem watch of video4linux
09:35:11.068 FuContext            added udev subsystem watch of msr
09:35:11.068 FuContext            added udev subsystem watch of mtd
09:35:11.068 FuContext            added udev subsystem watch of nvme
09:35:11.069 FuContext            added udev subsystem watch of drm
09:35:11.069 FuContext            added udev subsystem watch of drm_dp_aux_dev
09:35:11.069 FuContext            added udev subsystem watch of serio
09:35:11.069 FuPluginTest         init
[... cut ...]
        This stable release fixes the following issues:
        
        • Firmware updates to address security vulnerabilities.
        Issues:           CVE-2021-33159
                          CVE-2022-21181
                          CVE-2022-21166
                          CVE-2022-21151
                          CVE-2022-21127
                          CVE-2022-21125
                          CVE-2022-21123
                          CVE-2022-0005
                          CVE-2022-0004
      
```
```

hughsie, 2023-02-27:
and `sudo fwupdtool get-updates --verbose 388ff1f2-f836-48a5-9994-d4992e63faff` please?

hughsie, 2023-02-27:
So... you're on the latest, no? `current version is 1.23.0: 1.23.0=same` -- was this just a transient issue with the metadata perhaps?

paulmenzel, 2023-02-27:
The problem persisted this morning, so I wouldn’t say it’s a transient issue from the time perspective.

With your suggestion I ran `sudo fwupdtool get-devices --verbose`, that means with administrative privileges, and then the update was fetched(?) and I was able to apply it (from 1.15.0 to 1.23.0).

hughsie, 2023-02-27:
I can't explain any of that! :)

superm1, 2023-02-27:
> The problem persisted this morning, so I wouldn’t say it’s a transient issue from the time perspective.

As you already upgraded the firmware successfully I'll close this.

I think if this pops back up we should look at the daemon log as launched by systemd in verbose.  By running fwupdtool outside the systemd unit we could mask some other issue.

To do this modify the systemd unit for the daemon `fwupd.service` and add `--verbose` to command.  Make sure you reload the units, and then fetch the journal output for `fwupd.service` after reproducing.

closed 2023-02-27 by superm1