fwupd/fwupd #5593: Include generated man pages in release tarball

mkszuba, 2023-02-28:
**Is your feature request related to a problem? Please describe.**

Since 1.8.12 fwupd generates man pages from Markdown files, using pandoc. Unfortunately pandoc has a rather massive dependency tree, and many of these dependencies are rather fragile (i.e. can quite often fail to install). This may not be that much of a problem for systems based on binary packages but for source-based systems such as BSD ports or Gentoo Linux, requiring pandoc in order to have man pages puts a lot of strain on end users.

**Describe the solution you'd like**

fwupd already publishes official source tarballs of its releases so it would be great for these tarballs to include pre-generated man pages, which the users could then install directly.

**Describe alternatives you've considered**

It might be possible for Gentoo developers to generate man pages ourselves and publish them as a separate, Gentoo-specific tarball prior to making a new fwupd release available to users. However, that would be both less automated than having man-page generation integrated in the upstream release schedule and not directly of much help to everyone else building fwupd from sources.

**Additional context**

I am a maintainer of fwupd packages in Gentoo Linux.

hughsie, 2023-02-28:
Is there something lighter that we should use for going from Markdown to man? Maybe we could do something in Python locally?

superm1, 2023-02-28:
The problem with "generated" content going into the tarballs is that tarballs are a true snapshot of git.  So just like your action at release time this would mean a new action during release time and committing that during release time.

I recall @hughsie did something to lessen the dependency tree in arch.  Maybe something similar can be done for Gentoo?

I have opinions about the value of man pages by default that don't reflect the rest of the world.
But if you think they are valuable you could also just carry a local generated copy in your VCS.  They don't change that frequencly.

This could also be a good argument to have a special use flag that is by default disabled instead of something general like `man`