fwupd/fwupd #5596: KERNEL BUG: type attribute not exported

xxmlud, 2023-02-28:
**Describe the bug**
My computer had a CPU consumption problem because of the /usr/bin/fwupdmgr process.
I had to stop the service as follows

systemctl status fwupd.service

The following message appears in the log:

````shell
fwupd[9013]: 09:29:42.818 FuBiosSettings KERNEL BUG: 'type' attribute not exported: (failed to load type: Failed to open file <</sys/class/firmware-attributes/thinklmi/attributes/TotalMemoryEncryption/type>>>: File or directory does not exist)
fwupd[9013]: 09:29:43.066 FuEngine failed to add device usb:03:00:03: failed to get version: error status: 0x315
fwupd[9013]: 09:29:46.207 FuEngine failed to add device /sys/devices/pci0000:00/0000:00:1f.5: failed to mmap SPIBAR: Operation not allowed
````

**Steps to Reproduce**
This problem has not recurred.

**Expected behavior**
This Kernel BUG message should not appear.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
~ X /usr/bin/fwupdmgr --version
compile org.freedesktop.fwupd 1.8.10
compile com.hughsie.libjcat 0.1.12
runtime org.freedesktop.fwupd-efi 1.4
compile org.freedesktop.gusb 0.4.3
runtime com.hughsie.libjcat 0.1.12
runtime com.dell.libsmbios 2.4
runtime org.freedesktop.gusb 0.4.5
runtime org.freedesktop.fwupd 1.8.10
runtime org.kernel 6.1.12-arch1-1
```
The package already came installed with the operating system.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
LENOVO 20W1S58200
│
├─Dispositivo Desconocido:
│     ID de dispositivo:  63223cc2d1b0b8d4c45ef8cedcb1f45315d3ff20
│     GUID:               ca5c154c-a46a-59d8-bf77-0ca40522cc85 ← GPIO\ID_INT34C5:00
│   
├─0000:00:1f.5:
│     ID de dispositivo:  b04e387fb80d2b91f37a4d0c7b21461c451775e1
│     Totales:            Memory Technology Device
│     Proveedor:          DMI:LENOVO
│     GUIDs:              5f93d7e7-e282-59b9-b663-0146e382f8f6 ← MTD\NAME_0000:00:1f.5
│                         1d5d8ab8-c56e-59b0-9567-d4d306907af8 ← MTD\VENDOR_LENOVO&NAME_0000:00:1f.5
│                         1a6ba5df-2718-556f-ad63-d2b3f0f3c586 ← MTD\VENDOR_LENOVO&PRODUCT_20W1S58200&NAME_0000:00:1f.5
│     Indicadores del Dispositivo:• Dispositivo interno
│                         • Actualizable
│                         • Necesita rearrancar tras la instalación
│                         • Hay disponible verificación hash criptográfica
│   
├─11th Gen Intel Core™ i5-1145G7 @ 2.60GHz:
│     ID de dispositivo:  4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Versión actual:     0x000000a6
│     Proveedor:          Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│                         1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
[... 317 lines cut ...]
Qt Version: 5.15.8
Kernel Version: 6.1.12-arch1-1 (64-bit)
Graphics Platform: X11
Processors: 8 × 11th Gen Intel® Core™ i5-1145G7 @ 2.60GHz
Memory: 15.3 GiB of RAM
Graphics Processor: Mesa Intel® Xe Graphics
Manufacturer: LENOVO
Product Name: 20W1S58200
System Version: ThinkPad T14 Gen 2i
```

- Have you tried rebooting?
Yes, I have restarted the computer and I still get the message

For the moment I have not detected any serious effects.

```
~ ❯ ls -lh /sys/class/firmware-attributes/thinklmi/attributes/TotalMemoryEncryption/
.rw------- root root 4.0 KB Tue Feb 28 18:09:39 2023  current_value
.r--r--r-- root root 4.0 KB Tue Feb 28 18:09:39 2023  display_name
.r--r--r-- root root 4.0 KB Tue Feb 28 18:09:39 2023  possible_values
~ ❯ ls -lh /sys/class/firmware-attributes/thinklmi/attributes/TotalMemoryEncryption/Type
lsd: /sys/class/firmware-attributes/thinklmi/attributes/TotalMemoryEncryption/Type: No such file or directory (os error 2).
```
The file type does not exist.

hughsie, 2023-02-28:
@xxmlud how do you mean `had a CPU consumption problem` -- does the fwupdmgr or fwupd process spin at 100% CPU? The warning about the firmware type is valid.

superm1, 2023-02-28:
This is a link to the kernel bug: https://bugzilla.kernel.org/show_bug.cgi?id=216460

(@mrhpearson FYI)

xxmlud, 2023-02-28:
@hughsie , indeed, I had a problem where the "/usr/bin/fwupdmgr" process was at 100%, but it only happened to me once.

@superm1 , thanks for the contribution. is it necessary to close this issue?

hughsie, 2023-03-01:
I think so, closing as "it's a warning we want someone in Lenovo to fix" :)

closed 2023-03-01 by hughsie