fwupd/fwupd #5603: Cannot run update: failed to write data to efivarfs: Error writing to file descriptor: No space left on device

arenevier, 2023-03-04:
**Describe the bug**
Cannot run update: failed to write data to efivarfs: Error writing to file descriptor: No space left on device

**Steps to Reproduce**
$ fwupdmgr refresh
$ fwupdmgr update

**Expected behavior**
Should succeed. Instead, fails with the following:

```
/usr/bin/fwupdmgr update -y     
Devices with no available firmware updates: 
 • ASUE120D:00 04F3:31FB
 • 0000:00:1f.5
 • 6950
 • KINGSTON SNV2S2000G
 • System Firmware
 • UEFI Device Firmware
 • USB2.0 HD UVC WebCam
Writing…                 [                                       ]
failed to write data to efivarfs: Error writing to file descriptor: No space left on device
```

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.2.2-arch1-1
runtime   org.freedesktop.fwupd         1.8.12

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

pacman

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
ASUSTeK COMPUTER INC. ROG Zephyrus M16 GU603ZW_GU603ZW
│       
├─ASUE120D:00 04F3:31FB:  
│     Device ID:          983c01340ca2691209e663e8468b4bddf757670b
│     Summary:            Touchpad
│     Current version:    0x0002
│     Bootloader Version: 0x0001
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              d320ceaa-5ac9-5552-b535-ffe415c8786c ← HIDRAW\VEN_04F3&DEV_31FB
│                         e491b365-d6b0-5096-8049-88045ff0a7d9 ← HIDRAW\VEN_04F3&DEV_31FB&REV_00
│                         b1fd3244-1e51-57ce-91f0-c3a887a9ab4b ← HIDRAW\VEN_04F3&DEV_31FB&MOD_0022
│                         15e96dd6-4f5e-5d56-a991-2e63e82d1bd2 ← ELANTP\ICTYPE_11
│                         b6e57701-6954-5bcf-b517-dec27ba1d589 ← ELANTP\ICTYPE_11&MOD_0022
│                         f111e30f-4a4b-50be-8706-deff7e81f51c ← ELANTP\ICTYPE_11&MOD_0022&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable 
│                         
├─Unknown Device:         
[... 400 lines cut ...]
-rw-r--r-- 1 root root   881 Mar  4 11:56 PKDefault-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    10 Mar  4 11:56 PlatformLang-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    15 Mar  4 11:56 RemapportPathMapping-e5e0e2b2-5f15-4a5d-b208-42535541c680
-rw-r--r-- 1 root root    42 Mar  4 11:56 RstVmdV-193dfefa-a445-4302-99d8-ef3aad1a04c6
-rw-r--r-- 1 root root     5 Mar  4 11:56 SecureBoot-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    56 Mar  4 11:56 SetupCpuFeatures-ec87d643-eba4-4bb5-a1e5-3f3e36b20da9
-rw-r--r-- 1 root root     5 Mar  4 11:56 SetupMode-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root     5 Mar  4 11:56 SIDSUPPORT-7d3dceee-cbce-4ea7-8709-6e552f1edbde
-rw-r--r-- 1 root root   148 Mar  4 11:56 SignatureSupport-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    12 Mar  4 11:56 SmbiosEntryPointTable-4b3082a3-80c6-4d7e-9cd0-583917265df1
-rw-r--r-- 1 root root    12 Mar  4 11:56 SmbiosEntryPointTableF000-4b3082a3-80c6-4d7e-9cd0-583917265df1
-rw-r--r-- 1 root root    12 Mar  4 11:56 SmbiosScratchBuffer-4b3082a3-80c6-4d7e-9cd0-583917265df1
-rw-r--r-- 1 root root    12 Mar  4 11:56 SmbiosV3EntryPointTable-4b3082a3-80c6-4d7e-9cd0-583917265df1
-rw-r--r-- 1 root root    10 Mar  4 11:56 TbtSetupVolatileData-ec87d643-eba4-4bb5-a1e5-3f3e36b20da9
-rw-r--r-- 1 root root     6 Mar  4 11:56 Timeout-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    10 Mar  4 11:56 TPMPERBIOSFLAGS-7d3dceee-cbce-4ea7-8709-6e552f1edbde
-rw-r--r-- 1 root root    12 Mar  4 11:56 TpmServFlags-7d3dceee-cbce-4ea7-8709-6e552f1edbde
-rw-r--r-- 1 root root    15 Mar  4 11:56 TpvSetup-1c3483d5-1e7e-4450-9806-dede002c974b
-rw-r--r-- 1 root root    31 Mar  4 11:56 UIT_DATA-fe47349a-7f0d-4641-822b-34baa28ecdd0
-rw-r--r-- 1 root root    36 Mar  4 11:56 UIT_HEADER-fe47349a-7f0d-4641-822b-34baa28ecdd0
-rw-r--r-- 1 root root    36 Mar  4 11:56 UnlockIDCopy-eaec226f-c9a3-477a-a826-ddc716cdc0e3
-rw-r--r-- 1 root root     5 Mar  4 11:56 VendorKeys-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    68 Mar  4 11:56 WriteOnceStatus-4b3082a3-80c6-4d7e-9cd0-583917265df1

```

hughsie, 2023-03-05:
I think you need to ask ASUS for help, there's nothing we can really do. It looks like the EFI NVRAM really might be out of space. You might be able to reset it back to factory defaults in the system BIOS, but I'm doubtful if that will help. Sorry that's not more helpful.

closed 2023-03-05 by hughsie

arenevier, 2023-03-06:
Thank you.

If I look at the files in efivars, the biggest ones are db/dbx, which are the ones fwupdmgr is trying to update

```
% ls -Sl /sys/firmware/efi/efivars/ | head
total 0
-rw-r--r-- 1 root root 10160 Mar  6 07:33 dbx-d719b2cb-3d3a-4596-a3bc-dad00e67656f
-rw-r--r-- 1 root root 10160 Mar  6 07:33 dbxDefault-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root  6058 Mar  6 07:33 db-d719b2cb-3d3a-4596-a3bc-dad00e67656f
-rw-r--r-- 1 root root  6058 Mar  6 07:33 dbDefault-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root  3568 Mar  6 07:33 KEK-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root  3568 Mar  6 07:33 KEKDefault-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root  2052 Mar  6 07:33 _DMI-91b89306-5bac-4ae0-aab3-207ec12e989b
-rw-r--r-- 1 root root   881 Mar  6 07:33 PK-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   881 Mar  6 07:33 PKDefault-8be4df61-93ca-11d2-aa0d-00e098032b8c

```

goetzc, 2023-07-17:
Hi @arenevier, take a look here https://github.com/fwupd/firmware-lenovo/issues/53#issuecomment-1589611653. Those files taking up space might be dump files that are not really needed if the storage is full.

alexfrostdev, 2023-11-11:
> You might be able to reset it back to factory defaults in the system BIOS, but I'm doubtful if that will help.

I had same issue on Lenovo laptop. I found option `Restore factory keys`