fwupd/fwupd #5612: ThinkPad P14s Gen 2 AMD: failed to run update on reboot

jonsger, 2023-03-09:
**Describe the bug**
Doing a firmware upgrade on the Thinkpad P14s Gen 2 AMD doesn't work, as the update isn't done during reboot.
The system just boots as normal to the KDE desktop and the upgrade wasn't applied.

**Steps to Reproduce**
1) Run `fwupdmgr upgrade`
2) Answer "Y" to reboot question

**Expected behavior**
The System Firmware gets upgraded from `0.1.13`  to `0.1.21`.

**fwupd version information**
```
fwupdmgr --version
client version: 1.7.3
compile-time dependency versions
        gusb:   0.3.9

daemon version: 1.7.3
```
Installed via zypper on openSUSE Leap 15.4

<details>

<summary>**fwupd device information**</summary>

```shell
fwupdmgr get-devices --show-all-devices
21A0004XGE
│
├─Integrated Camera:
│     Device ID:          301046452a49d84af6356d23e43a684b8f10660f
│     Current version:    60.4
│     Vendor:             Azurewave (USB:0x13D3)
│     Serial Number:      0000
│     GUIDs:              1b2c4ed0-8e31-5d65-884b-a599e07e2260 ← USB\VID_13D3&PID_5406&REV_6004
│                         db842a92-1141-541b-965a-2733a5726e17 ← USB\VID_13D3&PID_5406
│     Device Flags:       • Updatable
│   
├─Ryzen 7 PRO 5850U with Radeon Graphics:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│                         79759cdc-94db-5098-be7b-eb02521fbbec ← CPUID\PRO_0&FAM_19&MOD_50
│                         20b595b0-5892-5870-8e4c-688133ad6e34 ← CPUID\PRO_0&FAM_19&MOD_50&STP_0
│     Device Flags:       • Internal device
│   
├─System Firmware:
│ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.13
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              8fb7e1f9-447b-41fb-8832-ae5ef955665f
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │                       2f587fcf-b814-56ef-8496-db1a4a256e5e ← UEFI\RES_{8FB7E1F9-447B-41FB-8832-AE5EF955665F}
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  267
│       Minimum Version:  267
│       Vendor:           UEFI:Linux Foundation
[... 484 lines cut ...]
├── initrd -> initrd-5.14.21-150400.24.46-default
├── initrd-5.14.21-150400.24.41-default
├── initrd-5.14.21-150400.24.46-default
├── symvers-5.14.21-150400.24.41-default.gz
├── symvers-5.14.21-150400.24.46-default.gz
├── sysctl.conf-5.14.21-150400.24.41-default
├── sysctl.conf-5.14.21-150400.24.46-default
├── vmlinux-5.14.21-150400.24.41-default.gz
├── vmlinux-5.14.21-150400.24.46-default.gz
├── vmlinuz -> vmlinuz-5.14.21-150400.24.46-default
├── vmlinuz-5.14.21-150400.24.41-default
└── vmlinuz-5.14.21-150400.24.46-default

11 directories, 360 files
```
</details>

**Additional questions**

- Operating system and version: openSUSE Leap 15.4
- Have you tried rebooting? Yes, multiple times. No changes.
- Is this a regression? Don't know, its the first time I'm using fwupd on this device.
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Need to check first

hughsie, 2023-03-09:
What fwupd-efi version please?

hughsie, 2023-03-09:
> Did I silently ordered somewhere the Super Premium Support

Invoices get sent out on Fridays :)

> fwupd-efi-1.2-150400.1.14.x86_64

That's kinda old -- can you try with 1.4 please?

hughsie, 2023-03-09:
> I tried installing a newer kernel on Leap

It's not the kernel that needs updating, it's the package called fwupd-efi, i.e. this thing: https://github.com/fwupd/fwupd-efi

You might be okay pulling the binary from the fedora fwupd-efi package and manually overwriting /usr/libexec/fwupd/efi/fwupdx64.efi and then turning Secure Boot off (as it'll be signed by the RH key, not the Suse key).

jonsger, 2023-03-09:
Okay, I disabled Secure Boot and did:

```shell
git clone https://github.com/fwupd/fwupd-efi
cd fwupd-efi/
git checkout 1.4
meson build
ninja -C build
ninja -C build install
cp /usr/local/libexec/fwupd/efi/fwupdx64.efi /usr/lib/fwupd/efi/
```

This alone didn't helped. As it now showed `fwupd-efi version 1.4` at boot, but with a QueryCapsuleCapabilities error.
After following [this hint](https://github.com/fwupd/fwupd/issues/620#issuecomment-588388351) I was able to update the System Firmware:
```shell
parted /dev/nvme0n1 set 1 esp on
```

So closing now.

Would be nice if one could get a hint, especially about this ESP thing being not set :)

superm1, 2023-03-09:
Can you please do us a favor and file a bug with your distro to get this fixed?  We can point any other future users with this distro to it too then.

hughsie, 2023-03-09:
@superm1 so is it expected that QueryCapsuleCapabilities fails if the ESP can't be found? It seems like we should be a bit less permissive about accepting FAT32 and other ESP-ish partitions like BDP. @mrhpearson can you share any details about what partition types the capsule loader can accept?

superm1, 2023-03-09:
> @superm1 so is it expected that QueryCapsuleCapabilities fails if the ESP can't be found?

That's up to the IBV how strict they want to be.  I've seen plenty accept payloads without those flags..

superm1, 2023-03-09:
How about this - what if we detect if that flag isn't set on _anything_ then **we** set it on what we're putting the capsule onto as a precaution?

hughsie, 2023-03-09:
I'm hesitant to just change the disk -- perhaps a CLI `WARNING` would be apt?

superm1, 2023-03-09:
Even if we used a high level tool to accomplish it? It should just be a bit.

How about we start out with warning and data collection for lvfs reports.

hughsie, 2023-03-10:
> data collection for lvfs reports

We can do better than that -- adding the data to the LVFS report we could add a known issue and direct the user to a wiki page explaining how to fix it. I don't think udisks gives you the interface to change a partition type so it's not something we can really do ourselves.

mrhpearson, 2023-03-10:
>  @mrhpearson can you share any details about what partition types the capsule loader can accept?

Interesting discussion. Don't know the answer but will ask the FW team for guidance.

mrhpearson, 2023-03-16:
Confirmed with FW team: disk partition should be “GPT”, then flag is “esp”.