fwupd/fwupd #5618: failed to update to 0: authentication signing error

personaingrata, 2023-03-13:
**Describe the bug**
today i updated "Lenovo Embedded Controller" via gnome-firmware on my Thinkpad P14s Gen 3 to 0.1.15. i signed fwupdx64.efi with my secure boot keys and the update went through. i thought business like usual but after rebooting, gnome-firmware reports "failed to update to 0: authentication signing error" and "Current Version 0"

**Expected behavior**
Embedded Controller being updated to  0.1.15

**fwupd version information**
fwupdmgr --version                                                      [2]
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.1.15-hardened1-1-hardened
runtime   org.freedesktop.fwupd         1.8.12

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.
fwupdmgr get-devices --show-all-devices                                    
LENOVO 21AKCTO1WW
│
├─Unbekanntes Gerät:
│     Gerätekennung:      6b5a06645e27c4e94b6e67d5185c46d11ec9ad67
│     GUID:               c96751f4-16e8-59ea-975d-69d3dac88d7d ← GPIO\ID_INTC1055:00
│   
├─12th Gen Intel Core™ i7-1280P:
│ │   Gerätekennung:      4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Aktuelle Version:   0x00000429
│ │   Anbieter:           Intel
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│ │                       ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│ │                       3ebbde86-d03e-549a-a8fd-02ebf9aa537a ← CPUID\PRO_0&FAM_06&MOD_9A&STP_3
│ │   Device Flags:       • Internes Gerät
│ │ 
│ ├─USB4 host controller:
│ │     Gerätekennung:    4f1cc6afcb7c5dc199ceb8db4d6be3bee2fc32fb
│ │     Zusammenfassung:  Unmatched performance for high-speed I/O
│ │     Anbieter:         INTEL (THUNDERBOLT:0x8087)
│ │     GUIDs:            5041b15d-6e1c-5681-86b3-bb1d56993357 ← THUNDERBOLT\VEN_8087&DEV_463E
│ │                       d651f364-21d8-5cc3-9790-dd019a8958ef ← THUNDERBOLT\VEN_8087&DEV_463E&REV_00
│ │                       e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│ │     Device Flags:     • Internes Gerät
│ │                       • System benötigt externe Stromquelle
│ │                       • Signierte Nutzdaten
│ │   
│ └─USB4 host controller:
│       Gerätekennung:    113b4d4c5f7344e472b94c02229739cd8132057d
│       Zusammenfassung:  Unmatched performance for high-speed I/O
│       Anbieter:         INTEL (THUNDERBOLT:0x8087)
│       GUIDs:            5041b15d-6e1c-5681-86b3-bb1d56993357 ← THUNDERBOLT\VEN_8087&DEV_463E
│                         d651f364-21d8-5cc3-9790-dd019a8958ef ← THUNDERBOLT\VEN_8087&DEV_463E&REV_00
│                         e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│       Device Flags:     • Internes Gerät
│                         • System benötigt externe Stromquelle
│                         • Signierte Nutzdaten
│     
├─Alder Lake-P Integrated Graphics Controller:
│     Gerätekennung:      5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Aktuelle Version:   0c
│     Anbieter:           Intel Corporation (PCI:0x8086)
[... 407 lines cut ...]
      dp: 03 0a 24 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b ad 38 cc bb f7 ed f0 4d 95 9c f4 2a a7 4d 36 50 / 03 18 04 00 / 7f ff 04 00
Boot0023  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35400)
      dp: 03 0a 25 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b ae a2 09 0a df de 21 4e 8b 3a 5e 47 18 56 a3 54 00 / 7f ff 04 00
Boot0024  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ca88c2349e7ae947beeb43038a5aeae700)
      dp: 03 0a 25 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b ca 88 c2 34 9e 7a e9 47 be eb 43 03 8a 5a ea e7 00 / 7f ff 04 00
Boot0025* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 14 / 03 05 06 00 0b 01 / 7f ff 04 00
Boot0026* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 14 / 03 05 06 00 0b 00 / 7f ff 04 00
Boot0027* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
      dp: 03 0a 24 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b 91 af 62 59 56 44 9f 41 a7 b9 1f 4f 89 2a b0 f6 / 7f ff 04 00
Boot0028* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
      dp: 03 0a 24 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b ae a2 09 0a df de 21 4e 8b 3a 5e 47 18 56 a3 54 / 7f ff 04 00

efivar -l | grep fw                                                                                                                                                     
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-7ddc47a5-a390-4571-a943-2636de3504d2-0

**Additional questions**

- Operating system and version: Arch, Gnome/Wayland, linux-hardened
- Have you tried rebooting? Sure
- Is this a regression? 
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2023-03-14:
Can you supply all the requested debugging data with the LANG set to en_US please? I'm afraid my German is super bad.

personaingrata, 2023-03-14:
thanks for looking into this, here's the requested output. please let me know if you need more information.

`$ fwupdmgr --version`                                                          
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.1.19-hardened1-1-hardened
runtime   org.freedesktop.fwupd         1.8.12

`$ fwupdmgr get-devices --show-all-devices`                                    
LENOVO 21AKCTO1WW
│
├─Unknown Device:
│     Device ID:          6b5a06645e27c4e94b6e67d5185c46d11ec9ad67
│     GUID:               c96751f4-16e8-59ea-975d-69d3dac88d7d ← GPIO\ID_INTC1055:00
│   
├─12th Gen Intel Core™ i7-1280P:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x00000429
│ │   Vendor:             Intel
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│ │                       ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│ │                       3ebbde86-d03e-549a-a8fd-02ebf9aa537a ← CPUID\PRO_0&FAM_06&MOD_9A&STP_3
│ │   Device Flags:       • Internal device
│ │ 
│ ├─USB4 host controller:
│ │     Device ID:        c202b5b5e24d8e97ce97f872ce34d40115323411
│ │     Summary:          Unmatched performance for high-speed I/O
│ │     Vendor:           INTEL (THUNDERBOLT:0x8087)
│ │     GUIDs:            5041b15d-6e1c-5681-86b3-bb1d56993357 ← THUNDERBOLT\VEN_8087&DEV_463E
│ │                       d651f364-21d8-5cc3-9790-dd019a8958ef ← THUNDERBOLT\VEN_8087&DEV_463E&REV_00
│ │                       e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│ │     Device Flags:     • Internal device
│ │                       • System requires external power source
│ │                       • Signed Payload
│ │   
│ └─USB4 host controller:
│       Device ID:        54c483b1c38d4297467bfdd5bef286c5a1e803c8
│       Summary:          Unmatched performance for high-speed I/O
│       Vendor:           INTEL (THUNDERBOLT:0x8087)
│       GUIDs:            5041b15d-6e1c-5681-86b3-bb1d56993357 ← THUNDERBOLT\VEN_8087&DEV_463E
│                         d651f364-21d8-5cc3-9790-dd019a8958ef ← THUNDERBOLT\VEN_8087&DEV_463E&REV_00
│                         e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│       Device Flags:     • Internal device
│                         • System requires external power source
│                         • Signed Payload
│     
├─Alder Lake-P Integrated Graphics Controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              eaad9970-8e4d-56da-88ab-41a8c1e2811f ← PCI\VEN_8086&DEV_46A6
│                         ed0b9458-c2f1-54c5-9063-dea8f75b4039 ← PCI\VEN_8086&DEV_46A6&REV_0C
│                         252c4720-3ca4-5c9e-869f-67c99c8a53a0 ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22EA
│                         ada74489-5c34-5122-a8bb-19079db09cfa ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22EA&REV_0C
│                         c4625510-a985-517c-8800-0ecfc6f68c8f ← PCI\VEN_8086&DEV_46A6&REV_00
│                         0f7adb81-1c81-5f97-9953-74396febcba2 ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22EA&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Embedded Controller:
│     Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
│     Summary:            UEFI ESRT device
│     Current version:    0
[... 773 lines cut ...]
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── grubx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux-hardened-fallback.img
├── initramfs-linux-hardened.img
├── initramfs-linux.img
├── intel-ucode.img
├── vmlinuz-linux
└── vmlinuz-linux-hardened

7 directories, 369 files

**Additional questions**

- Operating system and version: Arch, Gnome/Wayland, linux-hardened
- Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): pacman
- Have you tried rebooting? Yes
- Is this a regression?
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No