fwupd/fwupd #5642: Intel Management Engine (NUC 11) not valid?!

achimfrase, 2023-03-23:
**Describe the bug**
My System says that the ME-Firmware is not valid.
But I have the latest version installed
-> [https://downloadmirror.intel.com/774176/TN_0071_ReleaseNotes.pdf](url)

**Steps to Reproduce**
Read report from gnome-settings device security, I have attached it as text-file.
Also I am wondering why my system says **Lenovo**
I don't understand why, because this is an intel NUC 11.

├─Intel Management Engine:
│     Gerätekennung:      2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Zusammenfassung:    UEFI ESRT device
│     Aktuelle Version:   0.35.2039
│     Anbieter:           **Lenovo** (DMI:Intel Corp.)

**Expected behavior**
System should be secure.

**fwupd version information**

fwupdmgr --version
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.12
runtime   org.kernel                    6.2.7-300.fc38.x86_64

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
[all-devices.txt](https://github.com/fwupd/fwupd/files/11054373/all-devices.txt)
```

</details>

**Additional questions**

- Operating system and version:   Fedora Linux 38.20230323.0 (Silverblue Prerelease)
- Have you tried rebooting? -> yes
- Is this a regression? -> Never worked.

[device_security.txt](https://github.com/fwupd/fwupd/files/11054337/device_security.txt)

I hope I have placed that report at the correct place.

Regards, Achim

achimfrase, 2023-03-24:
Dear Richard, thanks for your advice.
To be honest, I still don't understand how the system determines that my ME-Version is to old.
Because the version number 0.35.2039 is the latest that is mentioned in the readme-file of the Firmware documentation.

[image]

So I think I am not going to take any action, because honestly I don't know whom to contact at intel.
I don't think that a "classic" support can help here.

Regards, Achim

hughsie, 2023-09-23:
>  I still don't understand how the system determines that my ME-Version is to old.

https://github.com/fwupd/fwupd/blob/main/plugins/pci-mei/fu-mei-common.c#L36

closed 2023-09-23 by hughsie