fwupd/fwupd #5643: failed to write data to efivarfs: Error writing to file descriptor: Invalid argument

w87, 2023-03-23:
**Describe the bug**
Can not update **UEFI dbx** package... Always running into:
`failed to write data to efivarfs: Error writing to file descriptor: Invalid argument`
Whether its via `plasma-discover` or `fwupdmgr update`. In the latter case:

```shell
root@pc:~# fwupdmgr update
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Devices with no available firmware updates: 
 • MHV2080BH PL
 • SSDPR-CX400-256-G2
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 78 to 217?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the dbx to the latest release from Microsoft which adds         ║
║ insecure versions of grub and shim to the list of forbidden signatures due   ║
║ to multiple discovered security updates.                                     ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures.If the installation fails, you will     ║
║ need to update shim and grub packages before the update can be deployed.     ║
║                                                                              ║
║ Once you have installed this dbx update, any DVD or USB installer images     ║
║ signed with the old signatures may not work correctly.You may have to        ║
║ temporarily turn off secure boot when using recovery or installation media,  ║
║ if new images have not been made available by your distribution.             ║
║                                                                              ║
║ UEFI dbx and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Decompressing…           [***************************************]
Writing…                 [***************************************]
failed to write data to efivarfs: Error writing to file descriptor: Invalid argument
root@pc:~#
```

**Steps to Reproduce**

1. `fwupdmgr update`
2. Confirm with Y

**Expected behavior**
Correct update.

**fwupd version information**

```shell
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-57-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

[... 540 lines cut ...]
```shell

From cat /etc/os-release:
VERSION_ID="22.04"
VERSION_CODENAME=jammy
ID_LIKE="ubuntu debian"

Operating System: KDE neon 5.26
KDE Plasma Version: 5.26.5
Kernel Version: 5.15.0-57-generic (64-bit)
Graphics Platform: X11
Graphics Processor: NVE4

```

- Have you tried rebooting?
  Yes, of course!
- Is this a regression?
  Yes?
- Are you using an NVMe disk?
  How do I check it?
- Is secure boot enabled?
  I think so.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
  It's not Lenovo hardware.

hughsie, 2023-03-23:
> System Product Name

What hardware is this? Are there any firmware updates available (perhaps using Windows to apply the update)?

w87, 2023-03-23:
> What hardware is this? Are there any firmware updates available (perhaps using Windows to apply the update)?

A PC. Processors: 8 × Intel® Core™ i7-3770K CPU @ 3.50GHz
Memory: 31.3 GiB of RAM
I don't know about any firmware updates available. I don't have Windows OS.

hughsie, 2023-03-23:
But what vendor? What model?

w87, 2023-03-23:
Local shop just completed CPU, mem, motherboard etc. for my requirements... So what do you mean by „model”?

w87, 2023-03-23:
Local shop just completed CPU, mem, motherboard etc. for my requirements... So what do you mean by „model”?

Anyway, thank you for such an instant response!
I hoped I'll just be able to Google solution in time... But months are passing, I can't upgrade some system packages because of the problem. A few I had to switch to older kernel version to have my OS booted at all (!) and it worries me more and more.
I appreciate your involvement so much!!!

hughsie, 2023-03-24:
Motherboard vendor and product.

w87, 2023-03-24:
```shell

# dmidecode -t 2
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 2.7 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: P8Z77-V LX
        Version: Rev X.0x
        Serial Number: 121205656802844
        Asset Tag: To be filled by O.E.M.
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: To be filled by O.E.M.
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0
```

hughsie, 2023-03-24:
Are there any BIOS update from ASUS for this hardware?

w87, 2023-03-24:
The most recent one is from 2014, AFAIR I have this version.

hughsie, 2023-03-24:
I don't think we can do anything I'm afraid.

closed 2023-03-24 by hughsie

w87, 2023-03-24:
Is there at least any way to safely remove UEFI dbx and be able to boot my PC correctly?
I mean, as of now I'm unable to upgrade many important system packages, because of UEFI dbx :cry:

w87, 2023-03-31:
> You could turn off secure boot or boot in BIOS compatibility mode.

Thank you, did.

Well, you say this, my package manager says that:

[image]

the **UEFI dbx** package can't be upgraded (_failed to write data to efivarfs: Error writing to file descriptor: Invalid argument_) and blocks upgrade of some other packages.

hughsie, 2023-03-31:
The dbx can't show if you're **running** in CSM mode. The dbx also won't block other packages. If you want to hide the entry you can blacklist the device or plugin in /etc/fwupd/daemon.conf.

w87, 2023-04-06:
I've noticed upgrades w/o _UEFI DBX_ on the list, so I clicked upgrade...
My PC become slow, couldn't work, moving Ctrl+Alt+F3 etc. worked but no luck running there anything... So I had to reboot.

After reboot, I'm not able to start my OS. In the meantime I've checked and had old bios after all. Updated it successfully. But right now I'm stuck:

[image]
[image]
[image]
[image]

This PC is my work tool, please help, I'm terrified and don't know what to do... Please help.

kinghat, 2023-06-25:
i also get this on my hp z440 with fedora kde:

```
$ fwupdmgr update
Devices with no available firmware updates: 
 • BC501 HFM128GDJTNG-8310A
 • BCM20702A0
 • PCIe SSD
 • Sabrent
Devices with the latest available firmware version:
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 220 to 371?                                            ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Insecure versions of the Microsoft Windows boot manager affected by Black    ║
║ Lotus were added to the list of forbidden signatures due to a discovered     ║
║ security problem.This updates the dbx to the latest release from Microsoft.  ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures.Applying this update may also cause     ║
║ some Windows install media to not start correctly.                           ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Writing…                 [                                       ]
failed to write data to efivarfs: Error writing to file descriptor: Invalid argument
```

system:

> Operating System: Fedora Linux 38
> KDE Plasma Version: 5.27.5
> KDE Frameworks Version: 5.107.0
> Qt Version: 5.15.10
> Kernel Version: 6.3.8-200.fc38.x86_64 (64-bit)
> Graphics Platform: Wayland
> Processors: 8 × Intel® Xeon® CPU E5-1620 v3 @ 3.50GHz
> Memory: 62.7 GiB of RAM
> Graphics Processor: AMD Radeon RX 470 Graphics
> Manufacturer: Hewlett-Packard
> Product Name: HP Z440 Workstation

i do have a win 10 install on one of my nvme drives which is meant for vm storage pass-through but i did install fedora after win 10 drive was present. i did boot the win 10 drive directly and applied all the updates as well. i have disabled that drive at boot but still see the UEFI dbx update. not sure if any of that matters or not.

hughsie, 2023-06-25:
@kinghat can you post the dbx section of your fwupdmgr get-devices output please

kinghat, 2023-06-25:
> @kinghat can you post the dbx section of your fwupdmgr get-devices output please

```
$ fwupdmgr get-devices 
Hewlett-Packard HP Z440 Workstation
│
├─82580 Gigabit Network Connection (Ethernet Server Adapter I340-T2):
│     Device ID:          ce6d39e7c84738d80b6899f99329c88aacd67559
│     Current version:    01
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              29b27eef-ddf3-5d19-a03c-a3771aad441b ← PCI\VEN_8086&DEV_1516
│                         018013ba-79c5-5b8a-bc7e-348f506192a0 ← PCI\VEN_8086&DEV_1516&REV_01
│                         dfc4fd9b-2096-542a-a1a1-7a54bb6bc8ac ← PCI\VEN_8086&DEV_1516&SUBSYS_808612B2
│                         17f8509a-c4c5-5a8a-a826-8f69af3d66f1 ← PCI\VEN_8086&DEV_1516&SUBSYS_808612B2&REV_01
│                         2776ed3d-41fb-54eb-88ac-1467a64cc30f ← PCI\VEN_8086&DEV_8D12
│                         2458ea1a-b0aa-5567-91f0-f7367a4b6715 ← PCI\VEN_8086&DEV_8D12&REV_D5
│                         83b986cb-49e0-5dee-be90-542378c64e62 ← PCI\VEN_8086&DEV_8D12&SUBSYS_103C212B
│                         e956c7e7-898f-59bf-9df5-fd225223bf9c ← PCI\VEN_8086&DEV_8D12&SUBSYS_103C212B&REV_D5
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─BC501 HFM128GDJTNG-8310A:
│     Device ID:          310f45f1f223064b5c16bf6dff31146755a64480
│     Summary:            NVM Express solid state drive
│     Current version:    80002C00
│     Vendor:             SK hynix (NVME:0x1C5C)
│     Serial Number:      NS91N546111801F1S
│     GUIDs:              f00c5df1-5b04-56b0-abc4-ec8796b0eb0b ← NVME\VEN_1C5C&DEV_1327
│                         3df71ce6-ec3c-5902-8215-7980cddffb5a ← NVME\VEN_1C5C&DEV_1327&REV_00
│                         3739ab6f-ed9a-550b-b441-e2ecb0e7862b ← NVME\VEN_1C5C&DEV_1327&SUBSYS_1C5C0000
│                         74fea6c6-0bbb-592e-bbce-73d8f8f32244 ← NVME\VEN_1C5C&DEV_1327&SUBSYS_1C5C0000&REV_00
│                         ce5fa4be-0fa0-57d6-a5bf-66677558d0f3 ← SK hynix BC501 HFM128GDJTNG-8310A
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─BCM20702A0:
│     Device ID:          6079b7ffd17ac41e26a182393d4401d7f3e622e7
│     Current version:    1.12
│     Vendor:             Broadcom Corp (USB:0x0A5C)
│     Serial Number:      5CF370885223
│     GUIDs:              a1e76ca3-d35b-53b8-a17c-454b700274ea ← USB\VID_0A5C&PID_21E8
│                         6faa2765-8bc2-51d3-bf6b-ffd6312c24d2 ← USB\VID_0A5C&PID_21E8&REV_0112
│     Device Flags:       • Updatable
│   
├─Ellesmere [Radeon RX 470/480/570/570X/580/580X/590] (Radeon RX 470):
│     Device ID:          5987c136e0024e0f45943927a31c5c4c711f3eb0
│     Current version:    115-D000PI20-100
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002, PCI:0x8086)
│     GUIDs:              55746bd7-4196-594a-9d39-0808222765d1 ← PCI\VEN_1002&DEV_67DF
│                         49baba56-43d7-57d0-8328-9844b9da6a4f ← PCI\VEN_1002&DEV_67DF&REV_CF
│                         55dd4b87-d741-50f8-b0b3-6722db2f2845 ← PCI\VEN_1002&DEV_67DF&SUBSYS_104304B0
│                         7c545160-ccfa-58b0-a6fd-156eae33a90b ← PCI\VEN_1002&DEV_67DF&SUBSYS_104304B0&REV_CF
│                         9f2489b7-b8ca-5a85-b4e4-85eb861e698c ← PCI\VEN_8086&DEV_2F04
│                         f4ddc23d-b403-5a10-a9ee-016cad34d810 ← PCI\VEN_8086&DEV_2F04&REV_02
│                         0f18c699-04f4-5aea-a9d7-c06c46e4cb51 ← PCI\VEN_8086&DEV_2F04&SUBSYS_80860000
│                         22f8af53-0255-5c3c-b3f1-8bba004de296 ← PCI\VEN_8086&DEV_2F04&SUBSYS_80860000&REV_02
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─PCIe SSD:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    ECFM13.3
│     Vendor:             Phison Electronics Corporation (NVME:0x1987)
│     Serial Number:      21032451200345
│     GUIDs:              a44eb54c-5441-56f2-8cc0-5e48964c6457 ← NVME\VEN_1987&DEV_5012
[... 69 lines cut ...]
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
└─Xeon CPU E5-1620 v3 @ 3.50GHz:
      Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
      Current version:    0x00000049
      Vendor:             Intel
      GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
                          30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
                          3dbf8c68-ea66-5b5e-8b50-e2e7c45dabdf ← CPUID\PRO_0&FAM_06&MOD_3F
                          079c5e08-ec0d-5042-bec5-42d236d81c1c ← CPUID\PRO_0&FAM_06&MOD_3F&STP_2
      Device Flags:       • Internal device
    
────────────────────────────────────────────────
Devices that were not updated correctly:
 • UEFI dbx (220 → 371)
Devices that have been updated successfully:
 • System Firmware (2.50 → 2.61)
Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection) [Y|n]: 
```

hughsie, 2023-06-26:
> UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503

Okay, so that's the normal MS PK, not the HP-specific one. @vathpela any idea on what we can do to debug this further? Thanks.

vathpela, 2023-06-26:
To me that looks like a firmware bug, but it's hard to know - either kernel is giving EINVAL on its own (which I don't think it is, but it's not impossible), or SetVariable() is returning EFI_INVALID_ARGUMENT and kernel is translating it. Either way that error code, as always, is effectively "I don't know what's wrong and now you don't either".

hughsie, 2023-06-26:
@kinghat I've asked HP to look at this issue now.

kimonberlin, 2023-06-26:
I haven't heard about this problem before, and it looks like a real thing. We're looking into it.

kimonberlin, 2023-06-28:
I was able to update the dbx from v13 to v371 without issues, using BIOS 2.61.
@hughsie what is the fwupdmgr command to update the dbx to a specific version (220), so I can then try 220 to 371? I can't find it using --help.

hughsie, 2023-06-28:
@kimonberlin `fwupdmgr install` with no arguments allows you to choose the device and the specific release.

kinghat, 2023-08-14:
@kimonberlin were you able to find the issue or need more info perhaps?

kimonberlin, 2023-08-14:
Unable to repro so far. Can you send your repsetup file?

jmontleon, 2023-10-31:
Seeing similar on a 2017 Pixelbook using MrChromeBox coreboot running Fedora 38. It happens whether or not Secure Boot is enabled.

```
$ uname -r
6.5.9-200.fc38.x86_64
```

```
BIOS Information
	Vendor: coreboot
	Version: MrChromebox-4.21.1
	Release Date: 10/29/2023
```

```$ sudo fwupdmgr update 
Devices with no available firmware updates: 
 • KUS040205M-B001
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 220 to 371?                                            ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Insecure versions of the Microsoft Windows boot manager affected by Black    ║
║ Lotus were added to the list of forbidden signatures due to a discovered     ║
║ security problem.This updates the dbx to the latest release from Microsoft.  ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures.Applying this update may also cause     ║
║ some Windows install media to not start correctly.                           ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Writing…                 [                                       ]
failed to write data to efivarfs: Error writing to file descriptor: Invalid argument
```

```
└─UEFI dbx:
      Device ID:          362301da643102b9f38477387e2193e57abaa590
      Summary:            UEFI revocation database
      Current version:    220
      Minimum Version:    220
      Vendor:             UEFI:Linux Foundation
      Install Duration:   1 second
      Update State:       Failed
      Update Error:       failed to write data to efivarfs: Error writing to file descriptor: Invalid argument
      Last modified:      2023-10-31 02:42
      GUIDs:              8301c527-1eb6-5e32-9302-165536509002 ← UEFI\CRT_343F345171011BCD073670DA224AD7B02DDA92BCA093CB1E171FBD2DCB7C6F3E
                          f6bc85a6-f0d7-549d-a677-9dd517bcd9ae ← UEFI\CRT_343F345171011BCD073670DA224AD7B02DDA92BCA093CB1E171FBD2DCB7C6F3E&ARCH_X64
                          c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
      Device Flags:       • Internal device
                          • Updatable
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Only version upgrades are allowed
                          • Signed Payload
    
────────────────────────────────────────────────
Devices that were not updated correctly:
 • UEFI dbx (220 → 371)
```

rafrafek, 2023-11-02:
For me enabling TPM2.0 in UTM (QEMU) helped. Maybe enabling it in a real hardware also helps?

jmontleon, 2023-11-02:
If it comes down to TPM then in the case of my device it might be expected it does not work right now.

Last I knew (from notes on the coreboot 4.20 Release (2023.05.15) at https://mrchromebox.tech/) TPM support is not yet enabled for CR50 devices using his firmware. @MrChromebox do you have any idea whether this is the cause of being unable to update the UEFI dbx with fwupdmgr?

MrChromebox, 2023-11-02:
> If it comes down to TPM then in the case of my device it might be expected it does not work right now.
>
> Last I knew (from notes on the coreboot 4.20 Release (2023.05.15) at [...]

I get this same dbx update failure on TPM1.2 Chromebooks which are fully working in edk2, so it's not that.

MrChromebox, 2023-11-02:
@jmontleon the issue I believe is that coreboot+edk2 doesn't support capsule updates, which is how fwupd is trying to deliver the update. IMO it should not be trying to do so on a system that doesn't support them

superm1, 2023-11-02:
No dbx doesn't go through capsules. It goes through uefi runtime services calls to write to nvram.

MrChromebox, 2023-11-02:
> No dbx doesn't go through capsules. It goes through uefi runtime services calls to write to nvram.

huh, well not sure why that's failing. that should be supported

superm1, 2023-11-02:
It's a secure variable, do you support that ?

MrChromebox, 2023-11-02:
I believe so, and the variable has the following flags set:
EFI_VARIABLE_NON_VOLATILE |
EFI_VARIABLE_RUNTIME_ACCESS |
EFI_VARIABLE_BOOTSERVICE_ACCESS |
EFI_VARIABLE_TIME_BASED_AUTHENTICATED_WRITE_ACCESS

superm1, 2023-11-03:
To the OP - does efibootmgr work to add/remove entries or change boot order or boot next?

That would prove that efi nvram writes work on general.

MrChromebox, 2023-11-03:
hey @superm1, @jmontleon and I are both using my coreboot+edk2 firmware which supports EFI NVRAM, and has a functional boot menu (can add/remove/reorder etc).

superm1, 2023-11-03:
> hey @superm1, @jmontleon and I are both using my coreboot+edk2 firmware which supports EFI NVRAM, and has a functional boot menu (can add/remove/reorder etc).

Specifically changing the boot entries from Linux is working?

The reason I'm asking is to try to narrow down where the fw bug is.

IE if it's a RT services after ExitBootServices bug or a bug with the authenticated variables.

MrChromebox, 2023-11-03:
> Specifically changing the boot entries from Linux is working?

I'll test that RQ now

MrChromebox, 2023-11-03:
> Specifically changing the boot entries from Linux is working?

yes, I changed the bootorder using `efibootmgr -o`, powered off then on, and the boot menu reflected my changes

MrChromebox, 2023-11-03:
I'll look into that in edk2 this weekend

jmontleon, 2023-11-03:
Just had a chance to try and I was able to add and remove entries using efibootmgr as well.

AkechiShiro, 2023-11-26:
I'm getting the same error in NixOS with SecureBoot in QEMU using OVMF, here is a bit more info on the QEMU machine configuration.
```xml
  <os firmware="efi">
    <type arch="x86_64" machine="pc-q35-7.1">hvm</type>
    <firmware>
      <feature enabled="no" name="enrolled-keys"/>
      <feature enabled="yes" name="secure-boot"/>
    </firmware>
    <loader readonly="yes" secure="yes" type="pflash">/usr/share/edk2/x64/OVMF_CODE.secboot.fd</loader>
    <nvram template="/usr/share/edk2/x64/OVMF_VARS.fd">/var/lib/libvirt/qemu/nvram/nixos-dev-unstable_VARS.fd</nvram>
    <bootmenu enable="yes"/>
  </os>
```
Could this error be due to the fact that `/var/lib/libvirt/qemu/nvram/nixos-dev-unstable_VARS.fd` isn't writable/is read-only ?

MrChromebox, 2023-11-26:
known issue, unknown cause. Next release will be pre-update to dbx 371

hughsie, 2023-11-26:
> /var/lib/libvirt/qemu/nvram/nixos-dev-unstable_VARS.fd

If you make it r/w does it work?

AkechiShiro, 2023-11-26:
So here is what I did so far, I chown on my host filesystem the nvram file, to my user, rebooted the qemu vm and here is the error, I get now : 
```
failed to write data to efivarfs: Error writing to file descriptor: No space left on device
```
So, I take it the nvram is full ? And I'd need to expand its storage but that's weird, I don't think it should be full.

AkechiShiro, 2023-11-26:
Maybe if I launch the VM using these options for drives, will it help : 
[image]
I'm not sure, I need to find out a way to convert the XML file to qemu command first.

sunbearc22, 2024-12-09:
Just installed Ubuntu 24.04.1 a few days ago. I notice this error every time I start the system: 

`failed to write data to efivarfs: Error writing to file descriptor: Invalid argument`
 
[image]

Clicking the [<button>Restart & Update</button>]() button won't work. 
```
$ uname -a
Linux Machine 6.8.0-49-generic #49-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov  4 02:06:24 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```
How to fix this issue?

sunbearc22, 2024-12-09:
After updating to the latest firmware, i.e.  [PRIME-X299-A-II-ASUS-180,](https://www.asus.com/us/motherboards-components/motherboards/prime/prime-x299-a-ii/helpdesk_bios?model2Name=Prime-X299-A-II),  the issue that I reported above no longer appears, i.e. seems resolved.

hughsie, 2024-12-09:
Good plan; I guess ASUS added a efivar workaround in the latest firmware. Maybe you could ask them to upload that firmware to the LVFS so it can be distributed to all users?

sunbearc22, 2024-12-09:
@hughsie What is LVFS? As far as I know, the firmware file can be downloaded by anyone from the hyperlink I gave above.

hughsie, 2024-12-09:
LVFS is this: https://fwupd.org/ :)

mattia-b89, 2025-03-16:
_I know, it's a **closed issue** but I do not know elsewhere to share my progress about this issue,_  
_feel free to copy/move the answer above._  

---

After years of searches, I have found a "solution" working for me:  
suspicious about this (apparently unrelated) error `[    4.441872] spi-nor spi0.0: probe with driver spi-nor failed with error -22
` from dmesg kernel log,  
I decided to try to get rid of it, so after investigating I tried by blacklisting a kernel module
```
#
# /etc/modprobe.d/06-spi-intel.conf
#

blacklist spi_intel_platform
```
**long story short,** this also fixed this long-term issue!

---

Arch Linux on a XPS 9343 machine (Intel i5-5200U)

kinghat, 2025-03-21:
i usually have to deselect the firmware update when i use discover to update my system so it doesnt error and hold everything else up. today the update had different text than normal so i thought i would leave it alone and the update went fine. no error and it appears everything is fine now 🤷‍♂

max-aigner, 2025-04-03:
When I do `fwupdmgr report-history` and upload the data to LVFS I get a link to the github page https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-dbx-efivarfs-IO-error which says this is a known issue due to firmware not supporting dbx updates and to update the BIOS or contact the manufacturer. 
This is way more helpful than `failed to write data to efivarfs: Error writing to file descriptor*`. Maybe it could be included in the error message to for clarification?
(ASUS P12R-E)

hughsie, 2025-04-03:
> This is way more helpful than

It's a wiki! :) Please add extra content as required.

max-aigner, 2025-04-03:
Sorry, I wasn't clear: I meant the wiki is more helpful than just the error message by itself. 

I think `fwupdmgr install` should also show the link to the wiki like `fwupdmgr report-history` does for the same error.  It would make it easier to find the helpful wiki because everyone uses `install` but (I guess) few people use `report-history`. 
Hopefully, this explains it better :)

hughsie, 2025-04-03:
> I think fwupdmgr install should also show the link to the wiki like fwupdmgr report-history does for the same error.

`fwupdmgr install` is "offline" -- it doesn't do any communication with the server. `fwupdmgr report-history` reports what happened back to the LVFS and gets the response -- if you use `fwupdmgr update` it'll ask you if you want to report if the update is "live" -- otherwise we need to do the 2nd step after the reboot comes back.

DjDiabolik, 2025-06-11:
I have right now the same issue on my Ubuntu 24.04.2 LTS!
```
diabolik@NUC-Ubuntu:~$ fwupdmgr update
Devices with no available firmware updates: 
 • HS-SSD-E100 512G
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 20241101 to 20250507?                                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ Some insecure versions of BiosFlashShell and Dtbios by DT Research Inc were  ║
║ added, due to a security vulnerability that allowed an attacker to bypass    ║
║ UEFI Secure Boot.                                                            ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Scrittura…               [                                       ]
failed to write data to efivarfs: Errore nello scrivere sul descrittore di file: Argomento non valido
diabolik@NUC-Ubuntu:~$ 
```
how i can fix this issue ?

hughsie, 2025-06-12:
Please see this: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-dbx-efivarfs-IO-error