fwupd/fwupd #5645: Wacom CTL 4100WL (Intuos BT S) not detected after update

hardboprobot, 2023-03-24:
**Describe the bug**

**Reporting on behalf of Allion**:

As far as I can tell (I didn't reproduce this myself), the device was updated at some point and after that it doesn't show up on the 'get-devices' list.

**Steps to Reproduce**
Update the device to fw 2.09 then run `fwupdmgr get-devices` again

**Expected behavior**
The device shouldn't be displayed by fwupdmgr

**fwupd version information**

```shell
fwupdmgr --version
```
```
runtime   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.2
compile   org.freedesktop.gusb          0.4.5
compile   com.hughsie.libjcat           0.1.8
runtime   org.kernel                    5.10.173-21969-gaa7861c4a57d
runtime   org.freedesktop.gusb          0.4.5
compile   org.freedesktop.fwupd         1.8.12
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Pre-installed in ChromeOS R113-15382.0.0

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

```
{
  "Devices" : [
    {
      "Name" : "ELAN2703:00 04F3:323B",
      "DeviceId" : "1f532bc465a37e0384a117d5f2da9c04a79b5de5",
      "InstanceIds" : [
        "HIDRAW\\VEN_04F3&DEV_323B",
        "HIDRAW\\VEN_04F3&DEV_323B&REV_00",
        "HIDRAW\\VEN_04F3&DEV_323B&MOD_0130",
        "ELANTP\\ICTYPE_13",
        "ELANTP\\ICTYPE_13&MOD_0130",
        "ELANTP\\ICTYPE_13&MOD_0130&DRIVER_HID"
      ],
      "Guid" : [
        "0a6e154e-7e6e-5b7e-ae3c-06bc247df555",
        "e3af00d0-57bf-5651-88c0-7314727f721a",
        "bfae2c46-3ffc-53eb-a994-b6c37c27b4b8",
        "4f9fe0ae-f26e-5c1d-8a4e-a7506fb02f96",
        "4184508c-a984-5a3c-893c-6bae98f2fcf4",
        "ba73bfad-c59c-5d19-82ae-c730143b5979"
      ],
      "Summary" : "Touchpad",
      "Plugin" : "elantp",
      "Protocol" : "tw.com.emc.elantp",
      "Flags" : [
        "internal",
        "updatable",
        "registered"
      ],
      "Vendor" : "ELAN Microelectronics",
      "VendorId" : "HIDRAW:0x04F3",
      "Version" : "0x000c",
      "VersionBootloader" : "0x0003",
      "VersionFormat" : "hex",
      "VersionRaw" : 12,
      "Icons" : [
        "input-touchpad"
      ],
      "Created" : 1679464165
    },
    {
      "Name" : "HapticPad EEPROM",
      "DeviceId" : "ae348b09411b8010d974d966a6434e4f69332ad7",
      "ParentDeviceId" : "1f532bc465a37e0384a117d5f2da9c04a79b5de5",
      "CompositeId" : "1f532bc465a37e0384a117d5f2da9c04a79b5de5",
      "InstanceIds" : [
        "HIDRAW\\VEN_04F3&DEV_323B&DRIVERIC_0002&MOD_0130",
        "ELANTP\\ICTYPE_13",
        "ELANTP\\ICTYPE_13&DRIVERIC_0002&MOD_0130",
        "ELANTP\\ICTYPE_13&DRIVERIC_0002&MOD_0130&DRIVER_HID"
      ],
      "Guid" : [
        "58279a02-cb82-5c04-a87b-8b7302187077",
        "4f9fe0ae-f26e-5c1d-8a4e-a7506fb02f96",
        "3a4b6035-912d-5c64-a1d2-4bcba1686e16",
        "35d08e43-460d-57b6-ba3c-ec671a4cdc9d"
      ],
      "Plugin" : "elantp",
      "Protocol" : "tw.com.emc.elantp.haptic",
      "Flags" : [
        "internal",
        "updatable",
        "registered"
      ],
      "Vendor" : "ELAN Microelectronics",
      "VendorId" : "HIDRAW:0x04F3",
      "Version" : "22040802",
      "VersionBootloader" : "0x0001",
      "VersionFormat" : "plain",
      "Icons" : [
        "input-touchpad"
      ],
      "Created" : 1679464165
    },
    {
      "Name" : "Display controller",
      "DeviceId" : "5792b48846ce271fab11c4a545f7a3df0d36e00a",
      "InstanceIds" : [
        "PCI\\VEN_8086&DEV_46B3",
        "PCI\\VEN_8086&DEV_46B3&REV_0C",
        "PCI\\VEN_8086&DEV_46B3&SUBSYS_808646B3",
        "PCI\\VEN_8086&DEV_46B3&SUBSYS_808646B3&REV_0C",
        "PCI\\VEN_8086&DEV_46B3&REV_00",
        "PCI\\VEN_8086&DEV_46B3&SUBSYS_808646B3&REV_00"
      ],
      "Guid" : [
        "86b8e8a0-4c9c-5607-96a7-04073caa6063",
        "af3ee7cb-cacc-57a0-8b4d-0f516ed2eba0",
        "d8f3a7d0-9f81-5ec6-bb0e-a295b84aa6b0",
        "348fe5b5-7c38-57bf-890c-75d0c507d269",
        "0223f94c-a22f-5abc-9e50-dffff268313a",
        "7b5c6ed9-13cd-5a7f-972a-c4ab9233bd83"
      ],
      "Plugin" : "optionrom",
      "Flags" : [
        "internal",
        "registered",
        "can-verify",
        "can-verify-image"
      ],
      "VendorId" : "PCI:0x8086",
      "Version" : "0c",
      "VersionFormat" : "plain",
      "Created" : 1679464162
    },
    {
      "Name" : "Integrated Webcam?",
      "DeviceId" : "08d460be0f1f9f128413f816022a6439e0078018",
      "Guid" : [
        "b585990a-003e-5270-89d5-3705a17f9a43"
      ],
      "Summary" : "Fake webcam",
      "Plugin" : "test",
      "Protocol" : "com.acme.test",
      "Flags" : [
        "updatable",
        "require-ac",
        "supported",
        "registered",
        "can-verify",
        "can-verify-image",
        "unsigned-payload"
      ],
      "Vendor" : "ACME Corp.",
      "VendorId" : "USB:0x046D",
      "Version" : "1.2.2",
      "VersionLowest" : "1.2.0",
      "VersionBootloader" : "0.1.2",
      "VersionFormat" : "triplet",
      "Icons" : [
        "preferences-desktop-keyboard"
      ],
      "Created" : 1679464162,
      "Releases" : [
        {
          "AppstreamId" : "org.fwupd.fakedevice.firmware",
          "RemoteId" : "fwupd-tests",
          "Name" : "FakeDevice",
          "Summary" : "Firmware for the ACME Corp Integrated Webcam",
          "Description" : "<p>Fixes another bug with the flux capacitor to prevent time going backwards.</p>",
          "Version" : "1.2.4",
          "Filename" : "fakedevice124.cab",
          "Checksum" : [
            "cffec02f8094d10552b68967425613d72d72ac02",
            "3fab34cfa1ef97238fb24c5e40a979bc544bb2b0967b863e43e7d58e0d9a923f"
          ],
          "License" : "GPL-2.0+",
          "Size" : 10,
          "Created" : 1497484800,
          "Locations" : [
            "https://storage.googleapis.com/chromeos-localmirror/lvfs/test/3fab34cfa1ef97238fb24c5e40a979bc544bb2b0967b863e43e7d58e0d9a923f-fakedevice124.cab"
          ],
          "Uri" : "https://storage.googleapis.com/chromeos-localmirror/lvfs/test/3fab34cfa1ef97238fb24c5e40a979bc544bb2b0967b863e43e7d58e0d9a923f-fakedevice124.cab",
          "Homepage" : "http://www.acme.com/",
          "Vendor" : "ACME Corp",
          "Flags" : [
            "is-upgrade"
          ]
        }
      ]
    },
    {
      "Name" : "USB4 host controller",
      "DeviceId" : "634c305bffd4653afe50f1f41a77a0c402a7751e",
      "InstanceIds" : [
        "THUNDERBOLT\\VEN_8087&DEV_463E",
        "THUNDERBOLT\\VEN_8087&DEV_463E&REV_00",
        "TBT-fixed"
      ],
      "Guid" : [
        "5041b15d-6e1c-5681-86b3-bb1d56993357",
        "d651f364-21d8-5cc3-9790-dd019a8958ef",
        "e72e778e-94f7-5ed2-b560-1c1262ee217c"
      ],
      "Summary" : "Unmatched performance for high-speed I/O",
      "Plugin" : "thunderbolt",
      "Protocol" : "com.intel.thunderbolt",
      "Flags" : [
        "internal",
        "require-ac",
        "registered",
        "signed-payload"
      ],
      "Vendor" : "INTEL",
      "VendorId" : "THUNDERBOLT:0x8087",
      "VersionFormat" : "pair",
      "Icons" : [
        "thunderbolt"
      ],
      "Created" : 1679464165
    },
    {
      "Name" : "USB4 host controller",
      "DeviceId" : "344564e2153b28f33ea40760f21be679da422e54",
      "InstanceIds" : [
        "THUNDERBOLT\\VEN_8087&DEV_463E",
        "THUNDERBOLT\\VEN_8087&DEV_463E&REV_00",
        "TBT-fixed"
      ],
      "Guid" : [
        "5041b15d-6e1c-5681-86b3-bb1d56993357",
        "d651f364-21d8-5cc3-9790-dd019a8958ef",
        "e72e778e-94f7-5ed2-b560-1c1262ee217c"
      ],
      "Summary" : "Unmatched performance for high-speed I/O",
      "Plugin" : "thunderbolt",
      "Protocol" : "com.intel.thunderbolt",
      "Flags" : [
        "internal",
        "require-ac",
        "registered",
        "signed-payload"
      ],
      "Vendor" : "INTEL",
      "VendorId" : "THUNDERBOLT:0x8087",
      "VersionFormat" : "pair",
      "Icons" : [
        "thunderbolt"
      ],
      "Created" : 1679464165
    }
  ]
}
```

**Additional questions**

- Operating system and version: ChromeOS R113-15382.0.0
- Have you tried rebooting? Not sure
- Is this a regression? Not sure

hughsie, 2023-03-24:
I think you need https://github.com/fwupd/fwupd/commit/cb9f823b868362db73ae2eb04da49f88fc710cb4 -- this fixed the problem for me. I can share the wacom firmware team emails if that would be helpful.

hardboprobot, 2023-03-27:
Thanks! if the fix will be in the next release (1.8.13) I think it'll be enough to remember that we need to test it again after that.

hughsie, 2023-03-27:
Yup, I'm planning 1.8.13 for the end of this week if that helps.

hughsie, 2023-03-27:
Lets close, but please re-open if that fails.

d4s, 2023-05-04:
@hughsie the same issue  for this device, **connected over BT**:
```
2023-04-11T03:05:03.237052Z WARNING kernel: [ 1740.455117] wacom 0005:056A:03C8.0006: unknown main item tag 0x0
2023-04-11T03:05:03.237097Z INFO kernel: [ 1740.455500] input: Wacom Intuos BT M Pen as /devices/virtual/misc/uhid/0005:056A:03C8.0006/input/input26
2023-04-11T03:05:03.238038Z INFO kernel: [ 1740.455805] input: Wacom Intuos BT M Pad as /devices/virtual/misc/uhid/0005:056A:03C8.0006/input/input28
2023-04-11T03:05:03.238064Z INFO kernel: [ 1740.456098] wacom 0005:056A:03C8.0006: hidraw2: BLUETOOTH HID v1.00 Device [Wacom Intuos PBM] on 38:7a:0e:9e:16:ea
```

@hughsie do you prefer to re-open this issue or create a new one dedicated for BT mode?

hughsie, 2023-05-05:
Let's just use this one. Got a -vv daemon log?

d4s, 2023-05-31:
Got at last.
I do not see the device in quirk files.

```
2023-05-30T08:54:08.279153Z WARNING fwupd[11224]: FuEngine             /sys/devices/virtual/misc/uhid/0005:056A:03C6.0003/hidraw/hidraw2 added FuUdevDevice
2023-05-30T08:54:08.279153Z WARNING fwupd[11224]:  Flags:                none
2023-05-30T08:54:08.279153Z WARNING fwupd[11224]:  BackendId:            /sys/devices/virtual/misc/uhid/0005:056A:03C6.0003/hidraw/hidraw2
2023-05-30T08:54:08.279153Z WARNING fwupd[11224]:  AcquiesceDelay:       2500
2023-05-30T08:54:08.279153Z WARNING fwupd[11224]:  Subsystem:            hidraw
2023-05-30T08:54:08.279153Z WARNING fwupd[11224]:  DeviceFile:           /dev/hidraw2
2023-05-30T08:54:08.279153Z WARNING fwupd[11224]:  SysfsPath:            /sys/devices/virtual/misc/uhid/0005:056A:03C6.0003/hidraw/hidraw2
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]: FuEngine             /sys/devices/virtual/misc/uhid/0005:056A:03C6.0003/hidraw/hidraw2 added FuUdevDevice
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  Name:                 Wacom Intuos PBS
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  Guid:                 4ba8ebc0-d251-5b5c-b5ad-d9a340b577e2 <- HIDRAW\VEN_056A&DEV_03C6 ?
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  Guid:                 38fbc2bf-7db8-50a2-b3a9-3370ef2c28d0 <- HIDRAW\VEN_056A&DEV_03C6&REV_00 ?
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  Flags:                none
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  VendorId:             HIDRAW:0x056A
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  Guid[quirks]:         d1141d81-b90d-53c9-9dc2-6c312a9afa21 <- HIDRAW\VEN_056A
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  Guid[quirks]:         eb45b692-59a3-5c1d-8ddb-950ccd241df4 <- HIDRAW
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  BackendId:            /sys/devices/virtual/misc/uhid/0005:056A:03C6.0003/hidraw/hidraw2
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  AcquiesceDelay:       2500
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  Vendor:               0x56a
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  Model:                0x3c6
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  Subsystem:            hidraw
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  DeviceFile:           /dev/hidraw2
2023-05-30T08:54:08.279840Z WARNING fwupd[11224]:  SysfsPath:            /sys/devices/virtual/misc/uhid/0005:056A:03C6.0003/hidraw/hidraw2
```

jigpu, 2023-06-06:
@d4s Updating the CTL-4100WL over Bluetooth is not supported. The Bluetooth mode uses a different VID:PID which is intentionally not listed in the quirks file because of this.

Does update over USB work as expected?

closed 2023-06-07 by d4s