fwupd/fwupd #5656: fwupd 1.8.13 core dumps

bulletmark, 2023-03-28:
fwupd updated on Arch today: `2023-03-29 08:01:38 fwupd 1.8.12-1 -> 1.8.13-1` so I did a refresh and attempted update of my devices:
```
$ sudo fwupdmgr update
Devices with no available firmware updates: 
 • Integrated Webcam HD
 • Internal SPI Controller
 • PC611 NVMe SK hynix 512GB
 • TPM
 • TPM 2.0
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19TB
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.15.0 to 1.16.0?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Firmware updates to address security vulnerabilities.                      ║
║                                                                              ║
║ XPS 13 9300 must remain plugged into a power source for the duration of the  ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Authenticating…          [***************************************]
Message recipient disconnected from message bus without replying
```

Last message is because fwupd core dumps:

```
           PID: 2070 (fwupd)
           UID: 0 (root)
           GID: 0 (root)
        Signal: 6 (ABRT)
     Timestamp: Wed 2023-03-29 09:28:19 AEST (1min 23s ago)
  Command Line: /usr/lib/fwupd/fwupd
    Executable: /usr/lib/fwupd/fwupd
 Control Group: /system.slice/fwupd.service
          Unit: fwupd.service
         Slice: system.slice
       Boot ID: b3b4b7369e204189ae33f5ce9b00d3fb
    Machine ID: 9eabba1a60b9408c970e3e88cfc74c46
      Hostname: lt
       Storage: /var/lib/systemd/coredump/core.fwupd.0.b3b4b7369e204189ae33f5ce9b00d3fb.2070.1680046099000000.zst (inaccessibl>
       Message: Process 2070 (fwupd) of user 0 dumped core.
                
                Stack trace of thread 2070:
                #0  0x00007fd0a45ea8ec n/a (libc.so.6 + 0x878ec)
                #1  0x00007fd0a459bea8 raise (libc.so.6 + 0x38ea8)
                #2  0x00007fd0a458553d abort (libc.so.6 + 0x2253d)
                #3  0x00007fd0a458629e n/a (libc.so.6 + 0x2329e)
                #4  0x00007fd0a45f4657 n/a (libc.so.6 + 0x91657)
                #5  0x00007fd0a45f52ec n/a (libc.so.6 + 0x922ec)
                #6  0x00007fd0a45f77d0 n/a (libc.so.6 + 0x947d0)
                #7  0x00007fd0a45f88ed malloc (libc.so.6 + 0x958ed)
                #8  0x00007fd0a43ba406 n/a (libsqlite3.so.0 + 0x17406)
                #9  0x00007fd0a43b6f50 n/a (libsqlite3.so.0 + 0x13f50)
                #10 0x00007fd0a43b7120 n/a (libsqlite3.so.0 + 0x14120)
                #11 0x00007fd0a444a311 n/a (libsqlite3.so.0 + 0xa7311)
                #12 0x00007fd0a44a62ad n/a (libsqlite3.so.0 + 0x1032ad)
                #13 0x00007fd0a44a8bc9 n/a (libsqlite3.so.0 + 0x105bc9)
                #14 0x00007fd0a4451cc7 n/a (libsqlite3.so.0 + 0xaecc7)
                #15 0x00007fd0a4420c0f n/a (libsqlite3.so.0 + 0x7dc0f)
                #16 0x00007fd0a4421403 n/a (libsqlite3.so.0 + 0x7e403)
                #17 0x00007fd0a442157a sqlite3_prepare_v2 (libsqlite3.so.0 + 0x7e57a)
                #18 0x00007fd0a4c7fa04 fu_history_remove_device (libfwupdengine.so + 0x7fa04)
                #19 0x00007fd0a4c7fc3b fu_history_add_device (libfwupdengine.so + 0x7fc3b)
[... 290 lines cut ...]
│     Update State:       Success
│     GUIDs:              02a963b2-82d6-5ab7-bd72-bcd6c2113772 ← 096d-2.0
│                         ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
│                         73730635-f6c2-53da-9df2-948bb5ac1022 ← DELL-TPM-2.0-STM-
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│   
└─Unifying Receiver:
      Device ID:          3d9add6326f028f1110de91231fd67ec59897503
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR24.11_B0036
      Bootloader Version: BOT03.02_B0009
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
      Device Flags:       • Updatable
                          • Supported on remote server
                          • Signed Payload
```

Happens every time. Rebooted and no change.

hughsie, 2023-03-29:
Hmm, I can reproduce -- and there's no changes in 1.8.13 that could explain this issue. Can you please install the debug symbols for both fwupd and sqlite3 please and then run the daemon with `sudo gdb --args /usr/libexec/fwupd/fwupd --verbose` using `r`  then `enter` to run it, then in another terminal do the update with `fwupdmgr update` and then when it crashes do  `bt`  then `enter`. Then post the full backtrace here -- thanks.

bbc2, 2023-03-29:
I have a very similar issue with different hardware (ThinkPad laptop). Unfortunately, I couldn't test with 1.8.12 so I don't know if this is due to the upgrade.

<details>
<summary>Stack trace</summary>

I got the stack trace using `DEBUGINFOD_URLS="https://debuginfod.archlinux.org/" coredumpctl gdb` in Arch Linux.

```gdb
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007f534223c953 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007f53421edea8 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007f53421d753d in __GI_abort () at abort.c:79
#4  0x00007f53421d829e in __libc_message (fmt=fmt@entry=0x7f534234f77e "%s\n") at ../sysdeps/posix/libc_fatal.c:150
#5  0x00007f5342246657 in malloc_printerr (str=str@entry=0x7f5342352400 "malloc_consolidate(): unaligned fastbin chunk detected") at malloc.c:5651
#6  0x00007f53422472ec in malloc_consolidate (av=av@entry=0x7f534238daa0 <main_arena>) at malloc.c:4736
#7  0x00007f53422497d0 in _int_malloc (av=av@entry=0x7f534238daa0 <main_arena>, bytes=bytes@entry=1072) at malloc.c:3951
#8  0x00007f534224a8ed in __GI___libc_malloc (bytes=bytes@entry=1072) at malloc.c:3305
#9  0x00007f5341ff2406 in sqlite3MemMalloc (nByte=1072) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:25585
#10 0x00007f5341feef50 in mallocWithAlarm (pp=<synthetic pointer>, n=<optimized out>) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:29287
#11 sqlite3Malloc (n=<optimized out>) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:29333
#12 sqlite3Malloc (n=<optimized out>) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:29327
#13 0x00007f5341fef120 in dbMallocRawFinish (db=0x55e9961aad30, n=<optimized out>) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:29638
#14 0x00007f5342082311 in sqlite3WhereBegin (pParse=0x7ffd35e3b620, pTabList=0x55e9964191f0, pWhere=0x55e99649c290, pOrderBy=0x0, pResultSet=0x0, pSelect=0x0, wctrlFlags=28, iAuxArg=1)
    at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:162812
#15 0x00007f53420de2ad in sqlite3DeleteFrom.constprop.0 (pParse=pParse@entry=0x7ffd35e3b620, pTabList=0x55e9964191f0, pWhere=0x55e99649c290, pLimit=0x0, pOrderBy=0x0)
    at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:124240
#16 0x00007f53420e0bc9 in yy_reduce.constprop.0 (yypParser=0x7ffd35e3ac30, yyruleno=<optimized out>, yyLookaheadToken=..., pParse=0x7ffd35e3b620, yyLookahead=<optimized out>)
    at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:171015
#17 0x00007f5342089cc7 in sqlite3Parser (yyminor=<error reading variable: DWARF-2 expression error: DW_OP_GNU_uninit must always be the very last op.>, yymajor=<optimized out>, yyp=0x7ffd35e3ac30)
    at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:172119
#18 sqlite3RunParser (pParse=<optimized out>, zSql=<optimized out>) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:42346
#19 0x00007f5342058c0f in sqlite3Prepare
    (db=db@entry=0x55e9961aad30, zSql=zSql@entry=0x7f534298cc48 "DELETE FROM history WHERE device_id = ?1;", nBytes=nBytes@entry=-1, prepFlags=prepFlags@entry=128, pReprepare=pReprepare@entry=0x0, ppStmt=ppStmt@entry=0x7ffd35e3b890, pzTail=0x0) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:137981
#20 0x00007f5342059403 in sqlite3LockAndPrepare (pzTail=0x0, ppStmt=0x7ffd35e3b890, pOld=0x0, prepFlags=128, nBytes=-1, zSql=0x7f534298cc48 "DELETE FROM history WHERE device_id = ?1;", db=0x55e9961aad30)
    at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:138056
#21 sqlite3LockAndPrepare
    (db=0x55e9961aad30, zSql=zSql@entry=0x7f534298cc48 "DELETE FROM history WHERE device_id = ?1;", nBytes=nBytes@entry=-1, prepFlags=prepFlags@entry=128, pOld=pOld@entry=0x0, ppStmt=ppStmt@entry=0x7ffd35e3b890, pzTail=0x0) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:6959
#22 0x00007f534205957a in sqlite3_prepare_v2
    (db=<optimized out>, zSql=zSql@entry=0x7f534298cc48 "DELETE FROM history WHERE device_id = ?1;", nBytes=nBytes@entry=-1, ppStmt=ppStmt@entry=0x7ffd35e3b890, pzTail=pzTail@entry=0x0)
    at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:138142
#23 0x00007f534287fa04 in fu_history_remove_device (self=self@entry=0x55e99618dce0, device=device@entry=0x55e99635f570, error=error@entry=0x7ffd35e3ba90) at ../fwupd-1.8.13/src/fu-history.c:902
#24 0x00007f534287fc3b in fu_history_add_device (self=0x55e99618dce0, device=0x55e99635f570, release=0x55e996472b20, error=0x7ffd35e3ba90) at ../fwupd-1.8.13/src/fu-history.c:759
#25 0x00007f534286cc00 in fu_engine_install_release
    (self=self@entry=0x55e996189e10, release=release@entry=0x55e996472b20, blob_cab=blob_cab@entry=0x55e9967d60a0, progress=0x55e9963a14e0, flags=flags@entry=FWUPD_INSTALL_FLAG_NONE, error=error@entry=0x7ffd35e3ba90)
    at ../fwupd-1.8.13/src/fu-engine.c:3191
#26 0x00007f534286d337 in fu_engine_install_releases
    (self=0x55e996189e10, request=<optimized out>, releases=0x55e996241a00, blob_cab=0x55e9967d60a0, progress=progress@entry=0x55e9961cedc0, flags=FWUPD_INSTALL_FLAG_NONE, error=0x7ffd35e3ba90)
    at ../fwupd-1.8.13/src/fu-engine.c:2714
#27 0x000055e99522c56b in fu_daemon_authorize_install_queue (helper_ref=helper_ref@entry=0x55e9967a8550) at ../fwupd-1.8.13/src/fu-daemon.c:737
[... cut ...]
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.13
runtime   org.kernel                    6.2.8-arch1-1
```

I hope this helps! If this turns out to be a different issue I'm happy to open a new issue in GitHub.

hughsie, 2023-03-29:
Is this somehow arch specific? Could you try downgrading libsqlite3 perhaps?

MichaelEischer, 2023-03-29:
I'm also seeing the same crash as in the original post. Downgrading the Arch sqlite package to 3.41.1-1 does not resolve the issue. But downgrading fwupdmgr to 1.8.12-1 while still using the latest sqlite package works. Deleting all files in `/var/{cache.lib}/fwupd` also does not fix the issue.

<details>
<summary>Log from fwupd --verbose</summary>

```
Mar 29 23:26:07 fwupd[8130]: 21:26:07.251 FuRelease            remote lvfs has kind=download and so marking as trusted
Mar 29 23:26:17 fwupd[8130]: 21:26:17.025 FuDeviceList         no replug or re-enumerate required
Mar 29 23:26:17 fwupd[8130]: 21:26:17.234 FuCabinet            verified metadata firmware.metainfo.xml: 3
Mar 29 23:26:17 fwupd[8130]: 21:26:17.234 FuCabinet            processing file: firmware.metainfo.xml
Mar 29 23:26:17 fwupd[8130]: 21:26:17.234 FuCabinet            processing release: 0.1.38
Mar 29 23:26:17 fwupd[8130]: 21:26:17.495 FuCabinet            verified payload firmware.bin: 3
Mar 29 23:26:17 fwupd[8130]: 21:26:17.597 FuEngine             composite update 1: d96de5c124b60ed6241ebcb6bb2c839cb5580786
Mar 29 23:26:17 fwupd[8130]: malloc_consolidate(): unaligned fastbin chunk detected
Mar 29 23:26:17 systemd[1]: fwupd.service: Main process exited, code=dumped, status=6/ABRT
Mar 29 23:26:17 systemd[1]: fwupd.service: Failed with result 'core-dump'.
```
</details>

<details>
<summary>Stacktrace</summary>

```
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007f3e648a8953 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007f3e64859ea8 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007f3e6484353d in __GI_abort () at abort.c:79
#4  0x00007f3e6484429e in __libc_message (fmt=fmt@entry=0x7f3e649bb77e "%s\n") at ../sysdeps/posix/libc_fatal.c:150
#5  0x00007f3e648b2657 in malloc_printerr (str=str@entry=0x7f3e649be400 "malloc_consolidate(): unaligned fastbin chunk detected") at malloc.c:5651
#6  0x00007f3e648b32ec in malloc_consolidate (av=av@entry=0x7f3e649f9aa0 <main_arena>) at malloc.c:4736
#7  0x00007f3e648b57d0 in _int_malloc (av=av@entry=0x7f3e649f9aa0 <main_arena>, bytes=bytes@entry=1072) at malloc.c:3951
#8  0x00007f3e648b68ed in __GI___libc_malloc (bytes=bytes@entry=1072) at malloc.c:3305
#9  0x00007f3e6465e406 in sqlite3MemMalloc (nByte=1072) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:25585
#10 0x00007f3e6465af50 in mallocWithAlarm (pp=<synthetic pointer>, n=<optimized out>) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:29287
#11 sqlite3Malloc (n=<optimized out>) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:29333
#12 sqlite3Malloc (n=<optimized out>) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:29327
#13 0x00007f3e6465b120 in dbMallocRawFinish (db=0x55a6e19fcdc0, n=<optimized out>) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:29638
#14 0x00007f3e646ee311 in sqlite3WhereBegin (pParse=0x7ffe3885e700, pTabList=0x55a6e1a9c080, pWhere=0x55a6e1c0acb0, pOrderBy=0x0, pResultSet=0x0, pSelect=0x0, wctrlFlags=28, iAuxArg=1)
    at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:162812
#15 0x00007f3e6474a2ad in sqlite3DeleteFrom.constprop.0 (pParse=pParse@entry=0x7ffe3885e700, pTabList=0x55a6e1a9c080, pWhere=0x55a6e1c0acb0, pLimit=0x0, pOrderBy=0x0) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:124240
#16 0x00007f3e6474cbc9 in yy_reduce.constprop.0 (yypParser=0x7ffe3885dd10, yyruleno=<optimized out>, yyLookaheadToken=..., pParse=0x7ffe3885e700, yyLookahead=<optimized out>) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:171015
#17 0x00007f3e646f5cc7 in sqlite3Parser (yyminor=<error reading variable: DWARF-2 expression error: DW_OP_GNU_uninit must always be the very last op.>, yymajor=<optimized out>, yyp=0x7ffe3885dd10)
    at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:172119
#18 sqlite3RunParser (pParse=<optimized out>, zSql=<optimized out>) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:42346
#19 0x00007f3e646c4c0f in sqlite3Prepare
    (db=db@entry=0x55a6e19fcdc0, zSql=zSql@entry=0x7f3e6518cc48 "DELETE FROM history WHERE device_id = ?1;", nBytes=nBytes@entry=-1, prepFlags=prepFlags@entry=128, pReprepare=pReprepare@entry=0x0, ppStmt=ppStmt@entry=0x7ffe3885e970, pzTail=0x0) at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:137981
#20 0x00007f3e646c5403 in sqlite3LockAndPrepare (pzTail=0x0, ppStmt=0x7ffe3885e970, pOld=0x0, prepFlags=128, nBytes=-1, zSql=0x7f3e6518cc48 "DELETE FROM history WHERE device_id = ?1;", db=0x55a6e19fcdc0)
    at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:138056
#21 sqlite3LockAndPrepare
    (db=0x55a6e19fcdc0, zSql=zSql@entry=0x7f3e6518cc48 "DELETE FROM history WHERE device_id = ?1;", nBytes=nBytes@entry=-1, prepFlags=prepFlags@entry=128, pOld=pOld@entry=0x0, ppStmt=ppStmt@entry=0x7ffe3885e970, pzTail=0x0)
    at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:6959
#22 0x00007f3e646c557a in sqlite3_prepare_v2 (db=<optimized out>, zSql=zSql@entry=0x7f3e6518cc48 "DELETE FROM history WHERE device_id = ?1;", nBytes=nBytes@entry=-1, ppStmt=ppStmt@entry=0x7ffe3885e970, pzTail=pzTail@entry=0x0)
    at /usr/src/debug/sqlite/sqlite-src-3410200/sqlite3.c:138142
#23 0x00007f3e6507fa04 in fu_history_remove_device (self=self@entry=0x55a6e19c0180, device=device@entry=0x55a6e1b7fa40, error=error@entry=0x7ffe3885eb70) at ../fwupd-1.8.13/src/fu-history.c:902
[... cut ...]
==8865==    by 0x48DBE49: fu_uefi_nvram_device_report_metadata_pre (fu-uefi-nvram-device.c:126)
==8865==    by 0x4B48849: fu_device_report_metadata_pre (fu-device.c:5012)
==8865==    by 0x48BCF45: fu_engine_add_release_metadata (fu-engine.c:2788)
==8865==    by 0x48BF849: fu_engine_install_release (fu-engine.c:3189)
==8865==    by 0x48C0336: fu_engine_install_releases (fu-engine.c:2714)
==8865==    by 0x11156A: fu_daemon_authorize_install_queue (fu-daemon.c:737)
==8865==    by 0x1116A9: fu_daemon_authorize_install_cb (fu-daemon.c:693)
==8865== 
```
</details>

bulletmark, 2023-03-29:
@hughsie regarding your request to me to run this debug symbols. After I raised this bug yesterday I downgraded fwupd to 1.8.12-1 and ran my pending BIOS update which worked successfully so unfortunately I can not test this issue again with 1.8.13-1 because there are no pending updates.

afontenot, 2023-03-30:
Incidentally, as this might be an Arch specific issue, I have reported this downstream here: https://bugs.archlinux.org/task/78039

hughsie, 2023-03-30:
@MichaelEischer the smoking gun - thanks!

freswa, 2023-03-30:
If anyone wants to bt with `gdb`, the easy way ist to install `debuginfod`, make sure `echo $DEBUGINFOD_URLS` prints something (otherwise re-login or run `/etc/profile.d/debuginfod.sh`).
You can then run `gdb` as mentioned by @hughsie via `sudo gdb --args /usr/libexec/fwupd/fwupd --verbose` and it will take care of the necessary downloads.