fwupd/fwupd #5658: How is ME Mfg mode checked with multiple physical MEI devices?

mefff, 2023-03-29:
I found two Windows Surface laptops that have 2 MEI devices, with different PCI config space, which I then assume correspond to two different physical devices. One on bdf 0.22.0 (DID 0x9d3a) and the other one on 0.22.4 (DID 0x9d3e).

Digging into an ME datasheet I found that it mentions that the HFS registers are only defined for the bdf 0.22.0 device. The problem is that the kernel is not checking, they are only checking for the device ID, hence I think that the pci_mei plugin would do the same, resulting in a false positive mfg mode active.

I couldn't debug it further, and fwupdtool showed the mfg mode disabled, so I can't confirm it. Could it be that the security attribute is being written for every mei device, and here the last one was the system one (no mfg mode)?

Regarding: 
https://github.com/fwupd/fwupd/blob/3d7d5dca673ae63633acc1f83d5f27e70d4d42fc/plugins/pci-mei/fu-pci-mei-plugin.c#L217

On this case, the `mei0` was the 0.22.4 device, so I think we should read the first directory here, instead of hardcoding the mei0. This device had a bunch of 0's as the versions, I'm even questioning if the kernel should dump that, or maybe is a vendor issue.

hughsie, 2023-03-30:
Hmm, so I tried that on my server with the ARC GPU -- i.e. mei_gsc and lots of UUIDs for each physical MEI controller. I only got one device matching against `PCI\DRIVER_mei_me`. I think the better way to fix this is to actually use the ME interface to read the HFSTSx registers -- rather than the config space.

hughsie, 2025-05-04:
I now have a server with two mei devices -- and it seems to work fine in 2.0.9.

closed 2025-05-04 by hughsie