fwupd/fwupd #5666: fwupdmgr does not find 8BitDo SN30 pro+ on Ubuntu 22.04.2

Time0o, 2023-03-30:
**Describe the bug**

I am unable to update the firmware or even recognize an otherwise functioning 8BitDo SN30 pro+ controller on Ubuntu 22.04.2.

**Steps to Reproduce**

Put the 8BitDo into Android pairing mode by pressing start + B and then connect it to a computer running Ubuntu 22.04.2 via USB, it shows up in `lsusb` but `fwupdmgr get-devices | grep 8Bit` outputs nothing.

**fwupd version information**

```shell
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.19.0-38-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10

```
Installed via `apt`.

<summary>**fwupd device information**</summary>

```shell
21CBCTO1WW
│
├─ELAN067B:00 04F3:31F8:
│     Device ID:          4dc958f59f7f61d6e4fee930e995b76baf216980
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0003
│     Vendor:             HIDRAW:0x04F3
│     GUIDs:              175e14ec-80c6-5c7e-beb3-2e2b24a16e3e
│                         abe7b9d9-0449-51d0-b89e-49f4b54780b2
│                         93386a84-d3f7-5ecc-bf58-7d4e14dc2a45
│                         4b808d27-c833-50cc-8177-9a401f5c08fe
│                         4070ab62-370f-529a-80fc-6e60d076e04b
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Embedded Controller:
│     Device ID:          ae1abd099407b1d95698d69b7273f7fed5c6f35c
│     Summary:            UEFI ESRT device
│     Current version:    0.1.17
│     Minimum Version:    0.1.17
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               ec01fae4-c67a-42b4-bada-a7c1b9900897
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          1bcca71b3480ec5662e5da8a742c4c805275ffbf
│     Current version:    54.20
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     GUIDs:              cdfbfdf3-2ee8-5c50-9c4a-7917691bc277
│                         2dccef5c-4211-5379-9463-a215f5968ef6
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    1.25.1932
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
[... 238 lines cut ...]
│     Summary:            UEFI ESRT device
│     Current version:    2097684
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               88523ddf-3aac-41f6-b15f-dc7bea434b39
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
      Summary:            UEFI ESRT device
      Current version:    3
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               240cf38f-3985-42c5-b77b-8e4bdaf0fcc1
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
```

lorn10, 2023-09-04:
By the way, I can confirm this problem also for the 8BitDo **Pro 2** gamepad. It is also not detected in Kubuntu 22.04 LTS.

And yes the underlying problem seems to be the change of the 8BitDo firmware handling / protocol mentioned in #1681.

hughsie, 2023-09-23:
8bitdo doesn't want to support this hardware unfortunately.

closed 2023-09-23 by hughsie