fwupd/fwupd #5668: Dell: Unable to update XPS 13 (9310) from 3.12.1 to 3.13.1

dguglielmi, 2023-03-30:
**Describe the bug**
Update from 3.12.1 to 3.13.1 freeze the system at [Decompressing].... stage on Gentoo Linux

**Steps to Reproduce**
1 - fwupmgr update

Devices with no available firmware updates: 
 • TPM 2.0
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19TB
 • KXG60PNV2T04 NVMe KIOXIA 2048GB
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 3.12.1 to 3.13.1?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Improved the stability of the system.                                      ║
║                                                                              ║
║ XPS 13 9310 must remain plugged into a power source for the duration of the  ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y

Freeze the entire system second after, need to hard reboot the system

**Expected behavior**
Update to 3.13.1

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.13
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.1.19-gentoo
runtime   org.freedesktop.fwupd         1.8.13

```

Installed by portage and I try master version compiled from sources

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Dell Inc. XPS 13 9310
│
├─WD19TB:
│ │   Device ID:          570867f4ddda9915445da59bd85cd0e0e507270e
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.03
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
[... 302 lines cut ...]
│           ├── mmx64.efi
│           └── shimx64.efi
├── grub
│   ├── grub.cfg
│   └── grubenv
├── initramfs-6.1.12-gentoo.img
├── initramfs-6.1.19-gentoo.img
├── lost+found
├── System.map-6.1.12-gentoo
├── System.map-6.1.19-gentoo
├── vmlinuz-6.1.12-gentoo
└── vmlinuz-6.1.19-gentoo

11 directories, 19 files

```

**Additional questions**

- Operating system and version: Gentoo Linux
- Have you tried rebooting? Yes, multiple time
- Is this a regression? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2023-03-31:
A dufe of https://github.com/fwupd/fwupd/issues/5656 perhaps?

dguglielmi, 2023-03-31:
I don't know if it's related to that issue specifically but the interesting part is that if I use 1.8.13, the fwupd service crash with:

`
Mar 30 22:29:25 xps13-9310 systemd[1]: Started fwupd.service.
Mar 30 22:29:38 xps13-9310 fwupd[2968]: double free or corruption (fasttop)
Mar 30 22:29:38 xps13-9310 systemd[1]: fwupd.service: Main process exited, code=dumped, status=6/ABRT
Mar 30 22:29:38 xps13-9310 systemd[1]: fwupd.service: Failed with result 'core-dump'.
`
but the system don't freeze at all

If I use my own compiled version from master I got:

`
fwupdmgr update 
Devices with no available firmware updates: 
 • TPM 2.0
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • KXG60PNV2T04 NVMe KIOXIA 2048GB
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 3.12.1 to 3.13.1?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Improved the stability of the system.                                      ║
║                                                                              ║
║ XPS 13 9310 must remain plugged into a power source for the duration of the  ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Decompressing…           [                                       ]
`
Then the system is completely freeze (local session + ssh session). Nothing in the journal.

hughsie, 2023-03-31:
I've just released 1.8.14 -- can you try that please.

dguglielmi, 2023-03-31:
Same issue with 1.8.14. I upgraded libjcat to 0.1.13, libxmlb to 0.3.11. Give a  try with SB disabled + "Allow firmware downgrade" in UEFI settings 
```
fwupdmgr update --version
compile   org.freedesktop.fwupd         1.8.14
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.1.19-gentoo
runtime   org.freedesktop.fwupd         1.8.14
```

The host freeze at the same stage with 1.8.14

```
 fwupdmgr update --verbose
(fwupdmgr:8471): GLib-DEBUG: 13:27:33.746: setenv()/putenv() are not thread-safe and should not be used after threads are created
(pkttyagent:8473): GLib-GIO-DEBUG: 13:27:33.748: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:8471): GLib-GIO-DEBUG: 13:27:33.758: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.758: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.759: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.759: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.759: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.760: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.760: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.760: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.760: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.760: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.760: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.760: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.760: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.760: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.760: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:8471): dconf-DEBUG: 13:27:33.760: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:8471): GLib-GIO-DEBUG: 13:27:33.767: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:8471): GLib-GIO-DEBUG: 13:27:33.768: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:8471): Fwupd-DEBUG: 13:27:33.819: Emitting ::status-changed() [idle]
Devices with no available firmware updates: 
 • TPM 2.0
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • KXG60PNV2T04 NVMe KIOXIA 2048GB
(fwupdmgr:8471): FuMain-DEBUG: 13:27:33.832: current version is 10704104: 10704104=same
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 3.12.1 to 3.13.1?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Improved the stability of the system.                                      ║
║                                                                              ║
║ XPS 13 9310 must remain plugged into a power source for the duration of the  ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Fwupd-INFO: 13:27:37.239: downloading https://fwupd.org/downloads/257e955cb1dc489d25f56e73645edd07774b1fc1f904fa13096b19a721d9e1b4-Dell_XPS_13_9310_U_System_BIOS_V3.13.1.cab
(fwupdmgr:8471): Fwupd-DEBUG: 13:27:37.242: Emitting ::status-changed() [downloading]
Downloading…             [                   -                   ]Fwupd-INFO: 13:27:37.999: download progress: 100%
Downloading…             [***************************************]Fwupd-INFO: 13:27:38.306: download progress: 0%
Downloading…             [  -                                    ]Fwupd-INFO: 13:27:38.381: download progress: 1%
Downloading…             [                                       ]Fwupd-INFO: 13:27:38.394: download progress: 2%
Downloading…             [                                       ]Fwupd-INFO: 13:27:38.399: download progress: 3%
Downloading…             [*                                      ]Fwupd-INFO: 13:27:38.405: download progress: 4%
Downloading…             [*                                      ]Fwupd-INFO: 13:27:38.409: download progress: 5%
Downloading…             [*                                      ]Fwupd-INFO: 13:27:38.413: download progress: 6%
[... cut ...]
Restarting device…       [                                       ](fwupdmgr:8471): Fwupd-DEBUG: 13:27:39.374: Emitting ::status-changed() [device-write]
Writing…                 [                                       ](fwupdmgr:8471): Fwupd-DEBUG: 13:27:39.374: Emitting ::status-changed() [scheduling]
Scheduling…              [                                       ](fwupdmgr:8471): Fwupd-DEBUG: 13:27:39.374: Emitting ::status-changed() [decompressing]
Decompressing…           [                                       ]
```

Unable to see/get kernel errors messages after the freeze.

dguglielmi, 2023-04-09:
I've made some progress, and it's not specifically related to the Dell firmware update. I manually updated it from the Dell UEFI interface and I still have the same behavior (system freeze during decompression) with the Secure Boot upgrade dbx 217 -> 220

dguglielmi, 2023-05-13:
Ok, I finally got It. The freezes on my system were visibly related to efivars files in /sys/firmware/efi/efivars/. I was able to reproduce a freeze by simply trying to add and/or modify boot entries order with efibootmgr.

Problem solved by purging /sys/firmware/efi/efivars/dump-type0* files.

closed 2023-05-13 by dguglielmi

hughsie, 2023-05-15:
Do you know what created the /sys/firmware/efi/efivars/dump-type0 files?

superm1, 2023-05-15:
Doesn't the kernel directly create those types of files when you have `CONFIG_EFI_PSTORE` set? The expectation is that they'll be cleared by userspace.

That's what systemd should have been doing; moving them to `/var/lib/systemd/pstore`

dguglielmi, 2023-05-15:
Sadly I deleted the files without making a copy. There were about 20-30 (dump-type0-xxxx) files.

My kernel config for PSTORE is:

```
# grep PSTORE /boot/config-6.1.24-gentoo 
CONFIG_EFI_VARS_PSTORE=m
# CONFIG_EFI_VARS_PSTORE_DEFAULT_DISABLE is not set
# CONFIG_MTD_PSTORE is not set
CONFIG_CHROMEOS_PSTORE=m
CONFIG_PSTORE=y
CONFIG_PSTORE_DEFAULT_KMSG_BYTES=10240
CONFIG_PSTORE_DEFLATE_COMPRESS=y
# CONFIG_PSTORE_LZO_COMPRESS is not set
# CONFIG_PSTORE_LZ4_COMPRESS is not set
# CONFIG_PSTORE_LZ4HC_COMPRESS is not set
# CONFIG_PSTORE_842_COMPRESS is not set
# CONFIG_PSTORE_ZSTD_COMPRESS is not set
CONFIG_PSTORE_COMPRESS=y
CONFIG_PSTORE_DEFLATE_COMPRESS_DEFAULT=y
CONFIG_PSTORE_COMPRESS_DEFAULT="deflate"
# CONFIG_PSTORE_CONSOLE is not set
# CONFIG_PSTORE_PMSG is not set
# CONFIG_PSTORE_FTRACE is not set
CONFIG_PSTORE_RAM=m
CONFIG_PSTORE_ZONE=m
CONFIG_PSTORE_BLK=m
CONFIG_PSTORE_BLK_BLKDEV=""
CONFIG_PSTORE_BLK_KMSG_SIZE=64
CONFIG_PSTORE_BLK_MAX_REASON=2
```

If interested, I will take a look if that kind of files reappear and try to understand why and by what they are created.

superm1, 2023-05-15:
Do you use systemd? Or something different?

dguglielmi, 2023-05-15:
Yes, I use systemd.  /var/lib/systemd/pstore is empty. I don't remember disabling systemd-pstore unit but:

```
# systemctl status systemd-pstore.service
○ systemd-pstore.service - Platform Persistent Storage Archival
     Loaded: loaded (/usr/lib/systemd/system/systemd-pstore.service; disabled; preset: enabled)
     Active: inactive (dead)
       Docs: man:systemd-pstore(8)
```

Would you recommend enable the unit ?

superm1, 2023-05-15:
Can you look at the logs for that unit? I think it just runs on boot.

dguglielmi, 2023-05-15:
The logs are empty for now:

```
# journalctl -u systemd-pstore
-- No entries --
```

If I start the unit manually I got:

```
# systemctl start systemd-pstore
# journalctl -u systemd-pstore
May 15 23:28:23 xps13-9310 systemd[1]: systemd-pstore.service was skipped because of an unmet condition check (ConditionDirectoryNotEmpty=/sys/fs/pstore).
```