fwupd/fwupd #5675: AMD Ryzen 3600 supported, obsoleted

loveisfoss, 2023-04-01:
**Describe the bug**
Device Security in Gnome reports "Checks Unavailable". Secure Boot is active.

```
$ fwupdmgr security --force
Host Security ID: HSI:INVALID:missing-data

HSI-1
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ Supported CPU:                 Failed

HSI-2
✔ IOMMU:                         Enabled
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ fwupd plugins:                 Untainted

This system has a low HSI security level.
 » https://fwupd.github.io/hsi.html#low-security-level

Host Security Events
  2023-03-25 18:35:20:  ✔ Kernel lockdown enabled
  2023-03-25 18:35:20:  ✔ Secure Boot enabled

...

{
                               "AppstreamId" : "org.fwupd.hsi.SupportedCpu",
                               "Created" : 1680346962,
                               "HsiLevel" : 1,
                               "HsiResult" : "valid",
                               "Name" : "Supported CPU",
                               "Summary" : "Processor Security Checks",
                               "Description" : "Each system should have tests to ensure firmware security.",
                               "Plugin" : "cpu",
                               "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SupportedCpu",
                               "Flags" : [
                                 "success",
                                 "obsoleted"
                               ],
                               "Guid" : [
                                 "b9a2dd81-159e-5537-a7db-e7101d164d3f",
                                 "997104da-ec11-5e04-9d94-ac5501fbf609",
                                 "b54f3c89-8a41-5b9f-9a5c-0367c19a9451",
                                 "d951a23c-a920-50ac-ad89-0660aa0be95c"
                               ]
                             },

...
```

**Expected behavior**
It would be nice to have more information on why the cpu is reported as supported but obsoleted.
Is there a list of obsoleted cpu's? 
Is there a wrong setting or something missing that falsely marks the cpu as obsoleted?
[... 211 lines cut ...]
│                         • Only version upgrades are allowed
│                         • Signed Payload
│   
└─WDC WD6402AAEX-00Z3A0:
      Device ID:          935d7cdf33f62258a740ff7b7bc500fdb6e73f01
      Summary:            ATA drive
      Current version:    05.01D05
      Vendor:             Western Digital (ATA:0x101C, OUI:0014ee)
      Serial Number:      WD-WCATR1695834
      GUIDs:              ed81a091-d452-56ac-896c-428ee3a27a8e ← IDE\WDC_WD6402AAEX-00Z3A0___________________05.01D05
                          958b31ca-5152-55eb-a49b-fca891f887c1 ← IDE\0WDC_WD6402AAEX-00Z3A0___________________
                          3b32a6dc-57e4-5639-989d-bbda276a9ca7 ← WDC WD6402AAEX-00Z3A0
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```
</details>

**Additional questions**

- Operating system and version: Fedora 38.
- Have you tried rebooting? Yes.
- Is this a regression? No.

loveisfoss, 2023-04-01:
Okay, after some reading here and there, I have understood that this might have to do with the motherboard bios not allowing access for the linux kernel/ccp to some specific functions.

#5415 seems like it is a similar case.

Could anybody confirm this?

superm1, 2023-04-01:
https://fwupd.github.io/libfwupdplugin/hsi.html#not-enough-info

Your firmware and SOC don't provide enough information to generate a report.

You can ask your system manufacturer to add support for this.

loveisfoss, 2023-04-02:
> https://fwupd.github.io/libfwupdplugin/hsi.html#not-enough-info
> 
> Your firmware and SOC don't provide enough information to generate a report.
> 
> You can ask your system manufacturer to add support for this.

Ok, thanks for the confirmation.

The question now is, what am I going to ask them to add support for?
Is it "support for the ccp linux driver" I am asking?

superm1, 2023-04-02:
Support for firmware attributes reporting.

https://github.com/torvalds/linux/blob/master/Documentation/ABI/testing/sysfs-driver-ccp

loveisfoss, 2023-04-03:
> Support for firmware attributes reporting.
> 
> https://github.com/torvalds/linux/blob/master/Documentation/ABI/testing/sysfs-driver-ccp

Unfortunately, this is the answer I got from MSI. :(

Regarding your concern, we have checked the BIOS code, there is no option to enable or Disable it. According to the link you sent, it may be AMD Code related issue, we suggest you try later BIOS with newer AMD code. Sorry for any inconvenience caused.

superm1, 2023-04-03:
That's unfortunate; but there is nothing else we can do from fwupd side.

closed 2023-04-03 by superm1

superm1, 2023-04-03:
JFYI - from kernel side I expect this fixes the message about a broken BIOS.
https://lore.kernel.org/linux-crypto/56d92baa-d05a-9b02-0195-7627187fdde1@amd.com/T/#t

loveisfoss, 2023-04-04:
> JFYI - from kernel side I expect this fixes the message about a broken BIOS. https://lore.kernel.org/linux-crypto/56d92baa-d05a-9b02-0195-7627187fdde1@amd.com/T/#t

So, if CCP would stop emitting this error, do you think that fwupd would consider the cpu to be supported?

superm1, 2023-04-04:
> > JFYI - from kernel side I expect this fixes the message about a broken BIOS. https://lore.kernel.org/linux-crypto/56d92baa-d05a-9b02-0195-7627187fdde1@amd.com/T/#t
> 
> So, if ccp  would stop emitting this error, do you think that fwupd would consider the cpu to be supported?

Nope. The error was just noise because a feature in ccp.ko didn't work on certain client processors.

The exporting information about security of the machine is a feature that the underlying hardware and firmware must provide.

loveisfoss, 2023-04-05:
Okay, but if I understand correctly, it is the ccp that gathers this information and shares it to other parts, like fwupd, right? Or, does fwupd access this information by itself?

superm1, 2023-04-05:
The confusing part is the driver is called "ccp" but it accesses multiple pieces of IP that are both accessible "through" the PCIe BAR for the Security Processor (SP) endpoint.

The CCP "IP" it accesses is for cryptographic operations on supported processors. This is the subset that emits the message you saw.

The other "IP" allows access into "TEE", "SEV". There are similar messages for those parts of the driver that can be emitted.

The part fwupd cares about is actually none of these. The same register that exports capabilities for TEE and SEV also has a bit to indicate the platform can export security attributes with very specific meanings. When that bit is set then all the attributes fwupd cares about are exported. Since it's not set then fwupd isn't able to measure them.

Hope that clarifies it.

loveisfoss, 2023-04-06:
Alright, thanks for the clarification and for your time!

superm1, 2024-05-29:
Here is a [patch series](https://lore.kernel.org/linux-crypto/oCZE3H8ZnqlQsatOdb4HH7_0t7E9fsRqVv1QodA3IeUSDCdx9DP1bWsLOVq6T5iGffMjzMequirUJSM4UpcA1y2LZVUvMsffulaArsPy2cs=@hughsie.com/T/#t) for the kernel that enables the feature on the older SoCs.