fwupd/fwupd #5681: firmware signature missing error for user signed GPG/PKCS#7 firmware

maheshbabu022, 2023-04-04:
Tried signing the .bin/.ROM with a self created key pair(PGP) and created a .cab file containing firmware.bin, firmware.bin.asc/.signed and firmware.metainfo.xml.
> gcab -c -v firmware.cab firmware.bin firmware.bin.asc firmware.metainfo.xml

Placed the public key under /etc/pki/fwupd/ and ran a install command in the newly created cab file.  
>fwupdmgr -v install firmware.cab

Getting the below error for signature missing or not trusted 
root@edge-Edge-Gateway-5200:/home/edge/mahesh/certtest# fwupdmgr -v install firmware.cab
(fwupdmgr:1467795): GLib-DEBUG: 23:52:14.371: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:1467795): GLib-GIO-DEBUG: 23:52:14.378: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.378: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.378: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.378: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.378: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.378: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.379: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.379: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.379: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.379: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.379: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.379: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.379: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.379: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.379: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:1467795): dconf-DEBUG: 23:52:14.379: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:1467795): GLib-GIO-DEBUG: 23:52:14.380: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:1467795): Fwupd-DEBUG: 23:52:14.382: Emitting ::status-changed() [idle]
(fwupdmgr:1467795): Fwupd-DEBUG: 23:52:14.394: Emitting ::status-changed() [decompressing]
Decompressing…           [  -                                    ](fwupdmgr:1467795): Fwupd-DEBUG: 23:52:14.455: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer

Is there any missing piece in the workflow?

superm1, 2023-04-04:
Did you restart the daemon after adding your cert?

hughsie, 2023-04-04:
Does https://github.com/fwupd/fwupd/blob/main/docs/only-trusted.md help?

maheshbabu022, 2023-04-04:
> Does https://github.com/fwupd/fwupd/blob/main/docs/only-trusted.md help?

This is good info for creating your own cab file. I Q here, is there a way to create the cab without creating .jcat file? is it a must?

hughsie, 2023-04-04:
I think we do support the asc detached signature too (without a jcat file), but you still need to add the GPG signing keys in /etc/pki/fwupd/ rather than being in your local keyring.

maheshbabu022, 2023-04-04:
> 

Got it. I got my firmware.bin signed from GPG signing infra and placed and added the keys in /etc/pki/fwupd and created a cab containing - .bin, .asc and .metadata. But still it was hitting the signature issue
Any missing piece here?

hughsie, 2023-04-04:
I think `sudo fwupdtool get-details foo.cab -vv` might help, e.g. I see:

	19:44:28.340 Jcat                 Using engine at /var/lib/fwupd/gnupg
	19:44:28.345 Jcat                 ignoring LVFS-CA.pem as not GPG public key
	19:44:28.345 Jcat                 ignoring ACME-CA.pem as not GPG public key
	19:44:28.346 Jcat                 Adding GnuPG public key /etc/pki/fwupd/GPG-KEY-Linux-Vendor-Firmware-Service
	19:44:28.353 Jcat                 importing key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2 [0] Success
	19:44:28.353 Jcat                 Adding GnuPG public key /etc/pki/fwupd/GPG-KEY-Linux-Foundation-Firmware
	19:44:28.358 Jcat                 importing key F2F0325648E1AE4956710198C6361787A0A849E1 [0] Success
	19:44:28.358 Jcat                 Adding GnuPG public key /etc/pki/fwupd-metadata/GPG-KEY-Linux-Foundation-Metadata
	19:44:28.362 Jcat                 importing key CAE0420259890141E701156E0A6DCEF6B46F3DBF [0] Success
	19:44:28.362 Jcat                 ignoring LVFS-CA.pem as not GPG public key
	19:44:28.362 Jcat                 Adding GnuPG public key /etc/pki/fwupd-metadata/GPG-KEY-Linux-Vendor-Firmware-Service
	19:44:28.367 Jcat                 importing key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2 [0] Success
	19:44:28.371 Jcat                 returned signature fingerprint 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
	19:44:28.371 Jcat                 verified: JcatResult:
	  Timestamp:             2022-12-20T18:17:58Z
	  Authority:             3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
	  JcatGpgEngine:
	    Kind:                gpg
	    VerifyKind:          signature

	19:44:28.371 Jcat                 got 1 PKCS7 signatures
	19:44:28.371 Jcat                 verified: JcatResult:
	  Timestamp:             2022-12-20T18:17:58Z
	  Authority:             O=Linux Vendor Firmware Project,CN=LVFS CA
	  JcatPkcs7Engine:
	    Kind:                pkcs7
	    VerifyKind:          signature

	19:44:28.371 FuCabinet            verified metadata firmware.metainfo.xml: 3
	19:44:28.371 FuCabinet            processing file: firmware.metainfo.xml
	19:44:28.371 FuCabinet            processing release: 3.0.1
	19:44:28.372 Jcat                 Jcat engine kind 'sha1' not allowed
	19:44:28.372 Jcat                 Jcat engine kind 'sha1' not allowed
	19:44:28.376 Jcat                 returned signature fingerprint 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
	19:44:28.376 Jcat                 verified: JcatResult:
	  Timestamp:             2022-12-20T18:17:58Z
	  Authority:             3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
	  JcatGpgEngine:
	    Kind:                gpg
	    VerifyKind:          signature

	19:44:28.376 Jcat                 got 1 PKCS7 signatures
	19:44:28.376 Jcat                 verified: JcatResult:
	  Timestamp:             2022-12-20T18:17:58Z
	  Authority:             O=Linux Vendor Firmware Project,CN=LVFS CA
	  JcatPkcs7Engine:
	    Kind:                pkcs7
	    VerifyKind:          signature

	19:44:28.376 FuCabinet            verified payload firmware.bin: 3

maheshbabu022, 2023-04-05:
Tried 

> Does https://github.com/fwupd/fwupd/blob/main/docs/only-trusted.md help?

Tried the steps mentioned in the above link and still see the signature issue

 fwupdmgr -v install firmware.cab --allow-reinstall
(fwupdmgr:1472366): GLib-DEBUG: 14:49:00.053: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:1472366): GLib-GIO-DEBUG: 14:49:00.059: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.059: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.060: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.060: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.060: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.060: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.060: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.060: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.060: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.060: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.060: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.060: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.060: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.060: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.061: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:1472366): dconf-DEBUG: 14:49:00.061: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:1472366): GLib-GIO-DEBUG: 14:49:00.061: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:1472366): Fwupd-DEBUG: 14:49:00.063: Emitting ::status-changed() [idle]
(fwupdmgr:1472366): Fwupd-DEBUG: 14:49:00.069: Emitting ::status-changed() [decompressing]
Decompressing…           [   -                                   ](fwupdmgr:1472366): Fwupd-DEBUG: 14:49:00.165: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
**firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer**

root@edge-Edge-Gateway-5200:/home/edge/mahesh/certlab# ls -la /etc/pki/fwupd
total 80
drwxr-xr-x 3 root root  4096 Apr  5 14:48 .
drwxr-xr-x 4 root root  4096 Apr  2 22:46 ..
-rw-r--r-- 1 root root  4250 Mar 29 12:48 0x1285491434D8786F
-rw-r--r-- 1 root root  3110 Mar 29 12:48 0x274E9C32857A9594
-rw-r--r-- 1 root root  1079 Mar 29 12:48 0x3CA66B4946770C59
-rw-r--r-- 1 root root  3206 Mar 29 12:48 0x756ba70b1019ced6
-rw-r--r-- 1 root root  6111 Mar 29 12:48 0xca77951d23b66a9d
-rw-r--r-- 1 root root  1635 Apr  5 14:29 ACME-CA.pem
-rw-r--r-- 1 root root  1046 Apr  2 20:11 fwupdtest.pem
-rw-r--r-- 1 root root  2169 Feb  7  2022 GPG-KEY-Linux-Foundation-Firmware
-rw-r--r-- 1 root root   959 Feb  7  2022 GPG-KEY-Linux-Vendor-Firmware-Service
drwxr-xr-x 3 root root  4096 Mar 29 12:46 keys
-rw-r--r-- 1 root root  1679 Feb  7  2022 LVFS-CA.pem
-rw-r--r-- 1 root root 13192 Mar 29 12:45 pgp_pub_keys.tgz
-rw-r--r-- 1 root root  1204 Apr  5 14:48 rhughes.pem

hughsie, 2023-04-05:
What does `sudo fwupdtool get-details foo.cab -vv` say?

maheshbabu022, 2023-04-05:
[fwupd_update_output.txt](https://github.com/fwupd/fwupd/files/11158330/fwupd_update_output.txt)

maheshbabu022, 2023-04-05:
got the issue- there were other pieces of pem file, which was getting linked and i happened to take that off and kept only ACME-CA.pem. It worked.

closed 2023-04-05 by hughsie

maheshbabu022, 2023-04-05:
> got the issue- there were other pieces of pem file, which was getting linked and i happened to take that off and kept only ACME-CA.pem. It worked.

Thank You for the support

maheshbaburamaiah, 2023-07-27:
The above issue is resurfacing on a different system, even after keeping ACME-CA.pem alone.

maheshbabu022, 2023-07-27:
@hughsie , The above issue is resurfacing in a different system, though i happened to isolate ACME-CA.pem for verification. Any missing pieces here?

hughsie, 2023-07-27:
> Any missing pieces here?

I don't have any additional debugging from you so I don't have anything at all to start with. I'm also wondering what you're trying to do...

maheshbabu022, 2023-07-27:
Im trying to create my own cab file with the keys that i have. Any particular logs that you would like to have?

hughsie, 2023-07-27:
`sudo fwupdtool get-details foo.cab -vv`

maheshbabu022, 2023-07-27:
The command helped me figure out somethings on my own and fix it as well. Thank you.
I did try something beyond something like this - 
I created the PKCS#7 signature  from jcat and copied it to a new file as to create firmware.bin.asc and eventually created a bundle containing only these - [firmware.bin. firmware.bin.asc(pkcs7 sign) and firmware.metainfo.xml]. 
This when i run through - sudo fwupdtool get-details foo.cab -vv , i get the below the error.
So Question- Do we only support PKCS#7 only in jcat and not as a standalone file format?

[image]

Incase you need the complete dump let me know.

hughsie, 2023-07-27:
> Do we only support PKCS#7 only in jcat and not as a standalone file format

I think we dropped support for the standalone pkcs format when we moved to JCat -- falling back only to .asc. Why not use the .jcat file i the .cab file?

maheshbabu022, 2023-07-27:
Our signing infra has some limitations in using jcat-tool. Looks like we need to generate the signature and import it using import command. Hope that should help.

hughsie, 2023-07-27:
> Our signing infra

That bit worries me a bit. Why are you auto-signing firmware? We're never going to ship vendor keys as part of fwupd.

maheshbabu022, 2023-08-02:
Nope this is more of a experiment before we get this done through signing procedure.
I have another 

> > Do we only support PKCS#7 only in jcat and not as a standalone file format
> 
> I think we dropped support for the standalone pkcs format when we moved to JCat -- falling back only to .asc. Why not use the .jcat file i the .cab file?

@hughsie ,  is there an option to skip verification of "firmware.metainfo.xml" as part .cab verification. I see skipping works automatically when we have a bundle that contains only GPG signature(and no jcat file). Internally we are not very keen of verifying the firmware.metainfo.xml, but only the firmware.bin.

maheshbabu022, 2023-08-02:
> > Our signing infra
> 
> That bit worries me a bit. Why are you auto-signing firmware? We're never going to ship vendor keys as part of fwupd.

Currently what im trying is more of experiments and once through internal process kicks to create and manage keys. So would not be using the experimental keys.

maheshbabu022, 2023-08-03:
Thank you that would be really helpful.
On the other question.... Can you share if its allowed and doable...
@hughsie , is there an option to skip verification of "firmware.metainfo.xml" as part .cab verification. I see skipping works automatically when we have a bundle that contains only GPG signature(and no jcat file). Internally we are not very keen of verifying the firmware.metainfo.xml, but only the firmware.bin.

hughsie, 2023-08-03:
Sure, follow the instructions on https://lvfs.readthedocs.io/en/latest/apply.html and I'll create an account for you on Wednesday when I'm back from holiday. The metadata signing should be optional iirc.