fwupd/fwupd #5687: Aiaiai H05 Firmware update (command line) fails; "Message recipient disconnected from message bus without replying"

g0-n, 2023-04-05:
I think I better open this new issue as we were discussing it over an already closed issue (#5251)

fwupd version:
> runtime   org.freedesktop.fwupd         1.8.6
> runtime   org.freedesktop.fwupd-efi     1.3
> compile   org.freedesktop.gusb          0.4.0
> runtime   com.hughsie.libjcat           0.1.11
> compile   com.hughsie.libjcat           0.1.11
> runtime   com.dell.libsmbios            2.4
> runtime   org.kernel                    5.13.0-valve36-1-neptune
> compile   org.freedesktop.fwupd         1.8.6
> runtime   org.freedesktop.gusb          0.4.2

Thank you for your patience.

g0-n, 2023-04-06:
Ok, I don't know if I did it right, gdb debugging kind of freeze, and when finally finish `bt` answered "no stack"...
Anyway, in case it helps I paste this in here:

> deck@g0nDeck ~> sudo gdb --args /usr/lib/fwupd/fwupd -vv
> [sudo] password for deck:
> GNU gdb (GDB) 12.1
> Copyright (C) 2022 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.
> Type "show copying" and "show warranty" for details.
> This GDB was configured as "x86_64-pc-linux-gnu".
> Type "show configuration" for configuration details.
> For bug reporting instructions, please see:
> <https://www.gnu.org/software/gdb/bugs/>.
> Find the GDB manual and other documentation resources online at:
>     <http://www.gnu.org/software/gdb/documentation/>.
> 
> For help, type "help".
> Type "apropos word" to search for commands related to "word"...
> Reading symbols from /usr/lib/fwupd/fwupd...
> (No debugging symbols found in /usr/lib/fwupd/fwupd)
> (gdb) r
> Starting program: /usr/lib/fwupd/fwupd -vv
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/usr/lib/libthread_db.so.1".
> 12:43:28:0257 FuDebug              Verbose debugging enabled (on console 1)
> 12:43:28:0275 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
> 12:43:28:0275 FuCommon             reading /etc/fwupd/daemon.conf with 2224 bytes
> 12:43:28:0275 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
> 12:43:28:0275 FuConfig             using autodetected max archive size 3.9 GB
> [New Thread 0x7ffff5a946c0 (LWP 6587)]
> 12:43:28:0276 FuRemoteList         loading /usr/share/fwupd/metainfo
> 12:43:28:0277 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
> 12:43:28:0277 XbSilo               file: 51023696-994a-dc3a-ba77-ec6cdbf7a312, current:51023696-994a-dc3a-ba77-ec6cdbf7a312, cached: (null)
> 12:43:28:0277 XbSilo               loading silo with file contents
> 12:43:28:0277 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
> 12:43:28:0277 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
> 12:43:28:0277 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
> 12:43:28:0277 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
> 12:43:28:0278 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
> 12:43:28:0278 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
> 12:43:28:0278 FuRemoteList         ordering lvfs-testing=lvfs+1
> 12:43:28:0278 FuRemoteList         ignoring unfound remote fwupd
> 12:43:28:0278 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
> 12:43:28:0278 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
> 12:43:28:0278 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
> 12:43:28:0278 Jcat                 loaded 1 certificates
> 12:43:28:0278 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
> 12:43:28:0278 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
> 12:43:28:0278 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
> 12:43:28:0278 Jcat                 loaded 1 certificates
> 12:43:28:0278 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
> 12:43:28:0278 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
> 12:43:28:0278 FuEngine             client certificate now exists: nothing to do
> 12:43:28:0278 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
> 12:43:28:0279 FuHistory            got schema version of 8
> 12:43:28:0321 FuPlugin             load(/usr/lib/fwupd-1.8.6/libfu_plugin_modem_manager.so)
> 12:43:28:0322 FuContext            added udev subsystem watch of platform
> 12:43:28:0322 FuContext            added udev subsystem watch of mei
> 12:43:28:0322 FuContext            added udev subsystem watch of block
> 12:43:28:0322 FuContext            added udev subsystem watch of pci
> 12:43:28:0322 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
> 12:43:28:0322 FuPluginDell         Using libsmbios 2.4
> 12:43:28:0322 FuContext            added udev subsystem watch of i2c
> 12:43:28:0322 FuContext            added udev subsystem watch of i2c-dev
> 12:43:28:0323 FuContext            added udev subsystem watch of hidraw
> 12:43:28:0323 FuContext            added udev subsystem watch of gpio
> 12:43:28:0323 FuContext            added udev subsystem watch of iommu
> 12:43:28:0323 FuContext            added udev subsystem watch of msr
[... 257 lines cut ...]
> 12:43:29:0295 FuEngine             requirement 1.0.3 ge 1.8.6 -> org.freedesktop.fwupd passed
> 12:43:29:0295 FuEngine             requirement 1.0.3 ge 1.8.6 -> org.freedesktop.fwupd passed
> 12:43:29:0296 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
> 12:43:29:0304 FuProgress           raw timing data was { 0.264, 0.001, 0.022, 0.001, 0.030, 0.003 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 82 0 7 0 9 1 ] at ../fwupd-1.8.6/plugins/uefi-capsule/fu-uefi-capsule-p [... cut ...]
> 14:43:29:0483 FuPlugin             finalize(modem_manager)
> 14:43:29:0484 FuPlugin             finalize(flashrom)
> 14:43:29:0484 FuPluginTest         destroy
> [Thread 0x7fffeeffd6c0 (LWP 6591) exited]
> [Thread 0x7fffeffff6c0 (LWP 6589) exited]
> [Thread 0x7ffff5a946c0 (LWP 6587) exited]
> [Thread 0x7ffff5d7ea00 (LWP 6584) exited]
> [Thread 0x7ffff481a6c0 (LWP 6588) exited]
> [New process 6584]
> [Inferior 1 (process 6584) exited normally]
> (gdb) bt
> No stack.
> (gdb) q
> deck@g0nDeck ~>

hughsie, 2023-04-07:
Exactly as you did with the Daemon there, but in another terminal tab then schedule the update and hopefully the Daemon then crashes with a backtrace.

g0-n, 2023-04-07:
Finally I got some crash reports!
I realized I could use the Firmware gnome GUI in the steam deck and that made it easier...
I attached the error window and the reports (trying to reinstall 1.6 or downgrading to 1.5). Also attached an error window when I tried to verify the firmware.

[image]
[image]
[firmware crash#1.txt](https://github.com/fwupd/fwupd/files/11179431/firmware.crash.1.txt)
[firmware crash#2.txt](https://github.com/fwupd/fwupd/files/11179433/firmware.crash.2.txt)

hughsie, 2023-04-07:
Okay, certainly getting there. Can you install the debuginfo data for the fwupd package and trigger those crashes again? e.g. https://wiki.archlinux.org/title/Debugging/Getting_traces -- thanks.

g0-n, 2023-04-09:
[fwupd crash#3.txt](https://github.com/fwupd/fwupd/files/11185315/fwupd.crash.3.txt)

Hello! I've been trying to install the data the better I could but I think wasn't successful. 
It seems I don't have that `debuginfod` command or so it's said (and I think that has something to do with the fact I'm doing all this in a Steam Deck)... I tried `coredumpctl gdb` and then inside gdb `bt full` but when I run the debugger again, keeps showing 

> (no debugging symbols found) "/lib/fwupd/fwupd"

I am sorry, I know this is maybe taking a bit too long. It's overwhelming for me...

Thank you very much

hughsie, 2023-04-21:
I think we really need a backtrace to debug this I'm afraid.

closed 2023-09-23 by hughsie