fwupd/fwupd #5689: fwupdmgr security --force reports Linux swap: Unencrypted, while it's encrypted

nvaert1986, 2023-04-06:
**Describe the bug**
fwupdmgr security --force reports Linux swap: Unencrypted, while it's encrypted

**Steps to Reproduce**
Rung fwupdmgr security --force reports and watch for Linux swap: Unencrypted,

**Expected behavior**
Linux swap: Enencrypted,

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.14
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.12
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.1.19-gentoo
runtime   org.freedesktop.fwupd         1.8.14
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

emerge fwupd

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
LENOVO 20M9001FMH
│
├─Cannon Lake LPC Controller:
│ │   Device ID:          71b31258b13a4b2793e529856a190f8fb02ad151
│ │   Current version:    10
│ │   Vendor:             Intel Corporation (PCI:0x8086)
│ │   GUIDs:              42ae6afa-cbfd-5e29-aa1b-943a87d1ec50 ← PCI\VEN_8086&DEV_A30E
│ │                       eda18be2-cab9-515d-b417-6a51da893121 ← PCI\VEN_8086&DEV_A30E&REV_10
│ │                       b3c53281-1520-5fbb-919b-800e2ad00788 ← PCI\VEN_8086&DEV_A30E&SUBSYS_17AA225F
│ │                       411f68e7-797c-5f87-a36a-9a4db27efde8 ← PCI\VEN_8086&DEV_A30E&SUBSYS_17AA225F&REV_10
│ │                       2d80f689-0b5e-5c4b-b6df-bd767f6e9f05 ← INTEL_SPI_CHIPSET\ID_PCH300
│ │   Device Flags:       • Internal device
│ │                       • Cryptographic hash verification is available
│ │ 
│ ├─BIOS:
│ │     Device ID:        ff7dbf2f6e354a5727c6ce1c466230f38bd26ff0
│ │     Vendor:           Intel Corporation (PCI:0x8086)
│ │     GUID:             2c7b8bd2-3e77-5730-aaf6-596136f5254d ← IFD\NAME_BIOS
│ │     Device Flags:     • Internal device
│ │                       • Cryptographic hash verification is available
│ │   
│ ├─Gigabit Ethernet:
│ │     Device ID:        11188287b93230d58f85f059dfab93e1d59724bb
│ │     Vendor:           Intel Corporation (PCI:0x8086)
│ │     GUID:             30ee4fad-d1cd-522f-9a32-e53f4a4269cc ← IFD\NAME_GBE
│ │     Device Flags:     • Internal device
│ │                       • Cryptographic hash verification is available
│ │   
│ └─Intel Management Engine:
│       Device ID:        7ef8a531d2413174034556f12dff8aa3bb4a8c30
│       Vendor:           Intel Corporation (PCI:0x8086)
│       GUID:             614566e7-4d60-55f5-9cb9-48c5e488a705 ← IFD\NAME_ME
│       Device Flags:     • Internal device
│                         • Device is locked
│                         • Cryptographic hash verification is available
│     
[... 252 lines cut ...]
│                         • Signed Payload
│   
└─UEFI Device Firmware:
      Device ID:          f95c9218acd12697af946874bfe4239587209232
      Summary:            UEFI ESRT device
      Current version:    4784132
      Minimum Version:    4784132
      Vendor:             DMI:LENOVO
      Update State:       Success
      Update Error:       BootOrder is locked in firmware setup
      GUID:               22c037d2-4402-452a-aaa6-c0e3f0a1c9ea
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
```

</details>

**Additional questions**

- Operating system and version: Gentoo Linux 2.13
- Have you tried rebooting? Yes
- Is this a regression? No

nvaert1986, 2023-04-06:
Entry in fstab
/dev/mapper/swap                none            swap            sw              0 0

Entry in crypttap
swap    /dev/nvme0n1p2  /dev/urandom    swap,,cipher=aes-cbc-essiv:sha256,size=256

Output of swapon:
NAME      TYPE      SIZE USED PRIO
/dev/dm-0 partition  32G   0B   -2

superm1, 2023-04-06:
duplicate of #4969

closed 2023-04-06 by superm1