fwupd/fwupd #5690: fwupdmgr security --force reports TPM PCR0 reconstruction: Not found while a TPM 2.0 chip is present

nvaert1986, 2023-04-06:
**Describe the bug**
fwupdmgr security --force reports TPM PCR0 reconstruction: Not found while a TPM 2.0 chip is present

**Steps to Reproduce**
Run fwupdmgr security --force

**Expected behavior**
 TPM PCR0 reconstruction: Valid

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.14
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.12
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.1.19-gentoo
runtime   org.freedesktop.fwupd         1.8.14
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
emerge fwupd

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
LENOVO 20M9001FMH
│
├─Cannon Lake LPC Controller:
│ │   Device ID:          71b31258b13a4b2793e529856a190f8fb02ad151
│ │   Current version:    10
│ │   Vendor:             Intel Corporation (PCI:0x8086)
│ │   GUIDs:              42ae6afa-cbfd-5e29-aa1b-943a87d1ec50 ← PCI\VEN_8086&DEV_A30E
│ │                       eda18be2-cab9-515d-b417-6a51da893121 ← PCI\VEN_8086&DEV_A30E&REV_10
│ │                       b3c53281-1520-5fbb-919b-800e2ad00788 ← PCI\VEN_8086&DEV_A30E&SUBSYS_17AA225F
│ │                       411f68e7-797c-5f87-a36a-9a4db27efde8 ← PCI\VEN_8086&DEV_A30E&SUBSYS_17AA225F&REV_10
│ │                       2d80f689-0b5e-5c4b-b6df-bd767f6e9f05 ← INTEL_SPI_CHIPSET\ID_PCH300
│ │   Device Flags:       • Internal device
│ │                       • Cryptographic hash verification is available
│ │ 
│ ├─BIOS:
│ │     Device ID:        ff7dbf2f6e354a5727c6ce1c466230f38bd26ff0
│ │     Vendor:           Intel Corporation (PCI:0x8086)
│ │     GUID:             2c7b8bd2-3e77-5730-aaf6-596136f5254d ← IFD\NAME_BIOS
│ │     Device Flags:     • Internal device
│ │                       • Cryptographic hash verification is available
│ │   
│ ├─Gigabit Ethernet:
│ │     Device ID:        11188287b93230d58f85f059dfab93e1d59724bb
│ │     Vendor:           Intel Corporation (PCI:0x8086)
│ │     GUID:             30ee4fad-d1cd-522f-9a32-e53f4a4269cc ← IFD\NAME_GBE
│ │     Device Flags:     • Internal device
│ │                       • Cryptographic hash verification is available
│ │   
│ └─Intel Management Engine:
│       Device ID:        7ef8a531d2413174034556f12dff8aa3bb4a8c30
│       Vendor:           Intel Corporation (PCI:0x8086)
│       GUID:             614566e7-4d60-55f5-9cb9-48c5e488a705 ← IFD\NAME_ME
│       Device Flags:     • Internal device
│                         • Device is locked
│                         • Cryptographic hash verification is available
[... 253 lines cut ...]
│                         • Signed Payload
│   
└─UEFI Device Firmware:
      Device ID:          f95c9218acd12697af946874bfe4239587209232
      Summary:            UEFI ESRT device
      Current version:    4784132
      Minimum Version:    4784132
      Vendor:             DMI:LENOVO
      Update State:       Success
      Update Error:       BootOrder is locked in firmware setup
      GUID:               22c037d2-4402-452a-aaa6-c0e3f0a1c9ea
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
```

</details>

**Additional questions**

- Operating system and version: Gentoo Linux 2.13
- Have you tried rebooting? Yes
- Is this a regression? Unknown

nvaert1986, 2023-04-06:
My kernel is compiled with TPM support and the TPM module is detected during boot:

dmesg | grep TPM Output:
[    0.000000] efi: TPMFinalLog=0x4fca2000 SMBIOS=0x4d441000 SMBIOS 3.0=0x4d434000 ACPI=0x4fd0e000 ACPI 2.0=0x4fd0e014 MEMATTR=0x429c9118 ESRT=0x4b2cc000 
[    0.001576] ACPI: TPM2 0x000000004D2F0000 000034 (v04 LENOVO TP-N2C   00001500 PTEC 00000002)
[    0.001669] ACPI: Reserving TPM2 table memory at [mem 0x4d2f0000-0x4d2f0033]
[    0.962370] tpm_tis STM7304:00: 2.0 TPM (device-id 0x0, rev-id 78)

securityfs support is compiled in as well, but the TPM module is not found there. I've tried resetting the TPM module in the UEFI firmware, but it doesn't resolve the issue.

ls -l /sys/kernel/security/ Output:
total 0
lr--r--r-- 1 root root 0 Apr  6 15:58 ima -> integrity/ima
drwxr-xr-x 3 root root 0 Apr  6 15:58 integrity
-r--r--r-- 1 root root 0 Apr  6 15:58 lsm

hughsie, 2023-04-06:
Does it work on this hardware with Fedora or Ubuntu LiveUSB? If so, the kernel support is missing for some reason.

superm1, 2023-04-06:
> fwupdmgr security --force reports TPM PCR0 reconstruction: Not found while a TPM 2.0 chip is present

fwupd is correct here with it's error message given the situation.  You don't have a TPM BIOS event log available to try to reconstruct and compare.

> Does it work on this hardware with Fedora or Ubuntu LiveUSB? If so, the kernel support is missing for some reason.

Specifically `/sys/kernel/security/tpm0/binary_bios_measurements`