fwupd/fwupd #5709: Dell WD19TB upgrade issue

dylanmtaylor, 2023-04-17:
**Describe the bug**

I have been trying to upgrade my WD19TB. 

I get to the point where it says 

Successfully installed firmware
The update will continue when the device USB cable has been unplugged.
The update will continue when the device USB cable has been unplugged.

and unplug the device for 5ish minutes. The updates don't seem to take though.

The prompts were answered as such:

╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Package level of Dell dock from 01.00.25.00 to 01.00.25.01?          ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ -Update the firmware package version of Dell docking station's composite     ║
║ devices.                                                                     ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall RTS5413 in Dell dock to 01.21?                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ -Same as previous release, there is no change for this component.            ║
║                                                                              ║
║ RTS5413 in Dell dock must remain connected for the duration of the update    ║
║ to avoid damage.                                                             ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall RTS5487 in Dell dock to 01.47?                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ -Same as previous release, there is no change for this component.            ║
║                                                                              ║
║ RTS5487 in Dell dock must remain connected for the duration of the update    ║
║ to avoid damage.                                                             ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall VMM5331 in Dell dock to 05.06.03?                                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ -Same as previous release, there is no change for this component.            ║
║                                                                              ║
║ VMM5331 in Dell dock must remain connected for the duration of the update    ║
║ to avoid damage.                                                             ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall WD19TB to 01.01.00.03?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ -Improved the display compatibility between systems and monitors.            ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Thunderbolt controller in Dell dock from 40.00 to 60.00?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ -Same as previous release, there is no change for this component.            ║
║                                                                              ║
║ Thunderbolt controller in Dell dock must remain connected for the duration   ║
║ of the update to avoid damage.                                               ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y

**Steps to Reproduce**
Steps to reproduce the behavior.

fwupdmgr upgrade

**Expected behavior**
[... 182 lines cut ...]
dual Lenovo 1440p monitors
Logitech unifying receiver for mouse/keyboard
2.5GBe usb-c ethernet adapter
Logitech C920s Pro HD Webcam

**Verbose daemon logs**
First enable daemon verbose logs collection.

```shell
fwupdmgr modify-config "VerboseDomains" "*"
```

Then try to reproduce the issue.  Even if it doesn't reproduce, please attach the
daemon verbose logs collected from the system journal.

```shell
journalctl -b -u fwupd.service
```

**Additional questions**

- Operating system and version: Ubuntu 23.04
- Have you tried unplugging the dock or any peripherals from your machine? Yes
- Have you tried to power cycle the dock from the AC adapter? Yes
- Is this a regression? Not sure.

hughsie, 2023-04-17:
> and unplug the device for 5ish minutes

Does the white LED flash when the dock is disconnected?

dylanmtaylor, 2023-04-17:
> > and unplug the device for 5ish minutes
> 
> Does the white LED flash when the dock is disconnected?

Yes. Three white blinks

hughsie, 2023-04-17:
@superm1 any ideas on this one?

superm1, 2023-04-17:
Everything looks correct to me here:
```
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   DeviceId:             9e3f96432729bff35a1346bf7da6947c3bc59e63
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   CompositeId:          5b80c4c065ca1a661535031a57e28bd1de0fb244
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Name:                 Thunderbolt controller in Dell dock
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Status:               device-restart
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Percentage:           1
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Guid:                 08a8c886-2818-544c-b1a3-588eb07ae487 <- THUNDERBOLT\VEN_00D4&DEV_B070
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Guid:                 48711a14-0aaa-55b8-a17f-f7c494b318c6 <- THUNDERBOLT\VEN_00D4&DEV_B070&REV_00
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Guid:                 c94770ca-1773-592c-b20a-e87243bc7cd0 <- TBT-00d4b070
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Guid:                 017a6b19-1f84-5be6-89f5-14ebd10a52de <- TBT-00d4b070-controller0-1
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Summary:              Thunderbolt controller
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Plugin:               thunderbolt
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Protocol:             com.intel.thunderbolt
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Flags:                require-ac|supported|registered|install-parent-first|dual-image|usable-during-update|updatable-hidden|skips-restart|signed-payload
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Problems:             update-in-process
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Vendor:               Dell Inc.
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   VendorId:             THUNDERBOLT:0x00D4
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   VendorId:             TBT:0x00D4
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Version:              40.00
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   VersionFormat:        pair
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Icon:                 thunderbolt
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   InstallDuration:      22
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Created:              2023-04-17
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Modified:             2023-04-17
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   UpdateError:          An update is in progress
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Guid[quirks]:         bf9358fb-ddfc-5d10-9d0e-88180feb70ae <- THUNDERBOLT\VEN_00D4
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Guid[quirks]:         f70b02b8-cac4-5138-aef2-83a4b2d3afef <- THUNDERBOLT\TYPE_THUNDERBOLT_DEVICE
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Guid[quirks]:         d239c23b-37f1-5392-9bb5-dd1c819f38f3 <- THUNDERBOLT
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   PhysicalId:           c8010000-0070-7c0e-035c-3314d051d121
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   BackendId:            /sys/devices/pci0000:00/0000:00:0d.2/domain0/0-0/0-1
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   AcquiesceDelay:       2500
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   CustomFlags:          dual-image
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   FirmwareSizeMin:      262144
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   FirmwareSizeMax:      524288
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Order:                1
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   PossiblePlugin:       thunderbolt
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Inhibit:              [update-in-process] An update is in progress
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Vendor:               0xd4
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Model:                0xb070
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Subsystem:            thunderbolt
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   SysfsPath:            /sys/devices/pci0000:00/0000:00:0d.2/domain0/0-0/0-1
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   AuthMethod:           nvm_authenticate_on_disconnect
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Device Type:          Thunderbolt device controller
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Safe Mode:            false
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Native mode:          false
Apr 16 23:46:10 Dylan-Precision-5570 fwupd[8933]:   Generation:           3
```
@CragW can you reproduce this perhaps?

@dylanmtaylor try the same suggestion as @esmil had in https://github.com/fwupd/fwupd/issues/4307 and see if that helps.  If it does, it might be a kernel issue.

superm1, 2023-04-17:
Before you try that - as a guess can you add `thunderbolt.clx_enabled=0` to your kernel command line and reboot?  It would be good to know if that still fails.

dylanmtaylor, 2023-04-17:
Okay, so here is what I figured out based on this conversation - I plugged the TB dock into the right USB-C on the computer which is NOT Thunderbolt. I ran the upgrade and it took 3-4 times as long, a good sign. Now it is showing it as updated when it is connected back to the TB port :)

$ fwupdmgr upgrade
Devices with no available firmware updates: 
 • Fingerprint Sensor
 • PC801 NVMe SK hynix 1TB
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • USB4 Retimer
 • USB4 Retimer
 • USB4 Retimer
 • USB4 Retimer
 • Unifying Receiver
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19TB
 • System Firmware
 • Thunderbolt controller in Dell dock

dwzrd86, 2023-04-22:
Ok so I have the same issue. HOWEVER, my System76 Darter Pro only has the 1 USB-C port that is also apparently thunderbolt.  Any suggestions on this?

esmil, 2023-04-22:
@dwzrd86 What I did was just to inject a USB hub between the dock and computer to force the connection into USB mode.

superm1, 2023-04-22:
> Ok so I have the same issue. HOWEVER, my System76 Darter Pro only has the 1 USB-C port that is also apparently thunderbolt.  Any suggestions on this?

Can you try the kernel module parameter I suggested to disable clx and see if it helps?

bionade24, 2023-04-23:
I do have the save issue. The update does not proceed after unplugging the USB(Thunderbolt) cable, the Dock only blinks once. I tried multiple times, I also applied the kernel flag. Sorry for the 1st log not being in English.

Without kernel cmdline flag:
```
1d [oskar:~] 130 % sudo fwupdmgr get-updates
Geräte mit keinen verfügbaren Firmware-Aktualisierungen:
 • Internal SPI Controller
 • KBG40ZPZ512G NVMe KIOXIA 512GB
 • UEFI Device Firmware
 • UEFI Device Firmware
Geräte mit der neuesten verfügbaren Firmware-Version:
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19TB
 • System Firmware
 • UEFI dbx
Dell Inc. XPS 13 9310 2-in-1
│
├─Package level of Dell dock:
│ │   Gerätekennung:      073624c16dd99abe01ba1da223a70321e4f29beb
│ │   Zusammenfassung:    A representation of dock update status
│ │   Aktuelle Version:   01.00.25.00
│ │   Anbieter:           Dell Inc. (USB:0x413C)
│ │   Installationsdauer: 5 Sekunden
│ │   GUIDs:              d8927ff5-a5b2-5618-848b-8e8bfb75b66f
│ │                       8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
│ │   Device Flags:       • Aktualisierbar
│ │                       • Unterstützt auf dem entfernten Server
│ │                       • Das Gerät kann sich nach Fehlern beim Aufspielen wiederherstellen
│ │                       • Das Gerät ist während der Dauer der Aktualisierung nutzbar
│ │                       • Unsignierte Nutzdaten
│ │
│ └─WD19/WD22/HD22 Series:
│       Neue Version:     01.00.25.01
│       Remote ID:        lvfs
│       Release ID:       13766
│       Zusammenfassung:  Firmware package version for WD19 Series Dock
│       Lizenz:           Proprietär
│       Größe:            24 Bytes
│       Erstellt:         2022-06-08
│       Dringlichkeit:    Hoch
│       Anbieter:         Dell
│       Dauer:            5 Sekunden
│       Release Flags:    • Vertrauenswürdige Metadaten
│                         • Ist Höherstufung
│       Beschreibung:
│       -Update the firmware package version of Dell docking station's composite devices.
│
└─Thunderbolt controller in Dell dock:
  │   Gerätekennung:      47379425778d9a6e477b0c13132001260cf8f324
  │   Zusammenfassung:    Thunderbolt controller
  │   Aktuelle Version:   43.00
  │   Anbieter:           Dell Inc. (THUNDERBOLT:0x00D4, TBT:0x00D4)
  │   Installationsdauer: 22 Sekunden
  │   GUIDs:              08a8c886-2818-544c-b1a3-588eb07ae487 ← THUNDERBOLT\VEN_00D4&DEV_B070
  │                       48711a14-0aaa-55b8-a17f-f7c494b318c6 ← THUNDERBOLT\VEN_00D4&DEV_B070&REV_00
  │                       c94770ca-1773-592c-b20a-e87243bc7cd0 ← TBT-00d4b070
  │                       d9d01d94-4749-58f4-b318-87cb92251fdb ← TBT-00d4b070-controller1-1
  │   Device Flags:       • Aktualisierbar
  │                       • System benötigt externe Stromquelle
  │                       • Unterstützt auf dem entfernten Server
  │                       • Zuerst auf dem übergeordneten Gerät installieren
  │                       • Gerät führt Aktualisierungen in Etappen durch
  │                       • Das Gerät ist während der Dauer der Aktualisierung nutzbar
  │                       • Signierte Nutzdaten
  │
  └─WD19 Thunderbolt:
        Neue Version:     60.00
        Remote ID:        lvfs
[... 178 lines cut ...]
⏚ [oskar:~] %
⏚ [oskar:~] % cat /boot/loader/entries/arch.conf
title Arch Linux
linux \vmlinuz-linux-zen
initrd \intel-ucode.img
initrd \initramfs-linux-zen.img
options rd.luks.name=5hd67d30-d63a-419e-9517-5f806da2g70a=cryptroot root=/dev/mapper/cryptroot rootflags=subvol=@root rw resume=/dev/mapper/cryptroot resume_offset=139571 rd.lu
ks.options=discard thunderbolt.clx_enabled=0
⌁69% [oskar:~] % LANG=C sudo -E fwupdmgr update
Devices with no available firmware updates:
 ? Internal SPI Controller
 ? KBG40ZPZ512G NVMe KIOXIA 512GB
 ? UEFI Device Firmware
 ? UEFI Device Firmware
????????????????????????????????????????????????????????????????????????????????
? Upgrade Package level of Dell dock from 01.00.25.00 to 01.00.25.01?          ?
????????????????????????????????????????????????????????????????????????????????
? -Update the firmware package version of Dell docking station's composite     ?
? devices.                                                                     ?
????????????????????????????????????????????????????????????????????????????????
Perform operation? [Y|n]: n
Request canceled
```

I will now try with a USB-C only port in a different device.

superm1, 2023-04-23:
Thanks. The kernel flag was a shot in the dark at what a big change to the thunderbolt driver since I knew this was working before.

I think @CragW will need to dig in more. If someone else hits this, I think it would be good to try it with the LTS kernel 5.15.y.

bionade24, 2023-04-24:
> Before you try that - as a guess can you add thunderbolt.clx_enabled=0 to your kernel command line and reboot? It would be good to know if that still fails.

> Thanks. The kernel flag was a shot in the dark at what a big change to the thunderbolt driver since I knew this was working before.

@superm1 @CragW While tracking down another regression that may be thunderbolt-related I looked into the kernel log:  ```
[    5.418336] thunderbolt: unknown parameter 'clx_enabled' ignored
```

The correct kernel flag is just `thunderbolt.clx=0`. So maybe you're on the right path, sadly I can't test it anymore :(

dwzrd86, 2023-04-24:
ggestions on this?
> 
> Can you try the kernel module parameter I suggested to disable clx and see if it helps?

> > Ok so I have the same issue. HOWEVER, my System76 Darter Pro only has the 1 USB-C port that is also apparently thunderbolt.  Any suggestions on this?
> 
> Can you try the kernel module parameter I suggested to disable clx and see if it helps?

Here is /proc/cmdline after I rebooted and added parameter:

`initrd=\EFI\Pop_OS-73f230dc-96e6-40c3-af43-32133c27833a\initrd.img root=UUID=73f230dc-96e6-40c3-af43-32133c27833a ro quiet loglevel=0 systemd.show_status=false splash thunderbolt.clx=0
`

Fetching updates:
`sudo fwupdmgr get-updates
[sudo] password for deemd: 
Devices with no available firmware updates: 
 • Sabrent
 • Thunderbolt host controller
 • Unifying Receiver
Devices with the latest available firmware version:
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19TB
Darter Pro
│
├─Package level of Dell dock:
│ │   Device ID:          073624c16dd99abe01ba1da223a70321e4f29beb
│ │   Summary:            A representation of dock update status
│ │   Current version:    01.00.25.00
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   5 seconds
│ │   GUIDs:              d8927ff5-a5b2-5618-848b-8e8bfb75b66f
│ │                       8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Unsigned Payload
│ │ 
│ └─WD19/WD22/HD22 Series:
│       New version:      01.00.25.01
│       Remote ID:        lvfs
│       Release ID:       13766
│       Summary:          Firmware package version for WD19 Series Dock
│       License:          Proprietary
│       Size:             24 bytes
│       Created:          2022-06-08
│       Urgency:          High
│       Vendor:           Dell
│       Duration:         5 seconds
│       Release Flags:    • Is upgrade
│       Description:      
│       -Update the firmware package version of Dell docking station's composite devices.
│     
└─Thunderbolt controller in Dell dock:
  │   Device ID:          72bb2207c7113bf8822d6e5ee6463da17f39eab2
  │   Summary:            Thunderbolt controller
  │   Current version:    43.00
  │   Vendor:             Dell Inc. (THUNDERBOLT:0x00D4, TBT:0x00D4)
  │   Install Duration:   22 seconds
  │   GUIDs:              08a8c886-2818-544c-b1a3-588eb07ae487 ← THUNDERBOLT\VEN_00D4&DEV_B070
  │                       48711a14-0aaa-55b8-a17f-f7c494b318c6 ← THUNDERBOLT\VEN_00D4&DEV_B070&REV_00
  │                       c94770ca-1773-592c-b20a-e87243bc7cd0 ← TBT-00d4b070
  │                       017a6b19-1f84-5be6-89f5-14ebd10a52de ← TBT-00d4b070-controller0-1
  │   Device Flags:       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Install to parent device first
[... 151 lines cut ...]
  │                       017a6b19-1f84-5be6-89f5-14ebd10a52de ← TBT-00d4b070-controller0-1
  │   Device Flags:       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Install to parent device first
  │                       • Device stages updates
  │                       • Device is usable for the duration of the update
  │                       • Signed Payload
  │ 
  └─WD19 Thunderbolt:
        New version:      60.00
        Remote ID:        lvfs
        Release ID:       13777
        Summary:          Firmware update for WD19 Thunderbolt
        License:          Proprietary
        Size:             417.8 kB
        Created:          2022-06-08
        Urgency:          High
        Vendor:           Dell
        Duration:         22 seconds
        Release Flags:    • Is upgrade
        Description:      
        -Same as previous release, there is no change for this component.
`
The flag doesn't seem to work.

dwzrd86, 2023-04-24:
I ran the updates again and then that command.

Output:
```
sudo fwupdtool activate
17:14:21:0379 FuRemoteList         loading /usr/share/fwupd/metainfo
17:14:21:0379 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
17:14:21:0380 XbSilo               file: 31145ac7-2f58-6063-8730-e3dfdfcdae93, current:31145ac7-2f58-6063-8730-e3dfdfcdae93, cached: (null)
17:14:21:0380 XbSilo               loading silo with file contents
17:14:21:0380 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
17:14:21:0380 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
17:14:21:0380 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
17:14:21:0380 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
17:14:21:0380 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
17:14:21:0380 FuRemoteList         ordering lvfs-testing=lvfs+1
17:14:21:0380 FuRemoteList         ignoring unfound remote fwupd
17:14:21:0380 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
17:14:21:0380 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
17:14:21:0380 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
17:14:21:0380 Jcat                 loaded 1 certificates
17:14:21:0380 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
17:14:21:0380 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
17:14:21:0380 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
17:14:21:0380 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
17:14:21:0381 Jcat                 loaded 1 certificates
17:14:21:0381 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
17:14:21:0381 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
17:14:21:0381 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
17:14:21:0381 FuEngine             client certificate now exists: nothing to do
17:14:21:0381 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
17:14:21:0381 FuHistory            got schema version of 7
17:14:21:0381 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
17:14:21:0391 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
17:14:21:0391 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
17:14:21:0391 XbSilo               file: 4d58b889-ccb1-4e19-9055-e3410f9e015b, current:4d58b889-ccb1-4e19-9055-e3410f9e015b, cached: (null)
17:14:21:0391 XbSilo               loading silo with file contents
17:14:21:0392 FuSmbios             trying to read /sys/class/dmi/id
17:14:21:0392 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
17:14:21:0392 FuHwids              not loading HwId overrides from /var/lib/fwupd/daemon.conf
17:14:21:0392 FuHwids              smbios property Manufacturer=System76
17:14:21:0392 FuHwids              smbios property EnclosureKind=9
17:14:21:0392 FuHwids              smbios property Family=
17:14:21:0392 FuHwids              smbios property ProductName=Darter Pro
17:14:21:0392 FuHwids              smbios property ProductSku=
17:14:21:0392 FuHwids              smbios property BiosVendor=coreboot
17:14:21:0392 FuHwids              smbios property BiosVersion=2020-05-20_de5eab8
17:14:21:0392 FuHwids              smbios property BiosMajorRelease=04
17:14:21:0392 FuHwids              smbios property BiosMinorRelease=0b
17:14:21:0392 FuHwids              smbios property FirmwareMajorRelease=00
17:14:21:0392 FuHwids              smbios property FirmwareMinorRelease=00
17:14:21:0392 FuHwids              smbios property BaseboardManufacturer=System76
17:14:21:0392 FuHwids              smbios property BaseboardProduct=Darter Pro
17:14:21:0396 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory
17:14:21:0396 FuEngine             ignoring: Error opening directory “/usr/share/fwupd/local.d”: No such file or directory
17:14:21:0396 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
17:14:21:0396 XbSilo               file: 32015f25-0802-1530-29e9-453f37d45159, current:32015f25-0802-1530-29e9-453f37d45159, cached: (null)
17:14:21:0396 XbSilo               loading silo with file contents
17:14:21:0397 FuEngine             2034 components now in silo
17:14:21:0418 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_thunderbolt.so)
17:14:21:0418 FuContext            added udev subsystem watch of thunderbolt
17:14:21:0418 FuEngine             plugins disabled: pci_mei, mtd, rts54hub, uefi_capsule, wacom_raw, dell_esrt, synaptics_mst, uefi_pk, synaptics_rmi, gpio, ep963x, dell_dock, colorhug, elantp, bios, uefi_recovery, nordic_hid, cpu, acpi_ivrs, parade_lspcon, hailuck, pixart_rf, logind, logitech_hidpp, emmc, tpm, elanfp, superio, uefi_dbx, acpi_phat, modem_manager, iommu, lenovo_thinklmi, nvme, redfish, ccgx, scsi, synaptics_cape, steelseries, thelio_io, pci_bcr, nitrokey, dell, ata, upower, amt, uf2, linux_lockdown, genesys, jabra, dfu, corsair, acpi_facp, optionrom, system76_launch, wacom_usb, synaptics_cxaudio, flashrom, vli, pci_psp, cfu, msr, linux_sleep, ch341a, linux_swap, acpi_dmar, ebitdo, goodixmoc, anal [... cut ...]
  Device Type:          Thunderbolt device controller
  Safe Mode:            false
  Native mode:          false
  Generation:           3

17:14:22:0524 FuEngine             Emitting PropertyChanged('Status'='idle')
Loading…                 [-                                      ]
17:14:22:0530 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
No devices to activate
17:14:22:0534 FuMain               No devices to activate
```
Same results also after plugging back in.

dwzrd86, 2023-04-24:
> @dwzrd86 What I did was just to inject a USB hub between the dock and computer to force the connection into USB mode.

I'll try this but I need to get a new hub as the one I have only has USB-C power input and not an actual port.  I will report back later this week.

dwzrd86, 2023-04-26:
@esmil can you tell me what hub you used?  I purchased this [https://www.amazon.com/dp/B09Z2167CN?psc=1&ref=ppx_yo2ov_dt_b_product_details](url) from Amazon. I've tried plugging it in to the port and then the dock into the usb-c data port and it does nothing.

closed 2023-09-23 by hughsie

SinBirb, 2024-02-17:
Why was this marked as completed? I have exactly the same issue with only one USB-C port on my notebook. Also I don't have a USB-C to USB-A dock or cable.

```
> fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.Passim        0.1.5
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.7.4-zen1-1-zen
runtime   org.freedesktop.fwupd         1.9.13
```

superm1, 2024-02-17:
> Why was this marked as completed? I have exactly the same issue with only one USB-C port on my notebook. Also I don't have a USB-C to USB-A dock or cable.

Please open a new issue with new logs.  You've a much newer version, it might be something else.

G0ldmember, 2024-09-20:
I had the same problem and found out that if you do NOT disconnect the DS / unplug the cable but just click 3 times on "close", the firmware upgrade process continued. Seems kind of buggy...

3ng7n33r, 2024-10-01:
> I had the same problem and found out that if you do NOT disconnect the DS / unplug the cable but just click 3 times on "close", the firmware upgrade process continued. Seems kind of buggy...

Same here, seems to be the solution

raheelk, 2025-07-03:
Had the same issue, Instead a USB-C to USB-A 3.1 adapter did the trick, updated all drivers that were otherwise stuck.

USB HUB work around didn't work for me. I guess it is hub specific, some work some don't, based on the anecdotal evidence in the issues.

The device also used to show up as a malfunctioning USB device when I tried to update via windows using the .exe, dock was not even recognised by the installer oe windows

This adapter literally revived the dock from the dead.

I am based in NZ, this is the exact adapter I used:
https://www.pbtech.co.nz/product/ADPDNX1046/Dynamix-A-USBCFAM-USB-C-Female-to-USB-A-Male-Adapt

UmbraAtrox, 2026-10-19:
Leaving this here because this issue and https://github.com/fwupd/fwupd/issues/6811 show up first on google when searching for W19TB not updating

Laptop: Framework 13 11th gen - has no non-tb ports afaik so couldn't try using those and i have no USB-C to USB-A adapter
Dock: Dell W19TB bought refurbished

All components needed an update and updated fine except "Package level of Dell dock" (updated 01.00.04.01 -> 01.01.03.00, instead of 01.01.03.01) and "Thunderbolt controller in Dell dock" (stayed same version 40.00).

Multiple cycles of fwupdmgr update -> wait until it tells you to unplug -> unplug USB  (power button flashed 3 times) & wait 30s -> unplug AC & wait 30s -> plug AC back in (power button flashed 3 times) & wait 30s did not get those two updated. 

There also was a message saying update will continue after unplug with a second Y/n prompt for Thunderbolt controller 40 -> 60 which after replug said device not found. (Sorry should have taken screencaps)

What's odd is even when those components didn't update the fwupdmgr upload thingy said they did when i ran fwupdmgr update after a update attempt and it asked to upload, i assume those results where uploaded despite being wrong since it also said those two have updates. 

The solution was: 

To get the initial working updates:
- fwupdmgr update -> wait until complete
- powercycle dock (unplug usb, 30s, unplug power, 30s, plug power, 30s, plug usb)
- 
To get the 2 remaining updates:
- run fwupdmgr update again and notice the two components on old firmware
- wait until it finishes and says to unplug USB -> don't unplug usb
- run "fwupdmgr activate" in another terminal -> That then said it activated fw on dell dock. 
- unplug & powercycle dock again with 30s waits inbetween. 

fwupdmgr update now says:
```
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19TB
 • Fingerprint Sensor
 • Thunderbolt controller in Dell dock
 • Unifying Receiver
```

fwupdmgr get-history 
```
Framework Laptop
│
├─RTS5413 in Dell dock:
│ │   Device ID:          12e83c17bab7a5fd17ce1cd8b314c3403415e237
│ │   Previous version:   01.21
│ │   Install Duration:   13 seconds
│ │   Update State:       Success
│ │   Last modified:      2026-10-19 14:55:31
│ │   GUID:               b27d25f1-019d-5718-b41a-02ddaefe5577
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Reported to remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │ 
│ └─WD19/WD22 USB 3.1 Generation 1 Controller Update:
│       New version:      01.25
│       Remote ID:        lvfs
│       Release ID:       144821
│       Summary:          Firmware update for WD19/WD22 USB 3.1 Generation 1
│       License:          Proprietary
│       Size:             65.5 kB
│       Created:          2026-07-14 15:33:32
│       Urgency:          High
│         Tested:         2026-10-14 00:00:00
│         Distribution:   chromeos 141
│         Old version:    01.62
│         Version[fwupd]: 2.0.13
[... 156 lines cut ...]
  │                       • Device is usable for the duration of the update
  │                       • Updatable
  │                       • Signed Payload
  │                       • Can tag for emulation
  │ 
  └─Thunderbolt Controller Update:
        New version:      60.00
        Remote ID:        lvfs
        Release ID:       144825
        Summary:          Firmware update for Thunderbolt
        License:          Proprietary
        Size:             417.8 kB
        Created:          2026-07-14 15:33:32
        Urgency:          High
          Tested:         2026-10-14 00:00:00
          Distribution:   chromeos 141
          Old version:    01.62
          Version[fwupd]: 2.0.13
        Vendor:           Dell
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:      
        This package contains the firmware update for Dell Dock WD19 and WD22TB4. The dock firmware update package contains utility and firmware for updating the dock components.- Improved the stability of the system.
        Checksum:         4bc576f1d44b4dfda66bca65847a463165e678a3e57a972c34ff7cf4248ad42f
```