fwupd/fwupd #5717: Unable to disable WARNING: UEFI capsule updates not available banner

joelsgp, 2023-04-18:
I got this message:
```
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
```
tried to disable it as suggested:
```
ile: /etc/fwupd/daemon.conf
───────┼──────────────────────────────────────────────────────────────────────────────────
   1   │ [fwupd]
   2   │ 
   3   │ # Allow blocking specific devices by their GUID
   4   │ # Uses semicolons as delimiter
   5   │ DisabledDevices=
   6   │ 
   7   │ # Allow blocking specific plugins
   8   │ # Uses semicolons as delimiter
   9   │ DisabledPlugins=test;test_ble;invalid;uefi

```
Then I restarted the daemon with systemctl restart fwupd.service. After that didn't work I rebooted. Now I'm still getting the banner.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.14
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.1.24-1-lts
runtime   org.freedesktop.fwupd         1.8.14
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): arch community repo package

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Micro-Star International Co., Ltd MS-7C02
│
├─Unknown Device:
│     Device ID:          a29de8124928f7dc31db026069ac1d601a56056f
│     GUID:               2d9ac22e-0ffe-5691-9e21-f9fecce8761c ← GPIO\ID_AMDIF030:00
│   
├─400 Series Chipset SATA Controller:
│     Device ID:          1b0e7fd057b65672902ebad0ad5099d23bd81891
│     Current version:    01
│     Vendor:             Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│     GUIDs:              40dd51ce-4d2b-5863-a55b-e5e05b828e18 ← PCI\VEN_1022&DEV_43C8
│                         37d930ee-e811-585c-a094-f430aa4d42fb ← PCI\VEN_1022&DEV_43C8&REV_01
│                         b315bc76-8da0-5bc6-82e8-296673694802 ← PCI\VEN_1022&DEV_43C8&SUBSYS_1B211062
│                         24f9394e-567d-573a-9e22-4ce1d0668241 ← PCI\VEN_1022&DEV_43C8&SUBSYS_1B211062&REV_01
│                         2d530482-50db-5cbc-97f7-6ae02507276d ← PCI\VEN_1022&DEV_1483
│                         9540483b-6bd7-580f-be16-212758d306d4 ← PCI\VEN_1022&DEV_1483&REV_00
│                         830befb6-3214-5c02-bde1-f65c116f71fc ← PCI\VEN_1022&DEV_1483&SUBSYS_10221234
│                         2cbddce9-c1ea-5441-afbd-9eb7778ff6f4 ← PCI\VEN_1022&DEV_1483&SUBSYS_10221234&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
[... 594 lines cut ...]
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── refind_linux.conf
└── vmlinuz-linux-lts

10 directories, 457 files

```

**Additional questions**

- Operating system and version: Arch 6.1.24-1-lts
- Have you tried rebooting? Yes
- Is this a regression? Don't know
- Are you using an NVMe disk? No
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2023-04-18:
> DisabledPlugins=test;test_ble;invalid;uefi

It's called `uefi_capsule` now -- if that works can you fix the wiki page please.

joelsgp, 2023-04-18:
Tried, rebooted and still getting the banner

hughsie, 2023-04-19:
Ahh, wrong plugin I think. Try using `bios` instead. `fwupdmgr get-plugins` will tell you which one it's coming from.

joelsgp, 2023-04-19:
Thanks, this fixed it

closed 2023-04-19 by joelsgp

joelsgp, 2023-04-19:
Done https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported