fwupd/fwupd #5720: fwupdate ThinkPad L14L15 Gen3 AMD System Firmware doesn't work

artificial-intelligence, 2023-04-18:
**Describe the bug**
A clear and concise description of what the bug is.
upgrade from 0.1.16 to 0.1.22 doesn't work
**Steps to Reproduce**
Steps to reproduce the behavior.
there is already a bugreport - I guess in the wrong - repository here: https://github.com/fwupd/firmware-lenovo/issues/320

maybe the issue board there could be closed?
**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
fwupdmgr --version                                                                                                                                                    ✔  at 17:57:37 
compile   org.freedesktop.fwupd         1.8.14
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.14
runtime   org.kernel                    6.2.10-200.fc37.x86_64

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
fwupd installed via dnf (it is afaik installed by default in fedora 37)
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
fwupdmgr get-devices --show-all-devices                                                                                                                               ✔  at 18:05:40 
LENOVO 21C5CTO1WW
│
├─AMD Ryzen 7 PRO 5875U with Radeon Graphics:
│ │   Gerätekennung:      4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Aktuelle Version:   0x0a50000d
│ │   Anbieter:           Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│ │                       79759cdc-94db-5098-be7b-eb02521fbbec ← CPUID\PRO_0&FAM_19&MOD_50
│ │                       20b595b0-5892-5870-8e4c-688133ad6e34 ← CPUID\PRO_0&FAM_19&MOD_50&STP_0
│ │   Device Flags:       • Internes Gerät
│ │ 
│ ├─GPIO controller:
│ │     Gerätekennung:    f685512aa07369c9e77742acef941d779d31e766
│ │     GUID:             37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│ │   
│ ├─Graphics Processing Unit (GPU):
│ │     Gerätekennung:    e32966e41f54b47823eef28bd8b4bd9e1018b923
│ │     Aktuelle Version: 113-BARCELO-004
│ │     Anbieter:         Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │     GUIDs:            a1a3e8c2-31df-508c-a926-7590fa528a99 ← PCI\VEN_1002&DEV_15E7
│ │                       efb3aa46-15af-5f26-a49b-3836243084f1 ← PCI\VEN_1002&DEV_15E7&REV_D1
│ │                       150ac9e3-676b-5621-b0ca-8418de52adef ← PCI\VEN_1002&DEV_15E7&SUBSYS_17AA50AE
│ │                       655c1563-ab48-5f46-aaa4-1048f41a96ad ← PCI\VEN_1002&DEV_15E7&SUBSYS_17AA50AE&REV_D1
│ │     Device Flags:     • Internes Gerät
│ │   
│ ├─Secure Processor:
│ │     Gerätekennung:    c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Anbieter:         Advanced Micro Devices, Inc. (PCI:0x1022)
[... 198 lines cut ...]
      Aktuelle Version:   65550
      Minimale Version:   65550
      Anbieter:           DMI:LENOVO
      Aktualisierungsstatus:Erfolg
      GUID:               4bea12df-56e3-4cdb-97dd-f133768c9051
      Device Flags:       • Internes Gerät
                          • Aktualisierbar
                          • System benötigt externe Stromquelle
                          • Benötigt einen Neustart nach der Installation
                          • Das Gerät ist während der Dauer der Aktualisierung nutzbar

let's me directly report a bug which points me here:

https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot

which says that it might be a bug involving secure boot, but I don't want to disable secure boot on this device.
```

</details>

**Additional questions**

- Operating system and version: fedora 37 workstation kde spin
- Have you tried rebooting? yes
- Is this a regression? don't know

artificial-intelligence, 2023-04-18:
also, all the linked bug reports at https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot are either closed or the linked PRs are merged, so should this really still be a problem? if so, where is the current open bug which can be tracked? is there another workaround beside degrading the security of the device in question?

hughsie, 2023-04-18:
I think the other bug is the appropriate place, sorry.

closed 2023-04-18 by hughsie