fwupd/fwupd #5728: /usr/lib/systemd/system-shutdown/fwupd.shutdown assumes /var/ is mounted

poettering, 2023-04-20:
**Describe the bug**

I noticed that the /usr/lib/systemd/system-shutdown/fwupd.shutdown script that fwupd installs assumes that it can access /var/. Thing is though: it cannot assume that. The "system-shutdown" callouts are invoked after everything but the root fs (and /run/ and some api vfs) are unmounted. Thus, on systems where /var/ is a separate partition the script cannot work.

I extended the docs about that in systemd now:

https://github.com/systemd/systemd/pull/27343

Instead of checking /var/ you could create a touch file or so in /run/ instead, which will be available this late, and then stop accessing /var/ from the script.

And yes, on many hosts /var/ is *not* split off, but on many others it will be, and this fwupd script won't work there.

(btw, we try to get shell scripts out of the boot and shutdown process. Might be nice to get rid of the script. You could for example symlink some binary of your choice into the dir, and then simply check argv[0] to see how you are invoked, i.e. as multi-call binary. But of course, this doesn't matter much)

**fwupd version information**

```shell
fwupdmgr --version$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.14
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.14
runtime   org.kernel                    6.1.6-200.fc37.x86_64
```

poettering, 2023-04-20:
BTW, why do you even need this? What does the script actually do?

hughsie, 2023-04-20:
> What does the script actually do?

In most cases it's when the [S]ATA firmware has been updated on a drive, and we need to do `ATA_OP_FLUSH_CACHE`, `ATA_OP_STANDBY_IMMEDIATE` and `ATA_OP_DOWNLOAD_MICROCODE` in the current boot, but when the system is otherwise idle and about to restart.

poettering, 2023-04-20:
> In most cases it's when the [S]ATA firmware has been updated on a drive, and we need to do `ATA_OP_FLUSH_CACHE`, `ATA_OP_STANDBY_IMMEDIATE` and `ATA_OP_DOWNLOAD_MICROCODE` in the current boot, but when the system is otherwise idle and about to restart.

Note that these callouts are invoked with the rootfs still mounted, and everything backing it still attached. i.e. if you have md or dm backing you rootfs then it will still be set up. Hence not sure what you mean by "idle" specifically.

An initrd can define a hook (we nowadays call this the "exitrd") that can unmount/detach the root fs too if that's desirable. dracut does that automatically on dm/md/lvm setups where the rootfs is located on such complex storage. Various initrd implementations then detach the root fs and everything they need, and then invoke `systemd-shutdown` a second time, which will then invoke the callouts a second time – but of course only those contained in the exitrd.

So long story short: if you want the harddisks to be entirely idle, then you have to run your stuff from the exitrd, and thus plug yourself into the exitrd somehow. Not sure how that works with dracut though, in detail. Would have to contact the dracut people about that...

closed 2023-09-23 by hughsie