fwupd/fwupd #5731: systemd timer fwupd-refresh.timer doesn't trigger on devices that reboot frequently

AidanGG, 2023-04-20:
**Describe the bug**
Due to the lines used in the systemd timer: https://github.com/fwupd/fwupd/blob/7fa422744c42d5f6a207d43c094d02dece64baf6/data/motd/fwupd-refresh.timer#L5-L8

the randomised delay is applied after reboots even when the last trigger time is far into the past (upstream report https://github.com/systemd/systemd/issues/21166). This means that when a device due for a refresh is left on for `n` hours, it has only an `n/12` probability of triggering the refresh.

**Steps to Reproduce**
Enable the `fwupd-refresh.timer`, leave the device off until the next scheduled refresh time (i.e. until 0600 or 1800 according to the `fwupd-refresh.timer`), then only use the device for about one hour at a time before rebooting. Notice that the timer will very rarely trigger.

**Expected behavior**
The timer triggers once every 12h but at randomised times within those intervals.

**fwupd version information**
Issue exists on main.

**Additional questions**
Maybe we could tweak the default values used in the timer, or direct users with specific needs to set their own values. But then, to follow good etiquette, what would the maximum frequency for refreshes be?

hughsie, 2023-04-20:
> what would the maximum frequency for refreshes be

Not more than twice a day I'd say. @superm1 got any ideas?

superm1, 2023-04-20:
We actually have a protection in place that we don't hit the server more than once every 24 hours by default.
https://github.com/fwupd/fwupd/blob/main/src/fu-util.c#L1810

So the 12 hours time was picked to try to get a request in there.  As we protect the server already I don't feel married to the timer value, so please feel free to make a PR with an argument for another default value .

AidanGG, 2023-04-20:
Oh right, the `fwupd-refresh.service` unit doesn't call `--force`, so the timer can be much more frequent. I can open a PR.

hughsie, 2023-04-20:
I think once per hour would be okay, no?

PR #5743 (2023-04-21): Have fwupd-refresh.timer trigger once per hour on average
Due to systemd timers applying RandomizedDelaySec upon reboot, the previous 12h maximum delay meant that the timer would rarely trigger on devices that reboot often (systemd/systemd#21166). We set the timer to trigger with a randomised delay of up to 1h, every hour.

Fixes #5731.

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit ca646958 (2023-04-21): Have fwupd-refresh.timer trigger once per hour on average

Due to systemd timers applying RandomizedDelaySec upon reboot, the previous 12h
maximum delay meant that the timer would rarely trigger on devices that reboot
often (systemd/systemd#21166). We set the timer to trigger with a randomised delay
of up to 1h, every hour.

Fixes #5731.
commit c41ffc44 (2023-04-28): Have fwupd-refresh.timer trigger once per hour on average

Due to systemd timers applying RandomizedDelaySec upon reboot, the previous 12h
maximum delay meant that the timer would rarely trigger on devices that reboot
often (systemd/systemd#21166). We set the timer to trigger with a randomised delay
of up to 1h, every hour.

Fixes #5731.