fwupd/fwupd #5738: plugins: nordic-hid: Invalid VID and PID of devices connected via dongle

MarekPieta, 2023-04-21:
**Describe the bug**
`nordic-hid` plugin reports VID and PID of the nRF Desktop dongle for peripherals connected via the dongle.
That  breaks our DFU mechanism (the *.cab packages cannot be properly matched because of unexpected GUID value).
Previously such peripherals used hardcoded VID and PID of `0x0000`, which was taken into account in GUIDs used by *.cab file.
Every peripheral was assigned an additional GUID with VID and PID of `0x0000` instead of peripheral's actual VID and PID.
With the current implementation the workaround cannot be applied - assigning every peripheral GUID related to each supported dongle the peripheral can be connected through would significantly limit interoperability.

**Steps to Reproduce**
Use `fwupdtool` with an nRF Desktop HID peripheral device connected through nRF Desktop dongle.
Check GUID strings reported for the device.

**Expected behavior**
The devices should report their own VID and PID.
Since this may not be possible for devices connected through nRF Desktop dongle (host has no direct access to the VID and PID values), we can either pass VID and PID over configuration channel protocol or use a fixed, hardcoded value (like 0x0000 which was used before).
Keep in mind that passing VID and PID over configuration channel would not be possible for devices with legacy firmware.

**fwupd version information**
Tested with `fwupdtool` built locally from sources on d15eab455e288e84603ca77d9f267c4fa58d571f.

**Additional details**
The `fwupd` regression seems introduced by the following commit c4ca02629487d3f26278f47e4720167b1ae04eb4.
So that `fwupd` versions since `1.7.5` seems affected.

MarekPieta, 2023-04-21:
Issue is fixed by the following PR: https://github.com/fwupd/fwupd/pull/5699. The PR is already merged.

We still need to introduce a way to bring back reporting VID and PID of `0x0000` for devices that do not report their VID and PID over configuration channel (devices with legacy firmware would not support the new functionality).

hughsie, 2023-05-09:
@MarekPieta okay to close this now?

MarekPieta, 2023-05-09:
Yes, since https://github.com/fwupd/fwupd/pull/5742 is merged I think we can close this one.
Thanks a lot for help with fixing this one

closed 2023-05-09 by MarekPieta