fwupd/fwupd #5740: Either delay daemon startup for retimers, or provide a client –wait

hughsie, 2023-04-21:
Most devices in fwupd are added during engine coldplug, and are present when the `org.freedesktop.fwupd` name appears on the system D-Bus. Any client that autostarted the daemon is able to call `GetDevices` or something like `fwupdmgr install foo.cab` without any delay and have the update applied. There are at least two classes of devices where the device or sub-device is added asynchronously from the initial coldplug. Bluetooth and thunderbolt retimer devices are all “missing” when the daemon appears on the bus, but appear anything from a few ms or several seconds later.

This means [_I think_] that if you do `fwupdmgr install retimer.cab` when the daemon is not already running, then fwupdmgr will block, the daemon will start, announce itself on the bus, `fwupdmgr` does `Install(*)` where **nothing matches**, and then `fwupdmgr` fails with an error. It’s not enough to put a “sleep 5” before the fwupdmgr call, unless you also do something like:

    fwupdmgr –version # to force the daemon to start
    sleep 5 # wait for async devices
    fwupdmgr install retimer.cab

So I guess we have a few ways we can fix this:

1. we allow the fwupd plugin to say to the engine “wait X seconds extra before claiming the bus name” and then all the devices are available using `GetDevices` as if they were added synchronously. This drawback is that we delay the fwupd startup for quite some time in the worst case (BT device not replying, which only times out after ~10 seconds) but the good bit is that the clients keeps working as before with no changes required.

2. we allow the plugin to say “wait for device XX to be added before claiming the bus name” which is slightly more complex as we need to tell the daemon *not* to do that if adding the device failed for some reason.

3. add a `--wait` command to `fwupdmgr` which tells the front-end to poll GetDevices for a number of seconds until the device shows up async. This isn’t ideal as it means changing the front-end clients – e.g. also gnome-software etc. Also not ideal as the first thing most clients do when they see fwupd appear on the bus is GetReleases to look for updates, which also isn’t going to work for the async devices.

4. just unconditionally wait say 5 seconds polling GetDevices if we do `fwupdmgr install file GUID` or `fwupdmgr update GUID` on the logic that if we’re specifying the GUID we really know what hardware we want to match. I think in most cases fwupd is probably started quite early in desktops and servers and ChromeOS early-boot firmware mode is the unusual case here.

I’m erring on 1 or possibly 2. Comments welcome. @superm1

superm1, 2023-04-21:
IMO Inherently this is because the client tries to do the update so aggressively.  It sees a daemon show up so it figures that it must be able to apply updates immediately.

IMO the knowledge of a retimer or a bluetooth device come and go and so you might need to try is something that such an aggressive client should handle on it's own.

As a counter-proposal, how about we have a specific new return code for this situation?  We have `2` to mean nothing to do, but what if we introduce `19` for `-ENODEV`?  Then clients can match that and know they called when the device wasn't present and they should try again later.

Sorry; but all the other ideas feel like hacks to me.

hughsie, 2023-04-23:
> how about we have a specific new return code for this situation

I'd be fine with that; @djcampello might need to do some funky shell loop in that case. To make his life easier, what about if we provided something like `fwupdmgr wait-device GUID` that can be used to wait for the device to showup in CLI scripts?

superm1, 2023-04-24:
Yes that's great