fwupd/fwupd #5741: ThinkPad T480 Thunderbolt controller not found

xvitaly, 2023-04-21:
**Describe the bug**
Trying to install T480 Thunderbolt controller update, but fwupd fails.

**Steps to Reproduce**
```
$ fwupdmgr get-upgrades
Devices with no available firmware updates: 
 • MZVLW256HEHP-000L7
 • TPM
 • Thunderbolt Controller
 • Thunderbolt Controller
 • Thunderbolt Controller
 • Thunderbolt host controller
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • System Firmware
```

```
$ wget https://fwupd.org/downloads/391c6be40422dc1d253f5a6b9bdf0c122ddc5f61f9fcb3ba4e61cf42a7c6f7bc-Lenovo-ThinkPad-T480-SystemFirmware-23.cab
$ fwupdmgr install ~/391c6be40422dc1d253f5a6b9bdf0c122ddc5f61f9fcb3ba4e61cf42a7c6f7bc-Lenovo-ThinkPad-T480-SystemFirmware-23.cab
Decompressing…           [ -                                     ]
No supported devices found
```

**Expected behavior**
The update should be installed.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.14
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.14
runtime   org.kernel                    6.2.10-200.fc37.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Installed from Fedora 37 repository.

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
LENOVO 20L50007RT
│
├─Core™ i7-8550U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
[... 252 lines cut ...]
├── initramfs-0-rescue-d81e28cb17d24c1e9176a83c32f34d23.img
├── initramfs-6.2.10-200.fc37.x86_64.img
├── initramfs-6.2.9-200.fc37.x86_64.img
├── loader
│   └── entries  [error opening dir]
├── lost+found  [error opening dir]
├── symvers-6.2.10-200.fc37.x86_64.gz -> /lib/modules/6.2.10-200.fc37.x86_64/symvers.gz
├── symvers-6.2.9-200.fc37.x86_64.gz -> /lib/modules/6.2.9-200.fc37.x86_64/symvers.gz
├── System.map-6.2.10-200.fc37.x86_64
├── System.map-6.2.9-200.fc37.x86_64
├── vmlinuz-0-rescue-d81e28cb17d24c1e9176a83c32f34d23
├── vmlinuz-6.2.10-200.fc37.x86_64
└── vmlinuz-6.2.9-200.fc37.x86_64

6 directories, 12 files
```

**Additional questions**

- Operating system and version: Fedora 37
- Have you tried rebooting? Yes
- Is this a regression? Maybe
- Are you using an NVMe disk? yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2023-04-21:
> Device requires AC power to be connected

Did you try on AC power?

Also the firmware you downloaded is targetting fe328c2d-5300-597b-a21d-991116db8ab6 which is a different GUID than either of your controllers. I don't think that's the correct file.

hughsie, 2023-04-21:
Unfortunately there are lots of t480 variants. Do you have N24TF23W?

xvitaly, 2023-04-21:
> Unfortunately there are lots of t480 variants. Do you have N24TF23W?

T480 (Type 20L5, 20L6) Laptop (ThinkPad) - Type 20L5. Required firmware update: N24TF23W.

From Lenovo support website:
[image]