fwupd/fwupd #5745: How can I debug `FuPluginIntelME failed to get public key ... generic failure [0xb]` error?

balacij, 2023-04-21:
**Describe the question**
When I run `sudo fwupdtool security`, I get the following error (I attached the `--full --verbose` version at the bottom):

```shell
sudo fwupdtool security                             
Loading…                 [*******************                    ]22:25:12.513 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]
Loading…                 [************************************** ]

Not enough data was provided to make an HSI calculation.
 » https://fwupd.github.io/hsi.html#not-enough-info
To ignore this warning, use --force
```

I suspect the reason why I'm not receiving the HSI calculation is because of that `FuPluginIntelMe`-related error (am I even right about this? I'm not sure). However, I am not quite sure how I should debug this. I've confirmed that `mei0` is loaded:

```shell
ls -l /dev/mei*
crw------- 1 root root 244, 0 Apr 21 18:04 /dev/mei0
```

and using coreboot's `intelmetool`:

<details>
<summary>`sudo ./intelmetool -m`</summary>

```shell
sudo ./intelmetool -m  
MEI found: [8086:a360] Cannon Lake PCH HECI Controller

ME Status   : 0x94000245
ME Status 2 : 0x9f58506

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Normal
ME: Current Operation State : M0 with UMA
ME: Current Operation Mode  : Normal
ME: Error Code              : No Error
ME: Progress Phase          : ROM Phase
ME: Power Management Event  : Non-power cycle reset
ME: Progress Phase State    : (null)

ME: Extend Register not valid

ME: Firmware Version 12.0.2145.92 (code) 12.0.2145.92 (recovery) 12.0.1420.32 (fitc)

ME Capability: Full Network manageability                 : OFF
ME Capability: Regular Network manageability              : OFF
ME Capability: Manageability                              : OFF
ME Capability: Small business technology                  : OFF
ME Capability: Level III manageability                    : OFF
ME Capability: IntelR Anti-Theft (AT)                     : OFF
ME Capability: IntelR Capability Licensing Service (CLS)  : ON
ME Capability: IntelR Power Sharing Technology (MPC)      : OFF
ME Capability: ICC Over Clocking                          : OFF
ME Capability: Protected Audio Video Path (PAVP)          : ON
ME Capability: IPV6                                       : OFF
ME Capability: KVM Remote Control (KVM)                   : OFF
ME Capability: Outbreak Containment Heuristic (OCH)       : OFF
ME Capability: Virtual LAN (VLAN)                         : ON
ME Capability: TLS                                        : ON
ME Capability: Wireless LAN (WLAN)                        : OFF
```
</details>

For some extra information, I am using NixOS on a Dell XPS 7590 with an Intel i7-9750H CPU. Please let me know if any other information is relevant and needed.
[... 510 lines cut ...]
✘ BIOS rollback protection:      [31m[1mDisabled[0m

[1mHSI-3[0m
✔ Intel BootGuard error policy:  [32m[1mValid[0m
✔ Pre-boot DMA protection:       [32m[1mEnabled[0m
✔ Suspend-to-idle:               [32m[1mEnabled[0m
✘ Intel CET Enabled:             [31m[1mNot supported[0m
✘ Suspend-to-ram:                [31m[1mEnabled[0m

[1mHSI-4[0m
✔ Intel SMAP:                    [32m[1mEnabled[0m
✘ Encrypted RAM:                 [31m[1mNot supported[0m

[1mRuntime Suffix -![0m
✔ fwupd plugins:                 [32m[1mUntainted[0m
✘ Linux kernel:                  [31m[1mTainted[0m
✘ Linux kernel lockdown:         [31m[1mFailed[0m
✘ Linux swap:                    [31m[1mUnencrypted[0m

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

```

</details>

hughsie, 2023-04-24:
> I suspect the reason why I'm not receiving the HSI calculation is because of that FuPluginIntelMe-related error

I don't think so; it's probably because  your CPU is unsupported for some reason. What does `sudo fwupdtool security --force` say?

> failed to get public key using /fpf/OemCred

Unfortunately this is the only way I know of getting the public key hash for legacy MKHI ME interfaces. Intel hasn't give me any more details unfortunately.

balacij, 2023-04-24:
Ah, that's interesting. I wasn't expecting that. Thank you for responding so quickly! :smile: 

The output of `sudo fwupdtool security --force` is:
```console
Loading…                 [*******************                    ]15:17:14.518 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]
Loading…                 [************************************** ]
Host Security ID: HSI:INVALID:missing-data

HSI-1
✔ CSME manufacturing mode:       Locked
✔ CSME override:                 Locked
✔ CSME v0:12.0.92.2145:          Valid
✔ Platform debugging:            Disabled
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ IOMMU:                         Enabled
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid
✘ BIOS rollback protection:      Disabled

HSI-3
✔ Intel BootGuard error policy:  Valid
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✘ Intel CET Enabled:             Not supported
✘ Suspend-to-ram:                Enabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✘ Linux kernel:                  Tainted
✘ Linux kernel lockdown:         Failed
✘ Linux swap:                    Unencrypted

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

```

hughsie, 2023-04-24:
> Host Security ID: HSI:INVALID:missing-data

Okay, weird. Can you **attach** the output of `sudo fwupdtool security --force --verbose --verbose` please.

Richard.

hughsie, 2023-04-24:
100% okay with me. It says the org.fwupd.hsi.Kernel.Lockdown plugin is missing data. What does `cat /sys/kernel/security/lockdown` say?

balacij, 2023-04-24:
```console
λ ~/ cat /sys/kernel/security/lockdown
cat: /sys/kernel/security/lockdown: No such file or directory
λ ~/ sudo cat /sys/kernel/security/lockdown                           
cat: /sys/kernel/security/lockdown: No such file or directory
```

Odd...

balacij, 2023-04-24:
Ah! Thank you again :smile: Have a great day!

superm1, 2023-04-24:
`SECURITY_LOCKDOWN_LSM`

flintflump, 2023-05-11:
Just a small question: It sounds intended that the check fails with missing-data if kernel lockdown is disabled via `SECURITY_LOCKDOWN_LSM`. Is there a reason that it isn't just reported as disabled?

not-a-dev-stein, 2023-09-14:
Sorry for reviving this issue, but I've been having the same issue on my machine, a Dell Inspiron 7567 running the latest 1.15.0 BIOS and Fedora Silverblue, ever since F37 all the way to now F39 (which I originally believed to just be down to DELL not updating the machine anymore, but today I decided to figure it out). Tried following the issue here to see if I managed to debug it and I'm even more confused. Everything looks the same until running `cat /sys/kernel/security/lockdown`, from which I get what appears to be normal results:

```
$ cat /sys/kernel/security/lockdown 
none [integrity] confidentiality
```
But I get the same issues when trying to run `sudo fwupdtool security --force` (it's all in portuguese, so if needed I'll translate):

```
$ sudo fwupdtool security --force
Carregando…              [********************                   ]20:14:56.556 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]
Carregando…              [************************************** ]
ID de segurança do host: HSI:0! (v1.9.5)

HSI-1
✔ Bloqueio de SPI:               Habilitado
✔ CPU suportada:                 Válido
✔ Chave de plataforma UEFI:      Válido
✔ Depuração da plataforma:       Desabilitado
✔ Escrita de SPI:                Desabilitado
✔ Modo de fabricação de csme:    Bloqueado
✔ PCRs vazios de TMP:            Válido
✔ Região BIOS do SPI:            Bloqueado
✔ Secure boot de UEFI:           Habilitado
✔ TPM v2.0:                      Encontrado
✔ Variáveis do serviço de arranque UEFI:Bloqueado
✔ substituição de csme:          Bloqueado
✘ csme v0:11.8.86.3877:          Inválido

HSI-2
✔ ACM protegido com Intel BootGuard:Válido
✔ Depuração da plataforma:       Bloqueado
✔ Fuse programável (OTP) do Intel BootGuard:Válido
✔ IOMMU:                         Habilitado
✔ Inicialização verificada com Intel BootGuard:Válido
✔ Intel BootGuard:               Habilitado
✔ Intel GDS mitigation:          Habilitado
✔ Reconstrução de PCR0 de TPM:   Válido

HSI-3
✔ Política de erro do Intel BootGuard:Válido
✘ Habilitado para Intel CET:     Não suportado
✘ Proteção de DMA pré-inicialização:Desabilitado
✘ Suspensão para RAM:            Habilitado
✘ Suspensão para inativo:        Desabilitado

HSI-4
✔ Intel SMAP:                    Habilitado
✘ RAM criptografada:             Não suportado

Sufixo de tempo de execução -!
✔ Lockdown do kernel Linux:      Habilitado
✔ Plugins do fwupd:              Descontaminado
✔ Swap do Linux:                 Criptografado
✘ Kernel Linux:                  Contaminado

Este sistema possui um nível de segurança de HSI baixo.
 » https://fwupd.github.io/hsi.html#low-security-level

Este sistema possui problemas de tempo de execução de HSI.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix
```

hughsie, 2023-09-15:
> Carregando…              [********************                   ]20:14:56.556 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]

This is the MKHI device not requesting the correct thing. Could you follow the instructions in https://blogs.gnome.org/hughsie/2023/05/09/msi-and-insecure-kms/ and get me the strace output please?

not-a-dev-stein, 2023-09-15:
Tried basically all of the CSME Tools options, and none of them worked on my Kaby Lake hardware (some of them didn't work with Secure Boot on and only gave me the unsupported hardware error after turning it off, some of them didn't work at all, and the ME System Tools only have Windows versions). Basically all of them except the v11 r46 version gave me the unsupported hardware platform error. And the error the v11 r46 gives me is this:

```

$ sudo ./CSME\ System\ Tools\ v11\ r46/MEInfo/LINUX64/MEInfo 

Intel(R) MEInfo Version: 11.8.92.4222
Copyright(C) 2005 - 2020, Intel Corporation. All rights reserved.

Driverless mode

Error 79: Fail to read FW Status Register value 

Unable to clean up MEInfo before exiting.
```

It might be a Silverblue limitation, but I'm not 100% sure.

hughsie, 2023-09-19:
Hmm, that's all I could think about on how to debug this. @CragW any idea what MEI tool should be used for a Dell XPS 7590 (Kaby Lake)? Thanks.

not-a-dev-stein, 2023-12-16:
> Tried basically all of the CSME Tools options, and none of them worked on my Kaby Lake hardware (some of them didn't work with Secure Boot on and only gave me the unsupported hardware error after [...]

A small update after all these months, as I got a teeny tiny small amount of progress. In the end the version that worked for my hardware really was v11 r46, but I only got it to work with the Windows version (which means I'm still unable to `strace` it). Not sure if this helps even a tiny amount in all of this, but at least here's the first of the two files you ask for in the blog post:

[not-a-dev-steiner-meinfo.txt](https://github.com/fwupd/fwupd/files/13694202/not-a-dev-steiner-meinfo.txt)

I still have no idea why I get this error on the Linux version, as I've tried it with pretty much everything I could think could affect this (secure boot on & off, SELinux on & off).

hughsie, 2023-12-18:
> I still have no idea why I get this error on the Linux version

Me either -- and I appreciate you getting the log from Windows and trying all the permutations. @CragW is there anything you can share about what we're supposed to do?