fwupd/fwupd #5747: Crash in fu_mei_hfsts1_to_string

bugaevc, 2023-04-22:
**Describe the bug**

```
Thread 1 "fwupd" received signal SIGSEGV, Segmentation fault.
0x00007ffff766a597 in __strchr_sse2 () from /lib64/libc.so.6
(gdb) bt
#0  0x00007ffff766a597 in __strchr_sse2 () at /lib64/libc.so.6
#1  0x00007ffff781d725 in g_strsplit () at /lib64/libglib-2.0.so.0
#2  0x00007ffff7ebad3b in fu_string_append
    (str=0x5555555884a0, idt=2, key=0x7ffff7db2bab "WorkingState", value=0x300000001 <error: Cannot access memory at address 0x300000001>)
    at ../libfwupdplugin/fu-string.c:336
#3  0x00007ffff7d1e538 in fu_mei_hfsts1_to_string (hfsts1=..., idt=2, str=0x5555555884a0) at ../plugins/pci-mei/fu-mei-common.c:209
#4  0x00007ffff7d1ef0d in fu_pci_mei_plugin_to_string (plugin=0x555555629860, idt=1, str=0x5555555884a0) at ../plugins/pci-mei/fu-pci-mei-plugin.c:40
#5  0x00007ffff7eebb59 in fu_plugin_add_string (self=0x555555629860, idt=0, str=0x5555555884a0) at ../libfwupdplugin/fu-plugin.c:414
#6  0x00007ffff7c878bc in fu_engine_load (self=0x55555559c4e0, flags=<optimized out>, flags@entry=78, progress=0x555555599d10, error=error@entry=0x7fffffffe208)
    at ../src/fu-engine.c:8543
#7  0x0000555555561b57 in fu_daemon_setup (self=self@entry=0x555555594280, socket_address=socket_address@entry=0x0, error=error@entry=0x7fffffffe208)
    at ../src/fu-daemon.c:2234
#8  0x000055555555b899 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:172
```

```
(gdb) p hfsts1 
$1 = {data = 2709540957, fields = {working_state = 13, mfg_mode = 1, fpt_bad = 0, operation_state = 1, fw_init_complete = 0, ft_bup_ld_flr = 0, update_in_progress = 0, 
    error_code = 5, operation_mode = 0, reset_count = 8, boot_options_present = 1, bist_finished = 0, bist_test_state = 0, bist_reset_request = 0, 
    current_power_source = 2, d3_support_valid = 0, d0i3_support_valid = 1}}
(gdb) l
203	    [ME_HFS_ERROR_DEBUG] = "debug-failure",
204	};
205	
206	void
207	fu_mei_hfsts1_to_string(FuMeiHfsts1 hfsts1, guint idt, GString *str)
208	{
209		fu_string_append(str, idt, "WorkingState", me_cws_values[hfsts1.fields.working_state]);
210		fu_string_append_kb(str, idt, "MfgMode", hfsts1.fields.mfg_mode);
211		fu_string_append_kb(str, idt, "FptBad", hfsts1.fields.fpt_bad);
212		fu_string_append(str,
(gdb) p me_cws_values
$2 = {0x7ffff7d993d5 "reset", 0x7ffff7db30aa "initializing", 0x7ffff7dc0fee "recovery", 0x7ffff7dbfa45 "test", 0x7ffff7db27c3 "disabled", 0x7ffff7db2ff4 "normal", 
  0x7ffff7db30b7 "wait", 0x7ffff7db30bc "transition", 0x7ffff7db6dcd "invalid"}
(gdb) bt full
#0  fu_mei_hfsts1_to_string (hfsts1=..., idt=2, str=0x5555555884a0) at ../plugins/pci-mei/fu-mei-common.c:208
#1  0x00007ffff7d1ef0d in fu_pci_mei_plugin_to_string (plugin=0x555555629860, idt=1, str=0x5555555884a0) at ../plugins/pci-mei/fu-pci-mei-plugin.c:40
        self = 0x555555629860
#2  0x00007ffff7eebb59 in fu_plugin_add_string (self=0x555555629860, idt=0, str=0x5555555884a0) at ../libfwupdplugin/fu-plugin.c:414
        priv = 0x5555556295f0
        vfuncs = <optimized out>
        name = <optimized out>
        flags = <optimized out>
        __func__ = "fu_plugin_add_string"
#3  0x00007ffff7c878bc in fu_engine_load (self=0x55555559c4e0, flags=<optimized out>, flags@entry=78, progress=0x555555599d10, error=error@entry=0x7fffffffe208)
    at ../src/fu-engine.c:8543
        plugin = 0x555555629860
        i = 57
        plugin_uefi = <optimized out>
        quirks_flags = <optimized out>
        plugins = 0x5555555a05c0
        host_emulate = <optimized out>
        checksums_approved = 0x5555555aee70
        checksums_blocked = 0x5555555ded00
        error_quirks = 0x0
        error_json_devices = 0x0
        error_local = 0x0
        str = 0x5555555884a0
        __func__ = "fu_engine_load"
#4  0x0000555555561b57 in fu_daemon_setup (self=self@entry=0x555555594280, socket_address=socket_address@entry=0x0, error=error@entry=0x7fffffffe208)
    at ../src/fu-daemon.c:2234
        machine_kind = 0x0
        progress = 0x555555599c90
        __func__ = "fu_daemon_setup"
#5  0x000055555555b899 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:172
        immediate_exit = 0
        timed_exit = 0
        socket_filename = <optimized out>
        options = {{long_name = 0x55555556285a "timed-exit", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7fffffffe200, description = 0x55555556282f "Exit after a small delay", arg_description = 0x0}, {long_name = 0x555555562865 "immediate-exit", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7fffffffe204, description = 0x555555562e18 "Exit after the engine has loaded", arg_description = 0x0}, {long_name = 0x0, short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x0, description = 0x0, arg_description = 0x0}}
        socket_address = 0x0
        error = 0x0
        argv0_file = 0x555555594040
        context = 0x555555597ec0
        daemon = 0x555555594280
        argv0_monitor = 0x0
        memory_monitor = 0x0
```

**Steps to Reproduce**

`sudo fwupd`, or just open Device Security in g-c-c.

**Expected behavior**
No crash.

**fwupd version information**
_Please provide the version of the daemon and client._

```shell
$ fwupdmgr --version
Failed to connect to daemon: GDBus.Error:org.freedesktop.DBus.Error.NameHasNoOwner: Could not activate remote peer: startup job failed.
```
:see_no_evil:

But really, this is 1.8.14 from fwupd-1.8.14-1.fc38

_Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):_ DNF

_Please provide the output of the fwupd devices recognized in your system._

```shell
$ fwupdmgr get-devices --show-all-devices
Failed to connect to daemon: GDBus.Error:org.freedesktop.DBus.Error.NameHasNoOwner: Could not activate remote peer: startup job failed.
```

(Of course!) Please let me know if there's any other way to give you the needed info without going through fwupd. The machine is Microsoft Surface Pro 6.

**Additional questions**

- _Operating system and version:_
    - F38
- _Have you tried rebooting?_
  - Yes (but it's not like rebooting can fix out-of-bounds array access...)
- _Is this a regression?_
    - Yes

hughsie, 2023-04-22:
Ohh uncool. What's the value of `hfsts1.fields.working_state` out of interest? Patch incoming in 5 mins.

bugaevc, 2023-04-22:
13. See my `p hfsts1` above.

hughsie, 2023-04-22:
We've never seen that state before. Can you share any details about your system that might be salient?

bugaevc, 2023-04-22:
I'm happy to provide any details if you tell me how to get them (such as maybe read some value from `/sys`?) -- unfortunately I'm rather clueless about hardware myself. My software is a regular F38 install with https://github.com/linux-surface/linux-surface stuff, in particular with their kernel build, if that's important. fwupd used to work beautifully until I updated to F38.

hughsie, 2023-04-22:
> The machine is Microsoft Surface Pro 6

Ah, I missed that bit. I suspect Microsoft is setting up the ME in an undocumented way. I'll ask my experts.

closed 2023-04-24 by hughsie

bugaevc, 2023-04-24:
Thank you! Will the hotfix make its way into F38 some time soon? I don't see any new `fwupd` versions on bodhi.

bugaevc, 2023-04-24:
That worked, thank you! On `1.9.1-0.1001.20230424git.fc38`, I'm only getting `FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]`, and a generic red "Checks Failed" / "Hardware does not pass checks." in Device Security, but no crash.