fwupd/fwupd #5748: Updating does not work - "Capsules not found"

Duvio, 2023-04-22:
When running the following:
```
$ sudo fwupdmgr update
Devices with no available firmware updates:
 • CT500MX500SSD1
 • UEFI dbx
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Embedded Controller from 0.1.35 to 0.1.36?                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Fixed Type-C Travel hub attached annoying PC isn't charging message.       ║
║                                                                              ║
║ 20JNS0CN00 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Update failure is a known issue, visit this URL for more information: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot
Successfully uploaded 1 report
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.80.3746 to 184.93.4323?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Updated includes a security fix.                                           ║
║                                                                              ║
║ 20JNS0CN00 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
Updating Intel Management Engine…                                ]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.70 to 0.1.73?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Update includes a security fix.                                            ║
║ • Enhancement to address security vulnerability.                             ║
║ • Updated the Diagnostics module to version 04.28.000.                       ║
║                                                                              ║
║ 20JNS0CN00 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
Updating System Firmware…[  -                                    ]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
An update requires a reboot to complete. Restart now? [y|N]: y
```
I get this error after rebooting:
```
fwupd-efi version 1.4
WARNING: QueryCapsuleCapabilities failed, assuming EfiResetWarm: Unsupported
WARNING: Could not apply capsule update: Not Found
WARNING: Could not apply capsules: Not Found
Reset System```
This should obviously not happen. The new versions are not installed.

```
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12                    compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
[... 96 lines cut ...]
  │
  ├─UEFI Platform Key:
  │     Device ID:        6924110cde4fa051bfdc600a60620dc7aa9d3c6a
  │     Summary:          Lenovo Ltd. PK CA 2012
  │     Vendor:           Lenovo Ltd.
  │     GUID:             71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E
  │
  └─UEFI dbx:
        Device ID:        362301da643102b9f38477387e2193e57abaa590
        Summary:          UEFI revocation database
        Current version:  83
        Minimum Version:  83
        Vendor:           UEFI:Linux Foundation
        Install Duration: 1 second
        GUIDs:            14503b3d-73ce-5d06-8137-77c68972a341 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649
                          5971a208-da00-5fce-b5f5-1234342f9cf7 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649&ARCH_X64
                          c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
        Device Flags:     • Internal device
                          • Updatable
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Only version upgrades are allowed
                          • Signed Payload
```

hughsie, 2023-04-22:
Can you try installing just the system firmware first? If that works then we need to add some additional requirements.

Duvio, 2023-04-23:
> Can you try installing just the system firmware first? If that works then we need to add some additional requirements.

Which one would that be? Which command do i have to run?

hughsie, 2023-04-23:
Use `fwupdmgr update a45df35ac0e948ee180fe216a5f703f32dda163f` to specify the exact device ID. If that works, yell, and I'll get Lenovo to update the requirements on the other firmware.

Duvio, 2023-04-23:
> Use `fwupdmgr update a45df35ac0e948ee180fe216a5f703f32dda163f` to specify the exact device ID. If that works, yell, and I'll get Lenovo to update the requirements on the other firmware.

Did that. Still the same error.

hughsie, 2023-04-23:
Would it be possible to try with Fedora? Just to rule out it being Debian specific?

Duvio, 2023-04-23:
I would rather not install another OS. But can i try with a Fedora live USB?

hughsie, 2023-04-24:
> But can i try with a Fedora live USB

Yes, I think so!

Duvio, 2023-04-24:
When decompressing it says that there is not enough space on /run/media/root/ANACONDA. I tried resolving it but failed.

Any other way where i do not have to install another OS?

mrhpearson, 2023-04-25:
Just to check - but this is a T470 right? 
Looking at the LVFS stats it's had 40k+ downloads so I'm wondering if this case is as basic as running out of EFI space? Can you check how much free space you have in the EFI partition - I've seen a few cases recently that seem related to that
(hughsie - let me know if I'm talking nonsense :P)

hughsie, 2023-04-25:
> as basic as running out of EFI space

This is 1.8.12 and the ESP file space check was added in 1.8.11 so I don't think it's that.

Duvio, 2023-04-25:
Yes, T470. Sadly it's not the available space. I gave the partition 500 MB and only 15 MB are used.

Duvio, 2023-05-22:
_**Bump**_

mrhpearson, 2023-06-14:
Yeah - let's do a new ticket so I can flag that to the FW team separately without things getting confusing. It might end up being the same root cause but it is a different FW team. 
Please tag me (then it goes right into my main inbox rather than getting filtered to my 'to review' folder)

Thanks!
Mark

thwalker3, 2023-06-14:
No problem.  Broken off into https://github.com/fwupd/fwupd/issues/5906

thwalker3, 2023-06-15:
Doh, now I feel like an idiot.  Somehow I managed to set up the disk on the new laptop (not a fresh install) without ESP set.  I'm actually kind of surprised it booted, but I guess direct EFI boot entries will do that.

As soon as I flipped the flag, things work and both UEFI and ME updated without a problem.

Sorry for the noise...

superm1, 2023-06-15:
I wonder if perhaps no partitions have esp flag set we should just set the one that was selected to flash from as esp as part of the flash process.

hughsie, 2023-06-15:
@superm1 my worry with doing that is that it's going to *freak out* forensic and monitoring agents. I'm also worried it'll change the disk in some way to not allow booting -- e.g. for an OS that only supports BDP and not ESP.

hughsie, 2023-06-15:
> parted /dev/nvme0n1 set 1 esp on

Does anyone know the programmatic way to do this? i.e. without calling `system("parted /dev/nvme0n1 set 1 esp on")`

superm1, 2023-06-15:
I guess https://www.gnu.org/software/parted/api/group__PedPartition.html#gf9ccfe496f0f6fe3ce34cd5ce4215b85 is one option

Duvio, 2023-06-15:
> Just to update here too - internal ticket to the FW team is LO-2516. Please also check that your EFI partition is
> 
>     * GPT formatted
> 
>     * has the esp flag set (see [ThinkPad P14s Gen 2 AMD: failed to run update on reboot #5612](https://github.com/fwupd/fwupd/issues/5612) - you need to run 'parted /dev/nvme0n1 set 1 esp on')
>       Just to rule those out.

It has the esp flag set and is GPT formatted.

mrhpearson, 2023-06-16:
Darn - I was hoping your issue was the same.
Checking with the FW team.

Duvio, 2023-08-16:
I tried again and now it somehow worked.

closed 2023-08-16 by Duvio

sharkcz, 2024-10-31:
I have a pretty similar experience trying to update Lenovo T14 Gen4. `Could not apply capsule update: Not Found` many times yesterday and all worked today on a first run. The only change I can see is that yesterday reboots were all with USB-C power supply connected. And before today the laptop spent time powered off without the power supply connected.