fwupd/fwupd #5760: "failed to read flash" error while flashing TUXEDO InfinityBook S15 Gen6 BIOS

Matombo, 2023-04-25:
**Describe the bug**
When trying to flash the TUXEDO InfinityBook S15 Gen6 BIOS using the local-install comand I get the error "failed to back up original firmware: failed to read flash [2]". Looking into the sourcecode of fwupd and flashrom the [2] is the return code of flashroms "flashrom_image_read" and means that the provided buffer size it not big enough to hold the rom image. Inserting a printf here shows that the BIOS flash chip ist 16MB while the buffer passed by fwupd is only 11MB. Finding out from where fwupd gets the 11MB size gets a little bit convoluted for someone not familiar with the codebase, thats why I'm opening this report.

**Steps to Reproduce**
Run `fwupdmgr local-install <*.cab file> <device id of BIOS chip>` on a TUXEDO InfinityBook S15 Gen6 BIOS

**Expected behavior**
The BIOS should be flashed without trowing errors and aborting.

**fwupd version information**

```shell
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.12
runtime   org.kernel                    6.2.0-20-generic
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

apt (ubuntu 23.04, installed while still beta, but updated since)

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
Notebook NS50MU
│
├─Unknown Device:
│     Device ID:          63223cc2d1b0b8d4c45ef8cedcb1f45315d3ff20
│     GUID:               ca5c154c-a46a-59d8-bf77-0ca40522cc85 ← GPIO\ID_INT34C5:00
│   
├─11th Gen Intel Core™ i7-1165G7 @ 2.80GHz:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x000000a6
│ │   Vendor:             Intel
│ │   GUID:               b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│ │                       2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│ │                       1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ └─USB4 host controller:
│       Device ID:        6d73fda73ce536d6dba904c9cef94fc097207a47
│       Summary:          Unmatched performance for high-speed I/O
│       GUID:             e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│       Device Flags:     • Internal device
│                         • System requires external power source
│                         • Signed Payload
│     
├─NS50MU (bios):
│ │   Device ID:          dbee8bd3b1ae0316ad143336155651eedb495a0e
│ │   Current version:    1.07
│ │   Vendor:             Notebook (DMI:INSYDE Corp.)
│ │   Update State:       Failed
│ │   Problems:           [31m[1m• An update is in progress[0m
│ │   Last modified:      2023-04-25 13:03
│ │   GUID:               230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │                       7e036735-f3d3-5fcb-af3a-f3b1356f6115 ← FLASHROM\VENDOR_Notebook&PRODUCT_NS50MU&REGION_BIOS
│ │                       f07dce59-cca7-51a7-8ca9-7d7d0358dd0e ← FLASHROM\GUID_60f53465-e8fc-5122-b79b-f7b03f063037
│ │                       dc0de287-201b-5240-b8d2-0e5cead707ea ← Notebook&Not Applicable&NS50MU&Not Applicable&Notebook&NS50MU
│ │                       8da9fe2b-90f4-5028-9bd9-c692b107d4ea ← Notebook&Not Applicable&NS50MU&Not Applicable
│ │                       0039fec6-9ecf-5921-9961-905e148e052a ← Notebook&Not Applicable&NS50MU
│ │                       6b26127f-384f-5ecc-9e94-02359da05f54 ← Notebook&Not Applicable&Notebook&NS50MU
[... 124 lines cut ...]
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          d96de5c124b60ed6241ebcb6bb2c839cb5580786
      Summary:            UEFI ESRT device
      Current version:    213
      Vendor:             DMI:INSYDE Corp.
      Update State:       Success
      GUID:               832af090-2ef9-7c47-8f6d-b405c8c7f156
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

</details>

**Additional questions**

- Operating system and version: ubuntu 23.04, installed while still beta, but updated since
- Have you tried rebooting? yes
- Is this a regression? don't know, only tried version 1.8.12

hughsie, 2023-04-25:
what's `sudo fwupdtool --verbose --verbose get-devices --plugins flashrom` say?

Matombo, 2023-04-25:
The "An update is in progress" message for the BIOS chip in the get-devices output is there after the flash faild until I reboot the device.

Matombo, 2023-04-25:
> what's `sudo fwupdtool --verbose --verbose get-devices --plugins flashrom` say?

```
$ sudo fwupdtool --verbose --verbose get-devices --plugins flashrom > out
13:27:59.884 FuDebug              Verbose debugging enabled (on console 1)
13:27:59.884 FuMain               locked /run/lock/fwupdtool
13:27:59.884 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
13:27:59.886 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
13:27:59.886 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
13:27:59.886 FuCommon             reading /etc/fwupd/daemon.conf with 2564 bytes
13:27:59.887 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
13:27:59.887 FuConfig             using autodetected max archive size 2,0 GB
13:27:59.887 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
13:27:59.887 FuRemoteList         loading /usr/share/fwupd/metainfo
13:27:59.887 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
13:27:59.887 XbSilo               file: 78b59426-0680-7db2-f42e-c5343f416b05, current:78b59426-0680-7db2-f42e-c5343f416b05, cached: (null)
13:27:59.888 XbSilo               loading silo with file contents
13:27:59.888 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
13:27:59.888 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
13:27:59.888 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
13:27:59.888 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
13:27:59.888 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
13:27:59.888 FuRemoteList         ordering lvfs-testing=lvfs+1
13:27:59.888 FuRemoteList         ignoring unfound remote fwupd
wird geladen …: 1%
13:27:59.888 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
13:27:59.888 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
13:27:59.888 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
13:27:59.888 Jcat                 loaded 1 certificates
13:27:59.888 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
13:27:59.888 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
13:27:59.888 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
13:27:59.888 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
13:27:59.888 Jcat                 loaded 1 certificates
13:27:59.888 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
13:27:59.888 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
13:27:59.888 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
13:27:59.888 FuEngine             client certificate now exists: nothing to do
wird geladen …: 2%
13:27:59.888 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
13:27:59.888 FuHistory            got schema version of 8
wird geladen …: 3%
wird geladen …: 4%
13:27:59.889 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-1.8.12/libfu_plugin_modem_manager.so)
13:27:59.894 FuContext            added udev subsystem watch of platform
13:27:59.894 FuContext            added udev subsystem watch of pci
13:27:59.894 FuContext            added udev subsystem watch of block
13:27:59.894 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
13:27:59.894 FuPluginDell         Using libsmbios 2.4
13:27:59.894 FuContext            added udev subsystem watch of i2c
13:27:59.894 FuContext            added udev subsystem watch of i2c-dev
13:27:59.894 FuContext            added udev subsystem watch of hidraw
13:27:59.894 FuContext            added udev subsystem watch of gpio
13:27:59.894 FuContext            added udev subsystem watch of mei
13:27:59.894 FuContext            added udev subsystem watch of iommu
13:27:59.894 FuContext            added udev subsystem watch of video4linux
13:27:59.895 FuContext            added udev subsystem watch of msr
13:27:59.895 FuContext            added udev subsystem watch of mtd
13:27:59.895 FuContext            added udev subsystem watch of nvme
13:27:59.895 FuContext            added udev subsystem watch of drm
13:27:59.895 FuContext            added udev subsystem watch of drm_dp_aux_dev
13:27:59.895 FuContext            added udev subsystem watch of serio
13:27:59.895 FuPluginTest         init
13:27:59.895 FuContext            added udev subsystem watch of thunderbolt
13:27:59.895 FuContext            added udev subsystem watch of tpm
13:27:59.895 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
13:27:59.895 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
13:27:59.895 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
13:27:59.895 XbSilo               file: a129a98c-e022-1ab5-82ff-2c48a521eb86, current:a129a98c-e022-1ab5-82ff-2c48a521eb86, cached: (null)
13:27:59.895 XbSilo               loading silo with file contents
[... 141 lines cut ...]
[... cut ...]
  Created:              2023-04-25

13:28:00.061 FuPlugin             finalize(flashrom)
13:28:00.061 FuPluginTest         destroy
13:28:00.062 FuProgress           raw timing data was { 0,000, 0,000, 0,002, 0,000 } -- steps were set as [ 1 1 3 95 ] but should have been [ 9 1 89 1 ] at ../libfwupdplugin/fu-context.c:810
13:28:00.062 FuProgress           raw timing data was { 0,001, 0,000, 0,001 } -- steps were set as [ 5 90 5 ] but should have been [ 34 3 63 ] at ../plugins/flashrom/fu-flashrom-plugin.c:269
```

Matombo, 2023-04-25:
In this output the firmware size is again 11MB, but the actual chip and firmware image is 16MB.

hughsie, 2023-04-25:
The SMBIOS is saying the system BIOS is 11MB in size. I think it'll need a quirk like https://github.com/fwupd/fwupd/blob/main/plugins/flashrom/flashrom.quirk#L68 -- can you try that? You can add that in /usr/share/fwupd/quirks.d/ and it should work fine if you restart fwupd.

Matombo, 2023-04-25:
Thanks, that helped, but i wasn't promted to unlock the ME nor did the ME get flashed. It also doesn't get flashed when I boot with ME unlocked regardless of net getting the promt and then flashing. At least the version number in the bios setting didn't change. (Sitenote: using flashrom directly the version number of the ME does get updates even when I didn't unlock the ME and flashrom does not throw any can't write error or something or fails the flash verification.

Also for some reason: after the flash the guid changed:

```
├─NS50 70MU (bios):
│ │   Device ID:          dbee8bd3b1ae0316ad143336155651eedb495a0e
│ │   Current version:    1.07
│ │   Vendor:             Notebook (DMI:INSYDE Corp.)
│ │   GUID:               230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │                       062858de-8c7e-5b00-8f20-1a2a46c2c14e ← FLASHROM\VENDOR_Notebook&PRODUCT_NS50_70MU&REGION_BIOS
│ │                       e061a780-1168-55b7-bea3-fbcf04341bb0 ← FLASHROM\GUID_bab4f96c-34e4-5b92-a785-4e4489b1c395
│ │                       9d3bda37-37ae-5fcc-813c-4b9a7143df8a ← Notebook&Not Applicable&NS50_70MU&Not Applicable&Notebook&NS50_70MU
│ │                       a5fe47a2-d2f3-5f90-a319-9cfafb6985f3 ← Notebook&Not Applicable&NS50_70MU&Not Applicable
│ │                       ea2ce7ca-ea69-5843-8973-a109417195e2 ← Notebook&Not Applicable&NS50_70MU
│ │                       8ac36364-05a8-5b41-bb60-190f9ae615b7 ← Notebook&Not Applicable&Notebook&NS50_70MU
│ │                       217e00fb-06b7-5939-8a94-e2d13bda7211 ← Notebook&Not Applicable&NS50_70MU&Not Applicable&INSYDE Corp.
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs shutdown after installation
│ │                       • Cryptographic hash verification is available
```

But I must add that i flashed from an unbranded clevo BIOS to a branded TUXEDO BIOS. Maybe this has influence on the GUID?

Matombo, 2023-04-25:
I need to restore a known state on the device, but it seems like old (also tuxedo) BIOS have a different FLASHROM\GUID_* then newer ones.

My 2 educated guesses: The NS50MU got a NS70MU variant after release, using the same maineboard but a different chassy and bigger screen -> newer BIOS versions have the dmi string NS50_70MU while older ones have NS50MU. Or, the device got a refresh with new higher clocked cpus, starting with BIOS 1.07.12 or so there is also a 1.07.12N variant, maybe the non N kept the old GUID with the N variant got a new one (or both got new ones), would be suboptimal because the N variant is the one for the older devices.

I will do some more research and then push a new quirk definition for the device. Thanks for the help so far.

For the ME not flashing I schould propably open a new issue, for this not getting to convoluted?

Matombo, 2023-04-25:
So I now factory flashed the device to the current Tuxedo branded firmware. Turns out in our internal cloud there is an older BIOS version that flashes the bios and the branding, and  afaik after that you are only supposed to flash the first 4MB of the bios chip for updates to not overwrite the branding (seems like the efi updater is doing this, but it is rebooting and doing a second automated flash step after that ... i need to do more research what the efi updater is actually doing).

But after doing this the BIOS is no longer recognized as such by fwupd:

```
$ sudo fwupdmgr get-devices --show-all
TUXEDO TUXEDO InfinityBook S 15 Gen6
│
├─Unknown Device:
│     Device ID:          63223cc2d1b0b8d4c45ef8cedcb1f45315d3ff20
│     GUID:               ca5c154c-a46a-59d8-bf77-0ca40522cc85 ← GPIO\ID_INT34C5:00
│   
├─11th Gen Intel Core™ i7-1165G7 @ 2.80GHz:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x000000a6
│ │   Vendor:             Intel
│ │   GUID:               b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│ │                       2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│ │                       1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ └─USB4 host controller:
│       Device ID:        5ad10833af65dd985af261b66cb60be552f136fa
│       Summary:          Unmatched performance for high-speed I/O
│       GUID:             e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│       Device Flags:     • Internal device
│                         • System requires external power source
│                         • Signed Payload
│     
├─SSD 860 EVO M.2 250GB:
│     Device ID:          e421b2fc248391f6fe3e55ddbb3c9043be068bd0
│     Summary:            ATA drive
│     Current version:    RVT24B6Q
│     Vendor:             Samsung (ATA:0x144D, OUI:002538)
│     Serial Number:      S5GFNS0N707641Y
│     GUID:               2ae995cc-df69-5c43-8904-7dda9a2426b7 ← IDE\Samsung_SSD_860_EVO_M.2_250GB___________RVT24B6Q
│                         2de487c2-cd3e-503c-8aea-a226bbb9da47 ← IDE\0Samsung_SSD_860_EVO_M.2_250GB___________
│                         b9a545c3-b0c6-5f21-b659-afa35ab61433 ← Samsung SSD 860 EVO M.2 250GB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    1880113170
│ │   Minimum Version:    1380122624
│ │   Vendor:             TUXEDO (DMI:INSYDE Corp.)
│ │   Update State:       Success
│ │   GUID:               ebd6363b-d954-5ae0-9cbe-7f609aad9ecf
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─BootGuard Configuration:
│       Device ID:        b0d4430dfa6bde9f0c22680df36dbc8c15c80753
│       Current version:  20
│       Vendor:           Intel Corporation (MEI:0x8086)
│       GUID:             dd17041c-09ea-4b17-a271-5b989867ec65
│                         f18c7464-aa01-5b3d-bd4f-6623597d0f70 ← MEI\VEN_8086&DEV_A0E0
│                         395a4458-f3a4-54b5-9947-e8b069f33c31 ← MEI\VEN_8086&DEV_A0E0&REV_20
│                         738bedff-32be-5c1c-a1e1-38715930d4e7 ← MEI\VEN_8086&DEV_A0E0&SUBSYS_155851A1
│                         02880a2c-a1d7-50ab-a38d-a9e08ad46296 ← MEI\VEN_8086&DEV_A0E0&SUBSYS_155851A1&REV_20
[... 16 lines cut ...]
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUID:               54ff3ed7-1ea3-5279-975f-93c13bf2aa53 ← PCI\VEN_8086&DEV_9A49
│                         3c11d563-4f83-5bd1-8369-75359ea6b6f6 ← PCI\VEN_8086&DEV_9A49&REV_01
│                         5cc29699-646b-5b2b-9830-b552fe75c219 ← PCI\VEN_8086&DEV_9A49&SUBSYS_155851A1
│                         891a4b3e-d99a-5dc2-b80b-a98fb956a00e ← PCI\VEN_8086&DEV_9A49&SUBSYS_155851A1&REV_01
│                         06f2a40f-75b6-5238-9fac-71ea35045c1c ← PCI\VEN_8086&DEV_9A49&REV_00
│                         12bee33e-348b-542a-9be5-4979e7227e94 ← PCI\VEN_8086&DEV_9A49&SUBSYS_155851A1&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─UEFI Device Firmware:
      Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
      Summary:            UEFI ESRT device
      Current version:    212
      Vendor:             DMI:INSYDE Corp.
      Update State:       Success
      GUID:               832af090-2ef9-7c47-8f6d-b405c8c7f156
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

This line `230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware` is still the same but there is another guid above it now and the flashrom/dmi guids are missing (probably because it is no longer recognized)

hughsie, 2023-04-25:
Do you need a DMI quirk with the new data? 'sudo fwupdtool hwids' is your friend

Matombo, 2023-06-22:
Figured both problems out by now (thanks to hughsie):
1. Needed a quirk because the returned flash size was based on the used region on the chip, not the physical size of the chip. Fixable with quirk.
2. After flashing the firmware the dmi strings changed so the quirks where no longer applied.

closed 2023-06-22 by Matombo