fwupd/fwupd #5777: Is it possible to force firmware upgrade on a Surface Book 2 with one defective battery

mirogl, 2023-05-01:
Hi,
I have a Surface Book 2, which has two batteries. The one in the tablet is defective - 0% and not charging - the other one in the base unit was replaced by myself and is 100% full. The tablet unit and the base unit are firmly attached together, since i cant detached the tablet as long, as the battery is not charged.

Under Windows 11 I got the message, that the firmware update is only working, if the battery is 40% charged. So Windows ignores, that the battery of the base unit is 100% charged - it just seams to count the tablet battery. 

My question now - is there any possibility to force the firmware update with fwupd - like described at https://github.com/linux-surface/surface-uefi-firmware 

with the command

`$ fwupdmgr install --allow-older --allow-reinstall --force <path to cab file>`

ignoring the 40% battery requirement ? 

I understand the requirement / safty measurements  - but i have two batteries in my notebook and the base unit is 100% full charged and will give power for >6 hours without the power charger. Currently there is no possibility to have no power while the firmware upgrade, since i cant detach the tablet without the tablet having also enough charge.

Therefore I would be glad, if I could force the firmware upgrade somehow.

Thanks
miro

hughsie, 2023-05-02:
Can you try adding `IgnorePower=true` to either /etc/fwupd/daemon.conf or /etc/fwupd/fwupd.conf and then restart the fwupd daemon please?

mirogl, 2023-05-02:
Hi @hughsie,
many thanks. I updated the /etc/fwupd/daemon.conf  and restarted the daemon.
I wanted to be sure to flash the latest firmware - therefore I used 

`$ fwupdmgr install  <path to cab file>
`
instead of  

`$ fwupdmgr install --allow-older --allow-reinstall --force <path to cab file>`

and got the message 

<pre>fwupdmgr install surfaceuefi_98.2.51968_6726b589-d1de-4f26-b2d7-7ac953210d39.cab 
wird entpackt …          [  -                                    ]
Specified firmware is older than installed &apos;98.2.51968 &lt; 1634490112&apos;
</pre>

How can I find out, if the  latest cab files I downloaded from Microsoft from https://www.microsoft.com/en-us/download/details.aspx?id=56261 included in SurfaceBook2_Win10_19041_22.080.2839.0.msi 

<pre>-rw-rw-r-- 1 miro miro 2047932 Apr 29 01:37 <span style="color:#C01C28"><b>SurfaceISH_36.2.14092_7a8be0e8-239e-452c-8281-ca184427982c.cab</b></span>
-rw-rw-r-- 1 miro miro 2048189 Apr 29 01:37 <span style="color:#C01C28"><b>surfaceme_184.92.4222_5f0f3ae0-5d9d-4d64-9385-1fa696ab1719.cab</b></span>
-rw-rw-r-- 1 miro miro  415215 Apr 29 01:37 <span style="color:#C01C28"><b>surfacesam_182.9.139_37da9c3d-6b50-4dbf-82b8-46ca912d98f2.cab</b></span>
-rw-rw-r-- 1 miro miro  378013 Apr 29 01:37 <span style="color:#C01C28"><b>SurfaceTouch_0_242.11.271_5773662e-2343-48b5-b018-db09eae2ea41.cab</b></span>
-rw-rw-r-- 1 miro miro  378187 Apr 29 01:37 <span style="color:#C01C28"><b>SurfaceTouch_242.0.261_5917bcbe-626f-4b76-89d1-b0a8b7a6707a.cab</b></span>
-rw-rw-r-- 1 miro miro 4473698 Apr 29 01:37 <span style="color:#C01C28"><b>surfaceuefi_98.2.51968_6726b589-d1de-4f26-b2d7-7ac953210d39.cab</b></span>
</pre>

are as new as the files under C:\Windows/Firmware from my Windows11 installation

<pre>-rwxr-xr-x 1 miro miro 2033968 Mai  2 11:56 <span style="color:#26A269"><b>SurfaceME_11.8.50.3448.bin</b></span>
-rwxr-xr-x 1 miro miro  402386 Mai  2 11:56 <span style="color:#26A269"><b>SurfaceSAM_182.2304.139.bin</b></span>
-rwxr-xr-x 1 miro miro  365183 Mai  2 11:56 <span style="color:#26A269"><b>SurfaceTouch_2.242.0.1.70.5.bin</b></span>
-rwxr-xr-x 1 miro miro 4458810 Mai  2 11:56 <span style="color:#26A269"><b>SurfaceUEFI_392.178.768.bin</b></span>
</pre>

Or how can I found out the version of the cab files, since the bin files have the version ?

I know it is offtopic - but don't want to test with old firmware

Thanks
Miro

hughsie, 2023-05-02:
> Or could I flash the .bin files directly

You can, but be careful with choosing the correct device. Use `fwupdmgr install-blob bin-file`

mirogl, 2023-05-02:
> 

Hi @hughsie 
Thanks  - what do you mean with the correct device ? Do you mean the notebook itself - it is a Surface Book 2 and the bin files are from a Windows 11 installation on the same disk of this machine - but won't install because of the power. I have a second Ubuntu partition on this device, where i will install it - so all the same device.

Just to be sure - could I somehow read out the version from the cab files to compare them ?

Thanks
Miro

hughsie, 2023-05-02:
> could I somehow read out the version from the cab files to compare them

Sure, "fwupdmgr get-details" should read them if they have the metainfo file included.

hughsie, 2023-05-02:
> Specified firmware is older than installed '98.2.51968 < 1634490112'

You might want to try a newer fwupd -- I think that would get you the actual version format and a sensible error.

mirogl, 2023-05-02:
Hi @hughsie ,
tried the `fwupdmgr install-blob bin-file` - but got the error, that the command is not found. Seems to be related to my version 1.8.12.

Therfore i unpacked the .cab files with 7-zip and could confirm, that they contain the latest versions - corresponding to the .bin file.

I tried to update with
`fwupdmgr install --allow-older --force surfaceuefi_98.2.51968_6726b589-d1de-4f26-b2d7-7ac953210d39.cab`

But it still just booted to the firmware update screen and skipped the firmwareupgrade, like before. The UEFI firmware version has not changed - so it seems, that the IgnorePower = true did not help for the SurfaceBook 2. Many thanks nevertheless for the support and hint.

hughsie, 2023-05-02:
Gah, that was supposed to be `fwupdtool install-blob bin-file` -- sorry.

> But it still just booted to the firmware update screen and skipped the firmwareupgrade, like before

You could try enabling debugging with EnableEfiDebugging=true in /etc/fwupd/uefi_capsule.conf and then do  `fwupdtool install-blob bin-file` -- then the debugging will be in the fwupdmgr get-results and get-devices output.

mirogl, 2023-05-02:
Hi @hughsie,

I tried - but got after execution and short "erasing" the following error - and selected 6 - and "N" - since I'm not sure, if the device will boot again.

<pre><span style="color:#26A269"><b>miro@miro-Surface-Book-2</b></span>:<span style="color:#12488B"><b>~/Downloads/surface-uefi-firmware/out</b></span>$ sudo fwupdtool install-blob SurfaceUEFI_392.178.768.bin 
wird geladen …           [*******                                ]<span style="color:#26A269">14:00:06.042 FuPluginIntelMe      </span><span style="color:#C01C28">failed to get public key using /fpf/OemCred: generic failure [0xb]</span>
Schreibvorgang …         [************                           ]
0.	Cancel
1.	3743975ad7f64f8d6575a9ae49fb3a8856fe186f (MZFLW256HEHP-000MV)
2.	a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
3.	349bb341230b1a86e5effe7dfe4337e1590227bd (UEFI Device Firmware)
4.	2292ae5236790b47884e37cf162dcf23bfcd1c60 (UEFI Device Firmware)
5.	f95c9218acd12697af946874bfe4239587209232 (UEFI Device Firmware)
6.	d96de5c124b60ed6241ebcb6bb2c839cb5580786 (UEFI Device Firmware)
Choose device [0-6]: 6
Waiting…                 [***************************************]]
An update requires a reboot to complete. Restart now? [y|N]: N
</pre>

How can I see now, if this was sucessfull ?

Thanks
Miro

hughsie, 2023-05-02:
> and short "erasing

I think that's a harmless (but scary!) bug -- ignore. I'd try rebooting.

hughsie, 2023-05-02:
> battery level is too low

Urgh, that's when the *firmware* is deciding the battery level is too low -- there's nothing we can do to work around that I'm afraid.

mirogl, 2023-05-02:
Hi @hughsie,
many thanks for the try ! Hoped there is a way to bypass this blocking decision by Microsoft.

Thanks
Miro

closed 2023-05-09 by hughsie

manny73, 2024-02-25:
Hi @mirogl,
I am in your identical situation too. Did you have find the trick?

Thanks
Riccardo

mirogl, 2024-02-25:
> Hi @mirogl, I am in your identical situation too. Did you have find the trick?
> 
> Thanks Riccardo

Hi,
no - I didn't find any other solution, then to change the battery to get the firmware upgrades installed.
Thanks
Mieo

pmarcum, 2024-05-22:
Hi @mirogl, I've been in your identical situation for the past 3 years.  The battery in the keyboard stopped charging first, and then about 6 months later, the battery in the tablet stopped charging, so I've been tethered to a power cord since.  When you say that the only solution you found was to "change the battery" to get the firmware upgrades installed, do you mean that you ended up changing the battery in the tablet itself?  If that's what you mean, how difficult was that task, and did you have to mess with the computer settings to get it to recognize the new battery, or did it automatically recognize the battery?

mirogl, 2024-05-22:
> Hi @mirogl, I've been in your identical situation for the past 3 years. The battery in the keyboard stopped charging first, and then about 6 months later, the battery in the tablet stopped charging, [...]

Hi @pmarcum,
yes - i changed the battery in the keyboard and in the tablet itself. To change the battery in the keyboard is not so difficult, since you just need to warm up the keyboard from the bottom to open it, since it is glued - you can find many videos in youtube for it. Then you need to heat up the botton and remove / change the battery (which is glued to the bottom), which is also not so difficult. But you should not heat the battery itself - only the bottom of the keyboard from the outside - with a fan or something like this. The benefit is, you can use the Surface book 2 again as a notebook without a power cord, but you can't detach the tablet or update the firmware. To be able to detach the tablet and update the firmware, you need to open the tablet by heating up the front glass, which is also glued,  and you have to unskrew many skrews afterwards to get to the battery - it is a very difficult task and you will mostly break the glass. For this you shoud try to find some mobile / tablet repair shop and ask them for a quote to exchange it, but i would not recommend to try it by yourself.

The battery will be automatically recognized - you don't need to mess around with something - but be sure to buy the correct replacement battery.