fwupd/fwupd #5778: Disable fu-self-test /fwupd/spawn-timeout when running on qemu

djcampello, 2023-05-02:
It seems that it takes a whole second for the spawn to happen: 02:28:24.545 --> 02:28:25.560

```
fwupd-1.8.13-r252: # FuSpawn-DEBUG: running '/bin/sh /tmp/portage/sys-apps/fwupd-1.8.13-r252/work/fwupd-1.8.13/src/tests/spawn.sh test'
fwupd-1.8.13-r252: (/tmp/portage/sys-apps/fwupd-1.8.13-r252/work/fwupd-1.8.13-build/src/fu-self-test:29): FuSpawn-DEBUG: 02:28:24.545: running '/bin/sh /tmp/portage/sys-apps/fwupd-1.8.13-r252/work/fwupd-1.8.13/src/tests/spawn.sh test'
fwupd-1.8.13-r252: # DEBUG: got 'this is a test'
fwupd-1.8.13-r252: ** (/tmp/portage/sys-apps/fwupd-1.8.13-r252/work/fwupd-1.8.13-build/src/fu-self-test:29): DEBUG: 02:28:25.560: got 'this is a test'
fwupd-1.8.13-r252: # DEBUG: got 'this is another line1'
fwupd-1.8.13-r252: ** (/tmp/portage/sys-apps/fwupd-1.8.13-r252/work/fwupd-1.8.13-build/src/fu-self-test:29): DEBUG: 02:28:27.326: got 'this is another line1'
fwupd-1.8.13-r252: # DEBUG: got 'this is another line2'
fwupd-1.8.13-r252: ** (/tmp/portage/sys-apps/fwupd-1.8.13-r252/work/fwupd-1.8.13-build/src/fu-self-test:29): DEBUG: 02:28:27.326: got 'this is another line2'
fwupd-1.8.13-r252: # DEBUG: got 'this is another line3'
fwupd-1.8.13-r252: ** (/tmp/portage/sys-apps/fwupd-1.8.13-r252/work/fwupd-1.8.13-build/src/fu-self-test:29): DEBUG: 02:28:27.326: got 'this is another line3'
fwupd-1.8.13-r252: # DEBUG: got 'this is another line4'
fwupd-1.8.13-r252: ** (/tmp/portage/sys-apps/fwupd-1.8.13-r252/work/fwupd-1.8.13-build/src/fu-self-test:29): DEBUG: 02:28:27.326: got 'this is another line4'
fwupd-1.8.13-r252: # DEBUG: got 'done!'
fwupd-1.8.13-r252: ** (/tmp/portage/sys-apps/fwupd-1.8.13-r252/work/fwupd-1.8.13-build/src/fu-self-test:29): DEBUG: 02:28:29.009: got 'done!'
fwupd-1.8.13-r252: ok 56 /fwupd/spawn
fwupd-1.8.13-r252: # FuSpawn-DEBUG: running '/bin/sh /tmp/portage/sys-apps/fwupd-1.8.13-r252/work/fwupd-1.8.13/src/tests/spawn.sh test'
fwupd-1.8.13-r252: (/tmp/portage/sys-apps/fwupd-1.8.13-r252/work/fwupd-1.8.13-build/src/fu-self-test:29): FuSpawn-DEBUG: 02:28:29.014: running '/bin/sh /tmp/portage/sys-apps/fwupd-1.8.13-r252/work/fwupd-1.8.13/src/tests/spawn.sh test'
fwupd-1.8.13-r252: Bail out! ERROR:../fwupd-1.8.13/src/fu-self-test.c:4274:fu_spawn_timeout_func: assertion failed (lines == 1): (0 == 1)
fwupd-1.8.13-r252: [1;36mstderr:[0m
fwupd-1.8.13-r252: **
fwupd-1.8.13-r252: ERROR:../fwupd-1.8.13/src/fu-self-test.c:4274:fu_spawn_timeout_func: assertion failed (lines == 1): (0 == 1)
fwupd-1.8.13-r252: qemu: uncaught target signal 6 (Aborted) - core dumped
```

hughsie, 2023-05-02:
How do we know we're in qemu? Is ther an env variable or something? We could just hide it behind  test-slow which is off by default

djcampello, 2023-05-02:
test-slow sounds good to me!