fwupd/fwupd #5781: fwupd_report_add_metadata_item: assertion 'value != NULL' failed

colttt, 2023-05-04:
**Describe the bug**
when I run refresh (or update) I got ```fwupd_report_add_metadata_item: assertion 'value != NULL' failed```

**Steps to Reproduce**
just run ```fwupd refresh```

**Expected behavior**
no error messages ;-)

**fwupd version information**
Please provide the version of the daemon and client.

```shell
flatpak run org.freedesktop.fwupd --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
compile   org.freedesktop.fwupd         1.8.14
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.3.1-debian+tj+
runtime   org.freedesktop.fwupd         1.8.14

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
via flatpak
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
flatpak run org.freedesktop.fwupd get-devices --show-all-devices
Loading…                 [*****                                  ]09:09:27.228 Fwupd                fwupd_report_add_metadata_item: assertion 'value != NULL' failed
Loading…                 [****************                       ]09:09:28.258 Fwupd                fwupd_report_add_metadata_item: assertion 'value != NULL' failed
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
LENOVO 20XL0012GE
│
├─USB2.0 Hub:
│ │   Device ID:          91c9a317b30acf7c0cd967f2ce6925d518ef19a7
│ │   Summary:            USB 3.x hub
│ │   Current version:    50.33
│ │   Vendor:             VIA Labs, Inc. (USB:0x17EF)
│ │   Install Duration:   15 seconds
│ │   GUIDs:              e2c96871-8be3-530e-a796-5652cee64eeb ← USB\VID_17EF&PID_3071
│ │                       dea51bff-550e-54e7-9b88-d05519771a15 ← USB\VID_17EF&PID_3071&REV_5033
│ │                       6f3f32bd-ac98-523e-9128-ef67941fb614 ← USB\VID_17EF&PID_3071&HUB_20
│ │                       ff324851-023a-508e-8304-f6231f512a64 ← USB\VID_17EF&PID_3071&SPI_C220
│ │                       eae87a37-9a0a-56ae-81b0-dda2516a1202 ← USB\VID_17EF&PID_3071&SPI_C220&REV_5033
│ │   Device Flags:       • Updatable
│ │                       • Cryptographic hash verification is available
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Unsigned Payload
│ │ 
│ └─USB2.0 Hub:
│       Device ID:        41af0a4d70ae1190499dc16830f8f4e856f07aee
│       Summary:          USB 3.x hub
│       Current version:  50.34
│       Vendor:           VIA Labs, Inc. (USB:0x17EF)
│       Install Duration: 15 seconds
│       GUIDs:            e2c96871-8be3-530e-a796-5652cee64eeb ← USB\VID_17EF&PID_3071
│                         f6540363-8288-5dce-86dd-be3656cee148 ← USB\VID_17EF&PID_3071&REV_5034
[... 743 lines cut ...]
Downloading…             [**************************             ]09:11:23.287 Fwupd                download progress: 76%
Downloading…             [*****************************          ]09:11:23.288 Fwupd                download progress: 83%
Downloading…             [********************************       ]09:11:23.288 Fwupd                download progress: 85%
Downloading…             [*********************************      ]09:11:23.288 Fwupd                download progress: 90%
Downloading…             [***********************************    ]09:11:23.289 Fwupd                download progress: 98%
Downloading…             [************************************** ]09:11:23.291 Fwupd                download progress: 100%
Downloading…             [***************************************]09:11:23.291 Fwupd                status-code was 200
09:11:23.292 FuMain               updating lvfs
09:11:23.355 FuEngine             ignoring: Error opening directory “/var/data/lib/fwupd/local.d”: No such file or directory
09:11:23.355 FuEngine             ignoring: Error opening directory “/app/share/fwupd/local.d”: No such file or directory
09:11:23.687 FuEngine             2052 components now in silo
09:11:23.716 FuRelease            remote lvfs has kind=download and so marking as trusted
09:11:23.723 FuRelease            remote lvfs has kind=download and so marking as trusted
09:11:23.724 FuRelease            remote lvfs has kind=download and so marking as trusted
09:11:23.724 FuRelease            remote lvfs has kind=download and so marking as trusted
09:11:23.730 FuRelease            remote lvfs has kind=download and so marking as trusted
09:11:23.730 Fwupd                fwupd_report_add_metadata_item: assertion 'value != NULL' failed
09:11:23.781 FuRelease            remote lvfs has kind=download and so marking as trusted
09:11:23.781 FuRelease            remote lvfs has kind=download and so marking as trusted
09:11:23.784 FuRelease            remote lvfs has kind=download and so marking as trusted
09:11:23.837 FuRelease            remote lvfs has kind=download and so marking as trusted
09:11:23.837 Fwupd                fwupd_report_add_metadata_item: assertion 'value != NULL' failed
09:11:23.880 FuEngine             resetting update motd timeout
```
</details>

hughsie, 2023-05-04:
Can you set `export G_DEBUG=fatal-criticals` and then get me a backtrace please?

colttt, 2023-05-04:
and here we go:
```
export G_DEBUG=fatal-criticals;  flatpak run org.freedesktop.fwupd refresh --verbose
10:50:04.206 FuDebug              verbose to info (on console 1)
Loading…                 [ -                                     ]10:50:04.206 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/dell-esrt.conf
10:50:04.207 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/lvfs-testing.conf
10:50:04.207 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/lvfs.conf
10:50:04.207 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/vendor-directory.conf
10:50:04.207 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/vendor.conf
10:50:04.207 FuRemoteList         enabled remotes: dell-esrt, lvfs
Loading…                 [                                       ]10:50:04.207 FuEngine             client certificate now exists: nothing to do
Loading…                 [*                                      ]10:50:04.210 FuContext            added udev subsystem watch of platform
10:50:04.210 FuContext            added udev subsystem watch of pci
10:50:04.210 FuContext            added udev subsystem watch of block
10:50:04.210 FuContext            added udev subsystem watch of i2c
10:50:04.210 FuContext            added udev subsystem watch of i2c-dev
10:50:04.210 FuContext            added udev subsystem watch of hidraw
10:50:04.210 FuContext            added udev subsystem watch of gpio
10:50:04.210 FuContext            added udev subsystem watch of mei
10:50:04.210 FuContext            added udev subsystem watch of iommu
10:50:04.210 FuContext            added udev subsystem watch of video4linux
10:50:04.210 FuContext            added udev subsystem watch of msr
10:50:04.210 FuContext            added udev subsystem watch of mtd
10:50:04.210 FuContext            added udev subsystem watch of nvme
10:50:04.211 FuContext            added udev subsystem watch of drm
10:50:04.211 FuContext            added udev subsystem watch of drm_dp_aux_dev
10:50:04.211 FuContext            added udev subsystem watch of serio
10:50:04.211 FuContext            added udev subsystem watch of thunderbolt
10:50:04.211 FuContext            added udev subsystem watch of tpm
10:50:04.211 FuQuirks             loading quirks from /app/share/fwupd/quirks.d
10:50:04.211 FuQuirks             loading quirks from /var/data/lib/fwupd/quirks.d
Loading…                 [*                                      ]10:50:04.212 FuContext            loading HwId overrides from /app/etc/fwupd/daemon.conf
10:50:04.212 FuContext            SMBIOS Manufacturer=LENOVO
10:50:04.212 FuContext            SMBIOS EnclosureKind=a
10:50:04.212 FuContext            SMBIOS Family=ThinkPad T14 Gen 2a
10:50:04.212 FuContext            SMBIOS ProductName=20XL0012GE
10:50:04.212 FuContext            SMBIOS ProductSku=LENOVO_MT_20XL_BU_Think_FM_ThinkPad T14 Gen 2a
10:50:04.212 FuContext            SMBIOS BiosVendor=LENOVO
10:50:04.212 FuContext            SMBIOS BiosVersion=R1MET53W (1.23 )
10:50:04.212 FuContext            SMBIOS BiosMajorRelease=01
10:50:04.212 FuContext            SMBIOS BiosMinorRelease=17
10:50:04.212 FuContext            SMBIOS FirmwareMajorRelease=01
10:50:04.212 FuContext            SMBIOS FirmwareMinorRelease=17
10:50:04.212 FuContext            SMBIOS BaseboardManufacturer=LENOVO
10:50:04.212 FuContext            SMBIOS BaseboardProduct=20XL0012GE
10:50:04.212 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/data/lib/fwupd/system.dtb
10:50:04.215 FuContext            added udev subsystem watch of firmware-attributes
10:50:04.246 FuBiosSettings       loaded 87 BIOS settings
Loading…                 [**                                     ]10:50:04.246 FuEngine             ignoring: Error opening directory “/var/data/lib/fwupd/local.d”: No such file or directory
10:50:04.246 FuEngine             ignoring: Error opening directory “/app/share/fwupd/local.d”: No such file or directory
10:50:04.248 FuEngine             2052 components now in silo
Loading…                 [***                                    ]10:50:04.259 FuEngine             plugins disabled: test, test_ble
Loading…                 [***                                    ]10:50:04.263 FuContext            battery threshold now 25
Loading…                 [ -                                     ]10:50:04.263 FuEngine             disabling plugin because: failed to startup using dell_esrt: UEFI firmware already supported
10:50:04.264 FuEngine             disabling plugin because: failed to startup using powerd: no service that owns the name for org.chromium.PowerManager
10:50:04.264 FuPlugin             mode was 0644, and needs to be 0640
10:50:04.264 FuEngine             disabling plugin because: failed to change permission of /app/etc/fwupd/redfish.conf
Loading…                 [****                                   ]10:50:04.266 FuContext            power state now ac-fully-charged
10:50:04.266 FuContext            battery level now 99
10:50:04.266 FuContext            lid state now closed
10:50:04.266 FuEngine             disabling plugin because: failed to startup using dell: firmware updating not supported: no structure with type de
10:50:04.266 FuEngine             disabling plugin because: failed to coldplug using acpi_phat: Failed to open file “/sys/firmware/acpi/tables/PHAT”: No such file or directory
[... cut ...]
10:53:12.973 FuDeviceList         ::added 349bb341230b1a86e5effe7dfe4337e1590227bd [System Firmware]
10:53:12.976 FuDeviceList         ::added 2292ae5236790b47884e37cf162dcf23bfcd1c60 [UEFI Device Firmware]
10:53:12.979 FuRelease            remote lvfs has kind=download and so marking as trusted
10:53:12.979 FuDevice             ignoring system power for f95c9218acd12697af946874bfe4239587209232 battery
10:53:12.979 Fwupd                fwupd_report_add_metadata_item: assertion 'value != NULL' failed
```

hughsie, 2023-05-04:
Ohh, I think you need to use something like: `flatpak run --env=G_DEBUG=fatal-criticals org.freedesktop.fwupd refresh --verbose` -- not sure how to get a backtrace out of that tho.

colttt, 2023-05-04:
that also does not work.. maybe this is usefull:
first you need access to the shell:
``` export G_DEBUG=fatal-criticals;  flatpak run --env=G_DEBUG=fatal-criticals --command=bash org.freedesktop.fwupd```
then run the command:
``` export G_DEBUG=fatal-criticals; fwupdtool refresh --verbose ```

in an other terminal window:
```
flatpak-coredumpctl org.freedesktop.fwupd
Executable /app/bin/fwupdtool doesn't seem to be a flatpaked application.
Running: `"flatpak" "run" "--filesystem=home" "--filesystem=/tmp" "--command=gdb" "--devel" "org.freedesktop.fwupd" "/app/bin/fwupdtool" "/tmp/tmprtixdj1r"`
GNU gdb (GDB) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-unknown-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /app/bin/fwupdtool...
(No debugging symbols found in /app/bin/fwupdtool)
[New LWP 3]
[New LWP 7]
[New LWP 8]
[New LWP 6]
[New LWP 5]
[New LWP 4]

warning: Section `.reg-xstate/3' in core file too small.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `fwupdtool refresh --verbose'.
Program terminated with signal SIGTRAP, Trace/breakpoint trap.

warning: Section `.reg-xstate/3' in core file too small.
#0  0x00007f610ff2643c in g_logv () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
[Current thread is 1 (Thread 0x7f610e768980 (LWP 3))]
(gdb) bt
#0  0x00007f610ff2643c in g_logv () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007f610ff266e3 in g_log () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007f6110290a52 in fu_release_load () at /app/lib/fwupd-1.8.14/libfwupdengine.so
#3  0x00007f61102794cb in  () at /app/lib/fwupd-1.8.14/libfwupdengine.so
#4  0x00007f6110280901 in fu_engine_get_releases_for_device () at /app/lib/fwupd-1.8.14/libfwupdengine.so
#5  0x00007f61102815af in  () at /app/lib/fwupd-1.8.14/libfwupdengine.so
#6  0x00007f611028438d in fu_engine_add_device () at /app/lib/fwupd-1.8.14/libfwupdengine.so
#7  0x00007f611044f4f2 in g_closure_invoke () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#8  0x00007f61104641e8 in signal_emit_unlocked_R.isra.0 () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#9  0x00007f611046b195 in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#10 0x00007f611046b363 in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#11 0x00007f6110506aa0 in fu_plugin_device_add () at /app/lib/fwupd-1.8.14/libfwupdplugin.so
#12 0x00007f611029ad0e in  () at /app/lib/fwupd-1.8.14/libfwupdengine.so
#13 0x00007f6110507c70 in fu_plugin_runner_coldplug () at /app/lib/fwupd-1.8.14/libfwupdplugin.so
#14 0x00007f6110272899 in  () at /app/lib/fwupd-1.8.14/libfwupdengine.so
#15 0x00007f61102872ff in fu_engine_load () at /app/lib/fwupd-1.8.14/libfwupdengine.so
#16 0x0000557128b2ee44 in  ()
#17 0x0000557128b309ac in  ()
#18 0x00007f61105e7294 in fu_util_cmd_array_run () at /app/lib/fwupd-1.8.14/libfwupdutil.so
#19 0x0000557128b2e182 in main ()
(gdb) 
```

I hope it's more usefull

hughsie, 2023-05-04:
I think the fix is this: https://gitlab.com/fwupd/lvfs-website/-/commit/a90c0734a8e5893f0b712534c29a93e9496c9bd5

I'll deploy that in ~1h.

colttt, 2023-05-04:
still happens, if it was a fwupd issue, I use flatpak it takes some time to refresh?!
There is no news version, and it still happens with the current version (1.8.14)

hughsie, 2023-05-04:
The CDN does take some time to refresh; maybe try `fwupdtool --force refresh` in a few hours?

colttt, 2023-05-04:
I will try that tomorrow

colttt, 2023-05-05:
still happens

hughsie, 2023-05-05:
So if you do `sudo -i flatpak run org.freedesktop.fwupd refresh --verbose --force` you still get the critical warning? I don't get that here, and I'm wondering if it's because you've somehow got/getting cached metadata?

colttt, 2023-05-05:
I would say yes:

```
flatpak run org.freedesktop.fwupd refresh --verbose --force
10:36:10.231 FuDebug              verbose to info (on console 1)
Loading…                 [ -                                     ]10:36:10.231 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/dell-esrt.conf
10:36:10.231 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/lvfs-testing.conf
10:36:10.231 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/lvfs.conf
10:36:10.231 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/vendor-directory.conf
10:36:10.231 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/vendor.conf
10:36:10.232 FuRemoteList         enabled remotes: dell-esrt, lvfs
Loading…                 [                                       ]10:36:10.232 FuEngine             client certificate now exists: nothing to do
Loading…                 [*                                      ]10:36:10.235 FuContext            added udev subsystem watch of platform
10:36:10.235 FuContext            added udev subsystem watch of pci
10:36:10.235 FuContext            added udev subsystem watch of block
10:36:10.235 FuContext            added udev subsystem watch of i2c
10:36:10.235 FuContext            added udev subsystem watch of i2c-dev
10:36:10.235 FuContext            added udev subsystem watch of hidraw
10:36:10.236 FuContext            added udev subsystem watch of gpio
10:36:10.236 FuContext            added udev subsystem watch of mei
10:36:10.236 FuContext            added udev subsystem watch of iommu
10:36:10.236 FuContext            added udev subsystem watch of video4linux
10:36:10.236 FuContext            added udev subsystem watch of msr
10:36:10.236 FuContext            added udev subsystem watch of mtd
10:36:10.236 FuContext            added udev subsystem watch of nvme
10:36:10.236 FuContext            added udev subsystem watch of drm
10:36:10.237 FuContext            added udev subsystem watch of drm_dp_aux_dev
10:36:10.237 FuContext            added udev subsystem watch of serio
10:36:10.237 FuContext            added udev subsystem watch of thunderbolt
10:36:10.237 FuContext            added udev subsystem watch of tpm
10:36:10.237 FuQuirks             loading quirks from /app/share/fwupd/quirks.d
10:36:10.237 FuQuirks             loading quirks from /var/data/lib/fwupd/quirks.d
Loading…                 [*                                      ]10:36:10.238 FuContext            loading HwId overrides from /app/etc/fwupd/daemon.conf
10:36:10.238 FuContext            SMBIOS Manufacturer=LENOVO
10:36:10.238 FuContext            SMBIOS EnclosureKind=a
10:36:10.238 FuContext            SMBIOS Family=ThinkPad T14 Gen 2a
10:36:10.238 FuContext            SMBIOS ProductName=20XL0012GE
10:36:10.238 FuContext            SMBIOS ProductSku=LENOVO_MT_20XL_BU_Think_FM_ThinkPad T14 Gen 2a
10:36:10.238 FuContext            SMBIOS BiosVendor=LENOVO
10:36:10.238 FuContext            SMBIOS BiosVersion=R1MET53W (1.23 )
10:36:10.238 FuContext            SMBIOS BiosMajorRelease=01
10:36:10.238 FuContext            SMBIOS BiosMinorRelease=17
10:36:10.238 FuContext            SMBIOS FirmwareMajorRelease=01
10:36:10.238 FuContext            SMBIOS FirmwareMinorRelease=17
10:36:10.238 FuContext            SMBIOS BaseboardManufacturer=LENOVO
10:36:10.238 FuContext            SMBIOS BaseboardProduct=20XL0012GE
10:36:10.238 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/data/lib/fwupd/system.dtb
10:36:10.241 FuContext            added udev subsystem watch of firmware-attributes
10:36:10.276 FuBiosSettings       loaded 87 BIOS settings
Loading…                 [**                                     ]10:36:10.276 FuEngine             ignoring: Error opening directory “/var/data/lib/fwupd/local.d”: No such file or directory
10:36:10.276 FuEngine             ignoring: Error opening directory “/app/share/fwupd/local.d”: No such file or directory
10:36:10.278 FuEngine             2053 components now in silo
Loading…                 [***                                    ]10:36:10.290 FuEngine             plugins disabled: test, test_ble
Loading…                 [***                                    ]10:36:10.293 FuContext            battery threshold now 25
Loading…                 [ -                                     ]10:36:10.293 FuEngine             disabling plugin because: failed to startup using dell_esrt: UEFI firmware already supported
10:36:10.295 FuEngine             disabling plugin because: failed to startup using powerd: no service that owns the name for org.chromium.PowerManager
10:36:10.295 FuPlugin             mode was 0644, and needs to be 0640
10:36:10.295 FuEngine             disabling plugin because: failed to change permission of /app/etc/fwupd/redfish.conf
Loading…                 [****                                   ]10:36:10.298 FuContext            power state now battery-discharging
10:36:10.298 FuContext            battery level now 97
10:36:10.298 FuContext            lid state now open
10:36:10.299 FuEngine             disabling plugin because: failed to startup using dell: firmware updating not supported: no structure with type de
10:36:10.299 FuEngine             disabling plugin because: failed to coldplug using acpi_phat: Failed to open file “/sys/firmware/acpi/tables/PHAT”: No such file or directory
[... cut ...]
10:36:12.451 FuRelease            remote lvfs has kind=download and so marking as trusted
10:36:12.451 FuRelease            remote lvfs has kind=download and so marking as trusted
10:36:12.454 FuRelease            remote lvfs has kind=download and so marking as trusted
10:36:12.469 FuRelease            remote lvfs has kind=download and so marking as trusted
10:36:12.469 Fwupd                fwupd_report_add_metadata_item: assertion 'value != NULL' failed
10:36:12.505 FuEngine             resetting update motd timeout
```

hughsie, 2023-05-05:
Huh. The LVFS change seemed not to work -- I'll try something less empty string and more dummy. Give me a few hours.

colttt, 2023-05-08:
jup, now it works fine..

closed 2023-05-08 by colttt