fwupd/fwupd #5782: plugins: nordic-hid: Issue with discovering one of the child devices fails discovery of other children and parent

MarekPieta, 2023-05-04:
**Describe the bug**
If we use a set of HID peripherals connected via dongle in the LVFS, the dongle is a parent device and the peripherals are children.
If fwupd device discovery fails for one of the peripherals, other devices cannot be accessed in the fwupd.

**Steps to Reproduce**
Use nRF Desktop dongle with gaming mouse and keyboard connected through the dongle.
Keyboard shall use configuration that **does not support the DFU module** (without support for configuration channel DFU).
During the discovery, the following log is observed:
```
the board has no support of bootloader runtime detection
Failed to add device /sys/devices/pci0000:00/0000:00:15.0/0000:03:00.0/usb3/3-2/3-2:1.0/0003:1915:52DC.000A/hidraw/hidraw1: the bootloader is not detected nor set via quirk
```
The warning log is caused by issue during keyboard discovery (keyboard used configuration without bootloader and DFU module). 
None of the devices can be accessed by the fwupd tool in that case.

**Expected behavior**
Issue when discovering one of the devices should not affect other devices (they should be used as separate devices).
With the current implementation, connecting a peripheral without DFU support through the dongle would fail discovery for all of the devices.

**fwupd version information**
fwupd built locally from the latest main (`ee3fd1f3713cd3966eed088c0a87fb41b92b9713`).
Running on Ubuntu 22.04.2 LTS.

**Additional context**
When the issue replicates, the `fu_nordic_hid_cfg_channel_add_peers` function seems to return `TRUE`.
Peers discovery itself is actually successful and we reach the end of the peer list.

The issue seems to happen a bit later, when keyboard is discovered through the dongle (can be noticed by looking at the log).
It seems like the error reported for child device is somehow improperly linked to parent.

hughsie, 2023-05-04:
Is fu_nordic_hid_cfg_channel_cmd_receive() returning with an error for the failing device? Can you attach a `sudo fwupdtool get-devices --plugins nordic_hid --verbose --verbose` log when it shows the failure please.

hughsie, 2023-05-04:
> Is fu_nordic_hid_cfg_channel_cmd_receive() returning with an error for the failing device

i.e. does the wip/hughsie/nordic-peer-poll branch fix this?

closed 2023-06-02 by hughsie