fwupd/fwupd #5786: SteelSeries Aerox 3 is not detected if connected over BT on Chrome OS

d4s, 2023-05-04:
**Describe the bug**
The mouse does not detected by `fwupd` on Chromebook if connected over BT.
Properly detected if paired, but  after power off/on cycle on the mouse it doesn't appeared in the devices list.
However after `fwupd` restart -- detected again, until power off/on.

**Steps to Reproduce**

1. Connect the mouse (already paired, switch on) to the Chromebook:
```
localhost ~ # dmesg | tail -n 5 
[17439.788352] input: Aerox 3 WL Mouse as /devices/virtual/misc/uhid/0005:0111:183A.0008/input/input32
[17439.788968] input: Aerox 3 WL Keyboard as /devices/virtual/misc/uhid/0005:0111:183A.0008/input/input33
[17439.789416] input: Aerox 3 WL as /devices/virtual/misc/uhid/0005:0111:183A.0008/input/input34
[17439.789811] input: Aerox 3 WL as /devices/virtual/misc/uhid/0005:0111:183A.0008/input/input35
[17439.790042] hid-generic 0005:0111:183A.0008: input,hidraw1: BLUETOOTH HID v1.08 Mouse [Aerox 3 WL] on d8:f8:83:ad:b2:d5

localhost ~ # ls -l /dev/hidraw*
crw-rw----. 1 root hidraw 239, 0 May  4 18:39 /dev/hidraw0
crw-rw----. 1 root hidraw 239, 1 May  4 23:29 /dev/hidraw1
crw-rw----. 1 root hidraw 239, 3 May  4 18:39 /dev/hidraw3
```
2. Check with `fwupdmgr` -- and it is **not detected**
```
localhost ~ # fwupdmgr get-devices | grep -B1 "Device ID:"
??DUTA42:
?     Device ID:          eafa35acc9a754da25475e870cca0b4460daa4ed
--
??Display controller:
?     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
--
??Generic Billboard Device:
?     Device ID:          8de32b71a6e39772a45c909c65c76d2d4d9ce428
--
??Integrated Webcam?:
?     Device ID:          08d460be0f1f9f128413f816022a6439e0078018
--
??Touchpad:
      Device ID:          4913a1ebc80c606184d022770fdc2db374078742
```

3. But if I do `restart fwupd` -- it is detected properly:

```
localhost ~ # restart fwupd
fwupd start/running, process 8518
localhost ~ # fwupdmgr get-devices | grep -B1 "Device ID:"
??Aerox 3 Wireless Mouse via Bluetooth:
?     Device ID:          ce2505af510d8b52c393427b0701c9dfbf0260b9
--
??DUTA42:
?     Device ID:          eafa35acc9a754da25475e870cca0b4460daa4ed
--
??Display controller:
?     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
--
??Generic Billboard Device:
?     Device ID:          8de32b71a6e39772a45c909c65c76d2d4d9ce428
--
??Integrated Webcam?:
?     Device ID:          08d460be0f1f9f128413f816022a6439e0078018
--
??Touchpad:
      Device ID:          4913a1ebc80c606184d022770fdc2db374078742
```

**Expected behavior**
Properly detected the connected over BT mouse.

**fwupd version information**

```
localhost ~ # fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.13
compile   com.hughsie.libxmlb           0.3.11
compile   org.freedesktop.gusb          0.4.5
compile   com.hughsie.libjcat           0.1.8
runtime   org.kernel                    5.15.109-19159-g476b5c53a228
runtime   org.freedesktop.gusb          0.4.5
compile   org.freedesktop.fwupd         1.8.13
```
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
localhost ~ # fwupdmgr get-devices --show-all-devices 
Google Nightfury
│
├─DUTA42:
│     Device ID:          eafa35acc9a754da25475e870cca0b4460daa4ed
│     Current version:    0x0300000000000000
│     Vendor:             Samsung/Sandisk/LG (EMMC:0x000015)
│     Serial Number:      0x95e4bcb7
│     GUIDs:              c039ed61-1dac-544a-94be-a0ef017d8d9e ← EMMC\NAME_DUTA42
│                         8e3b9011-0e80-5eb4-a671-03e574378e8b ← EMMC\NAME_DUTA42&REV_0x0300000000000000
│                         5496aba3-9171-589d-b035-36ffc4327d52 ← EMMC\MAN_0015&OEM_0100
│                         c4529c82-3a2e-5bf4-8162-58c452718e1c ← EMMC\MAN_0015&OEM_0100&NAME_DUTA42
│                         972230c9-65e2-5f7e-865c-2258a7c62dd7 ← EMMC\MAN_0015&NAME_DUTA42&REV_0x0300000000000000
│                         b1b43144-05d0-56a8-90ba-6582dbdd3334 ← EMMC\MAN_0015&OEM_0100&NAME_DUTA42&REV_0x0300000000000000
│                         9a1b1223-fd3e-5e6f-af93-f12fb2f8405a ← EMMC\0021&0256&DUTA42
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Display controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             PCI:0x8086
│     GUIDs:              5589d313-88e5-54d3-8f4c-b2ee3882d4bc ← PCI\VEN_8086&DEV_9B41
│                         9779d679-0f31-5dd5-b3c2-f6ff275a9ca9 ← PCI\VEN_8086&DEV_9B41&REV_02
│                         2ffa2655-4d89-5462-81f9-8db939585b5e ← PCI\VEN_8086&DEV_9B41&SUBSYS_80862212
│                         e4ef1dc4-be60-5ebc-99b7-0c2b8475c0eb ← PCI\VEN_8086&DEV_9B41&SUBSYS_80862212&REV_02
│                         25be981a-c8a2-5d3a-82cc-39c2a795effb ← PCI\VEN_8086&DEV_9B41&REV_00
│                         d5a11444-73b3-5589-b182-de5c6cd56e3c ← PCI\VEN_8086&DEV_9B41&SUBSYS_80862212&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Generic Billboard Device:
│     Device ID:          8de32b71a6e39772a45c909c65c76d2d4d9ce428
│     Current version:    7.99.36.0
│     Vendor:             Fresco Logic, Inc (USB:0x1D5C)
│     Install Duration:   15 seconds
│     GUIDs:              4b8e93aa-7207-5706-9e90-b55306343c84 ← USB\VID_1D5C&PID_7102
│                         5564faf2-5478-5d9c-9e11-29039b6ae7e0 ← USB\VID_1D5C&PID_7102&REV_0100
│                         790a1b00-9cf9-5220-af78-ef690d303fa8 ← USB\VID_1D5C&PID_7102&CID_63
│     Device Flags:       • Updatable
│   
├─Integrated Webcam™:
│     Device ID:          08d460be0f1f9f128413f816022a6439e0078018
│     Summary:            Fake webcam
│     Current version:    1.2.2
│     Minimum Version:    1.2.0
│     Bootloader Version: 0.1.2
│     Vendor:             ACME Corp. (USB:0x046D)
│     GUID:               b585990a-003e-5270-89d5-3705a17f9a43
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Cryptographic hash verification is available
│                         • Unsigned Payload
│   
└─Touchpad:
      Device ID:          4913a1ebc80c606184d022770fdc2db374078742
      Current version:    0x0004
      Bootloader Version: 0x0000
      Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
      GUIDs:              8f7d987d-6dc3-53be-9b5d-fc561029b01e ← I2C\NAME_ELAN0000:00
                          af1e280d-76eb-542b-a71a-2358e20df38f ← HIDRAW\VID_04F3&PID_0400
                          389d8748-dae0-5d7b-956c-f9183a0b8899 ← HIDRAW\VEN_04F3&DEV_0400&MOD_00D2
                          3852e430-731f-55fa-a0e1-f2ff3b818c9f ← ELANTP\ICTYPE_0D
                          161af12c-a58a-53d2-b8c3-fe5de7d80ebe ← ELANTP\ICTYPE_0D&MOD_00D2
                          9e7693d2-c2a9-59ff-b0df-d432e589000b ← ELANTP\ICTYPE_0D&MOD_00D2&DRIVER_ELAN_I2C
      Device Flags:       • Internal device
                          • Updatable

```

**Additional questions**

- Operating system and version: 
```
GOOGLE_CRASH_ID=ChromeOS
ID_LIKE=chromiumos
ID=chromeos
NAME=Chrome OS
VERSION=115
VERSION_ID=115
BUILD_ID=15444.0.0
```
- Have you tried rebooting?
  N/A
- Is this a regression?
  N/A

d4s, 2023-05-04:
Not sure if it is a bug of `fwupd` or Chrome OS.
With Logitech M650B BT mouse `fwupd` works as expected.

hughsie, 2023-05-05:
Different behaviour on regular Linux? Different behaviour with minijail disabled? If so, can you get a -vv --no-timestamp on both operating systems and we can compare what's happening. Thanks

d4s, 2023-05-05:
Sorry, looks like to start `fwupd` without `minijail` is a problem.
2 logs in `minjail` with `-vv --no-timestamp` and the single log from the host.

1.  Reconnect the device after switch on:
```
2023-05-05T12:18:19.686107Z WARNING fwupd[27525]: 12:18:19.686 FuEngine             /sys/devices/virtual/misc/uhid/0005:0111:183A.0012/hidraw/hidraw2 added FuUdevDevice:
2023-05-05T12:18:19.686107Z WARNING fwupd[27525]:  Flags:                none
2023-05-05T12:18:19.686107Z WARNING fwupd[27525]:  BackendId:            /sys/devices/virtual/misc/uhid/0005:0111:183A.0012/hidraw/hidraw2
2023-05-05T12:18:19.686107Z WARNING fwupd[27525]:  AcquiesceDelay:       2500
2023-05-05T12:18:19.686107Z WARNING fwupd[27525]:  Subsystem:            hidraw
2023-05-05T12:18:19.686107Z WARNING fwupd[27525]:  DeviceFile:           /dev/hidraw2
2023-05-05T12:18:19.686107Z WARNING fwupd[27525]:  SysfsPath:            /sys/devices/virtual/misc/uhid/0005:0111:183A.0012/hidraw/hidraw2
2023-05-05T12:18:19.687808Z WARNING fwupd[27525]: 12:18:19.687 FuDevice             unknown device overwriting name value: Aerox 3 WL->Aerox 3 Wireless Mouse via Bluetooth
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]: 12:18:19.689 FuEngine             /sys/devices/virtual/misc/uhid/0005:0111:183A.0012/hidraw/hidraw2 added FuUdevDevice:
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  Name:                 Aerox 3 Wireless Mouse via Bluetooth
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  Guid:                 14b18a05-6162-58f6-bae4-a6ff3c8b66c1 <- HIDRAW\VEN_0111&DEV_183A ?
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  Guid:                 a6afb856-9ef7-53a9-9445-d923fcc6d57d <- HIDRAW\VEN_0111&DEV_183A&REV_00 ?
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  Flags:                none
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  VendorId:             HIDRAW:0x0111
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  Icon:                 input-mouse
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  Guid[quirks]:         3b2ba11c-4c0e-57c4-a26c-050ad9ca5317 <- HIDRAW\VEN_0111
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  Guid[quirks]:         eb45b692-59a3-5c1d-8ddb-950ccd241df4 <- HIDRAW
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  BackendId:            /sys/devices/virtual/misc/uhid/0005:0111:183A.0012/hidraw/hidraw2
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  AcquiesceDelay:       2500
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  PossiblePlugin:       steelseries
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  Vendor:               0x111
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  Model:                0x183a
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  Subsystem:            hidraw
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  DeviceFile:           /dev/hidraw2
2023-05-05T12:18:19.689223Z WARNING fwupd[27525]:  SysfsPath:            /sys/devices/virtual/misc/uhid/0005:0111:183A.0012/hidraw/hidraw2
2023-05-05T12:18:19.694568Z WARNING fwupd[27525]: 12:18:19.692 FuDevice             already set GType to FuSteelseriesFizzHid, ignoring FuSteelseriesFizzHid
2023-05-05T12:18:19.694568Z WARNING fwupd[27525]: 12:18:19.694 FuDevice             using ce2505af510d8b52c393427b0701c9dfbf0260b9 for hid
2023-05-05T12:18:19.694568Z WARNING fwupd[27525]: 12:18:19.694 FuPluginSteelSeries  Version:
2023-05-05T12:18:19.694568Z WARNING fwupd[27525]:       | 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
2023-05-05T12:18:19.694568Z WARNING fwupd[27525]: -------+------------------------------------------------------------------------------------------------
2023-05-05T12:18:19.694568Z WARNING fwupd[27525]: 0x0000 | 04 90 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
2023-05-05T12:18:19.694568Z WARNING fwupd[27525]: 0x0020 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
2023-05-05T12:18:19.694568Z WARNING fwupd[27525]: 0x0040 | 00 
2023-05-05T12:18:19.874181Z WARNING fwupd[27525]: 12:18:19.874 FuPluginSteelSeries  Version:
2023-05-05T12:18:19.874181Z WARNING fwupd[27525]:       | 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
2023-05-05T12:18:19.874181Z WARNING fwupd[27525]: -------+------------------------------------------------------------------------------------------------
2023-05-05T12:18:19.874181Z WARNING fwupd[27525]: 0x0000 | 05 12 13 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
2023-05-05T12:18:19.874181Z WARNING fwupd[27525]: 0x0020 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
2023-05-05T12:18:19.874181Z WARNING fwupd[27525]: 0x0040 | 00 
2023-05-05T12:18:19.874181Z WARNING fwupd[27525]: 12:18:19.874 FuEngine             failed to add device /sys/devices/virtual/misc/uhid/0005:0111:183A.0012/hidraw/hidraw2: /sys/devices/virtual/misc/uhid/0005:0111:183A.0012/hidraw/hidraw2 failed but no error set
```

2. Detected while `fwupd` started:
```
2023-05-05T12:26:52.269226Z WARNING fwupd[28927]: FuEngine             /sys/devices/virtual/misc/uhid/0005:0111:183A.0015/hidraw/hidraw2 added FuUdevDevice:
2023-05-05T12:26:52.269226Z WARNING fwupd[28927]:  Flags:                none
[... cut ...]
May 05 14:40:38 amber fwupd[351405]: 0x0000 | 05 12 13 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
May 05 14:40:38 amber fwupd[351405]: 0x0020 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
May 05 14:40:38 amber fwupd[351405]: 0x0040 | 00
May 05 14:40:38 amber systemd[1]: fwupd.service: Main process exited, code=killed, status=11/SEGV
May 05 14:40:38 amber systemd[1]: fwupd.service: Failed with result 'signal'.
```

hence the issue is `fwupd` -- I'll check the core reason.

hughsie, 2023-05-05:
>  On the host I have a segmentation fault:

Woah, super uncool -- but easy enough to pinpoint.

closed 2023-05-07 by d4s