fwupd/fwupd #5787: Firmware update support for stock RP2040 bootloader

jackpot51, 2023-05-05:
**Is your feature request related to a problem? Please describe.**

A number of devices are using the RP2040 microcontroller, and System76 is now planning to do the next launch and thelio-io products using this microcontroller. It comes with a ROM bootloader that has an ID of `2e8a:0003` and basically requires a firmware file to be placed onto a USB storage device. While it is possible to implement a second stage bootloader supporting another protocol, for simplicity and for recovering from SPI chip issues, being able to flash firmware when the ROM bootloader is detected is a critical feature.

**Describe the solution you'd like**

The process should be similar to the AVR microcontroller DFU bootloader flashing process. The ID of the bootloader should be provided along with the ID of the non-bootloader device. When fwupd detects the switch from the non-bootloader device to the bootloader device, it should mount the USB storage device and copy the firmware file to it.

**Describe alternatives you've considered**

The best alternative would be to include a second stage bootloader supporting DFU. This presents several problems:
- I believe it would take longer to implement than having a new plugin for the RP2040's default ROM bootloader
- It requires more device-side code which increases firmware size where it is far more costly than increasing userspace code in fwupd
- It does not provide for flashing completely bricked systems that may not have a functional second stage bootloader, but should still have the ROM bootloader
- It is not generic to all usages of the RP2040 - different devices or firmwares may have different second stage bootloaders, whereas the ROM bootloader will always be present

**Additional context**

I am completely willing to implement this plugin, and intend to do so in order to deliver firmware updates for new launch and thelio-io revisions using the RP2040. As such, I have assigned myself to this issue. If there are any recommendations based on this information above, please let me know.

jackpot51, 2023-05-05:
Yes, it is, I didn't notice that plugin before.

jackpot51, 2023-05-05:
Seems like this will be even easier than I thought.

hughsie, 2023-05-05:
> Seems like this will be even easier than I thought.

Maybe; but I still think you'll need to do a new plugin to implement the System76-specific detach (to get the device into bootloader without holding a button) and to get the runtime firmware version -- although perhaps that just the bcdRevision on the USB interface. On my test Pico board I get:

	14:04:06.326 FuMain               FuUf2Device:
	  DeviceId:             6e2cd00486e8398afebdd95cd5b909ae0cb9fa42
	  Name:                 RP2
	  Guid:                 874a17a9-dd06-5e8b-9007-09bc9be85b74 ← BLOCK\VEN_2E8A&DEV_0003
	  Guid:                 2ba00449-c24d-5f0e-8e58-182924f02911 ← BLOCK\VEN_2E8A&DEV_0003&REV_02
	  Guid:                 6dd89c00-e9c8-5add-a69e-9f00fb08f52f ← UF2\FAMILY_E48BFF56
	  Guid:                 08022771-a1e2-55b9-b682-a156b1b4b40d ← USB\VID_2E8A&PID_0003
	  Guid:                 ee5d2b24-742f-5b95-af04-5755e5a619cf ← USB\VID_2E8A&PID_0003&UUID_000A-04D4
	  Guid:                 abb9adbc-77dd-5c23-9ffd-b3039abccb4e ← UF2\BOARD_RPI-RP2
	  Serial:               E0C912952D54
	  Plugin:               uf2
	  Protocol:             com.microsoft.uf2
	  Flags:                updatable|registered|will-disappear|can-verify|can-verify-image|unsigned-payload
	  Vendor:               Raspberry Pi
	  VendorId:             BLOCK:0x2E8A
	  VendorId:             USB:0x2E8A
	  Version:              02
	  VersionFormat:        number
	  Created:              2023-05-05

jackpot51, 2023-05-05:
The detach will be identical to system76-launch plugin, thankfully.

hughsie, 2023-05-05:
Sooo maybe this is going to be super simple. Add a chunk of quirk lines to system76-launch.quirk and see what happens? One tiny nit I can see is that the launch plugin needs `org.usb.dfu` in `fu_system76_launch_device_init()` but maybe just add `com.microsoft.uf2` as well and see if it works?

jackpot51, 2023-05-05:
I will try it, thanks.

hughsie, 2023-05-05:
want me to add BLOCK:0x2E8A to the system76 LVFS vendor restrictions?

jackpot51, 2023-05-05:
Yes, please.

jackpot51, 2023-05-05:
By the way, this will also wrap up two longstanding issues with thelio-io. Firstly, the bootloader reset will be done via HID just like launch, instead of through a CDC-ACM kernel driver. Secondly, it will require physical access just like launch.

hughsie, 2023-05-05:
> want me to add BLOCK:0x2E8A to the system76 LVFS vendor restrictions?

Done.

jackpot51, 2023-05-05:
I have changed the issue to be something more vendor specific. I believe the support for RP2040 is already technically there, with the UF2 plugin, and I just need to add the right quirks and adjust the launch plugin to work with that.

jackpot51, 2023-08-22:
I tried to upload the firmware in [thelio_io_2_0.21.0-65-g0c3e4c.zip](https://github.com/fwupd/fwupd/files/12410309/thelio_io_2_0.21.0-65-g0c3e4c.zip), but got this message:

```
Failed to upload: File is not supported: Metadata is invalid: No valid UpdateProtocol com.microsoft.uf2 found
```

hughsie, 2023-08-22:
Oopa, I needed to mark the protocol as public. Can you try again now please.

jackpot51, 2023-08-22:
Thanks, I was able to upload.

jackpot51, 2023-08-25:
There is one more problem I foresee once #6117 and #6111 are mreged. We currently have the update in embargo-system76. There is one reported problem on page https://fwupd.org/lvfs/firmware/54637/problems:

### Firmware protocol requires fwupd report for ‘Thelio Io Device Update’
com.system76.thelio_io_2.firmware
This firmware requires the uploading ODM or responsible OEM to upload a fwupd success report from the target hardware. 

However, our vendor is set up as "Do Not Track", and my attempt to upload a successful fwupd report has not shown up.

hughsie, 2023-08-25:
Hmm, I'll admit I didn't think about how Do Not Track interacts with the requirement for a signed upload. What happens now when you upload a success report, either signed or anonymous? e.g. if you run fwupd with --verbose what does the server respond with?

jackpot51, 2023-08-25:
```
FuMain-INFO: 09:22:09.302: server message: Certificate serial 7f42f749a692a9c49ccbdf1f571e96ae83836099 did not match user; 62a9ebc24aab1c20e6dbcd52133ec5d7f765eb41 will not accept reports
```

hughsie, 2023-08-25:
@jackpot51 I understand the ideological idea of DoNotTrack -- so we just don't count the users downloading the firmware -- but are you conceptually okay with the idea of accepting QA success reports from vendor that uploaded the report, any 3rd party QA team and unauthenticated random end-users? e.g. my proposal would be just to drop this check: https://gitlab.com/fwupd/lvfs-website/-/blob/master/lvfs/reports/routes.py?ref_type=heads#L311

jackpot51, 2023-08-25:
I am okay with this.

jackpot51, 2023-08-25:
If we are the only vendor using DoNotTrack, and the only thing still being skipped is incrementing a download counter, I don't think it is worth maintaining this feature. You can unset DoNotTrack on our account.

hughsie, 2023-08-25:
> and the only thing still being skipped is incrementing a download counter

Ohh it's a bit more than that; we also keep stats of what countries the firmwares are going out to, the fwupd versions of the requester and that kind of thing. We collect the country data for ITAR/EAR compliance but it's also requested by some vendors. Do you want me to enable it for your account (you're indeed the only vendor using it) and you can switch back if you're not comfortable?

hughsie, 2023-08-25:
I've also deployed https://gitlab.com/fwupd/lvfs-website/-/commit/8ec1e66f59fb203c85bbe27346496030b6d00197 so if you could try the report upload in the meantime and see if that gets accepted. You might have to wait a few mins for the new containers to spin up.

jackpot51, 2023-08-25:
Thanks, I will try uploading again. I am talking about the download data internally.

jackpot51, 2023-08-25:
Uploading the report worked, https://fwupd.org/lvfs/firmware/54637/analytics/reports/2, now it can be moved to Testing or Stable.

hughsie, 2023-08-25:
I think you want a Client fwupd version requirement of 1.9.5, no?

jackpot51, 2023-08-25:
Yep, I will set that.

jackpot51, 2023-08-25:
Done, required fwupd version is now >= 1.9.5

jackpot51, 2023-08-25:
I talked internally, and we specifically do not want to track the number of downloads.

closed 2023-08-25 by jackpot51