fwupd/fwupd #5789: ccgx: HP Thunderbolt Dock G4  update times out with "Message did not receive a reply (timeout by message bus) after 5 minutes

bleungatchromium, 2023-05-06:
**Describe the bug**
A clear and concise description of what the bug is.
Running an "allow-reinstall" update of the HP Thunderbolt G4 results in the dock disconnecting and rebooting, but `fwupd` times out after 5 minutes, rather than waiting the 732 seconds this dock has as a RemoveDelay.
The update is actually successful, and the dock boots up and reconnects to the system after 573 seconds.

**Steps to Reproduce**
Steps to reproduce the behavior.
`fwupdmgr install --allow-reinstall 5bace0ea25c135919a5a3337b25f585d75736f04235276b40fe4331f8c644612-Thunderbolt_Dock_G4_V1.4.16.0`
Wait 5 minutes.

**Expected behavior**
A clear and concise description of what you expected to happen.
The process waits up to the 12.2 minutes that RemoveDelay is defined for this dock.

**fwupd version information**
Please provide the version of the daemon and client.
1.8.

```
runtime   org.freedesktop.fwupd         1.8.13
compile   com.hughsie.libxmlb           0.3.11
compile   org.freedesktop.gusb          0.4.5
compile   com.hughsie.libjcat           0.1.8
runtime   org.kernel                    5.10.179-22448-g82b751593a36-dirty
runtime   org.freedesktop.gusb          0.4.5
compile   org.freedesktop.fwupd         1.8.13

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

As a part of ChromeOS.

<details>

<summary>**HP Dragonfly Pro Chromebook**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version: ChromeOS 115.15445.0
- Have you tried rebooting?
- Is this a regression?
Nope.

hughsie, 2023-05-06:
Hmm. For install we're using `g_dbus_connection_send_message_with_reply()` with a `timeout_msec` of `G_MAXINT` which means *no timeout*. Given I've also not seen this -- and I've done much longer updates than 5 minutes (a Logitech keyboard takes ~20 mins!) I suspect this is a dbus-server timeout that's specific somehow to ChromeOS. Does ChromeOS perhaps set a limit in `/usr/share/dbus-1/system.conf` ?

bleungatchromium, 2023-05-07:
I'll take a look. Thanks for the pointer.

djcampello, 2023-05-09:
I see the following on a dut:

```
localhost ~ # grep -R timeout /usr/share/dbus-1/
/usr/share/dbus-1/session.conf:  <limit name="service_start_timeout">120000</limit>  
/usr/share/dbus-1/session.conf:  <limit name="auth_timeout">240000</limit>
/usr/share/dbus-1/session.conf:  <limit name="pending_fd_timeout">150000</limit>
/usr/share/dbus-1/system.conf:  <!-- <limit name="service_start_timeout">25000</limit> -->
/usr/share/dbus-1/system.conf:  <!-- <limit name="auth_timeout">5000</limit> -->
/usr/share/dbus-1/system.conf:  <!-- <limit name="pending_fd_timeout">150000</limit> -->
/usr/share/dbus-1/interfaces/org.freedesktop.ModemManager1.Modem.xml:        @timeout: The number of seconds to wait for a response.
/usr/share/dbus-1/interfaces/org.freedesktop.ModemManager1.Modem.xml:      <arg name="timeout"  type="u" direction="in"  />
```

I don't think these timeouts are relevant as the one it should be limiting is unset: `reply_timeout`

Logs show issues with the retimer but these seem unrelated and much earlier:

```
2023-05-05T20:18:14.497038Z WARNING fwupd[4758]: 20:18:14.496 FuEngine             failed to add device usb:02:00:03: NVM read error: hub NVM read error: maximum tries exceeded: GET_MMIO failed to set control on mbox register index [0x11]: USB error on device 8087:0b40 : No such device (it may have been disconnected) [-4]
2023-05-05T20:18:14.497836Z WARNING fwupd[4758]: 20:18:14.497 FuEngine             failed to probe device usb:02:00:03:04: USB error on device 03f0:3488 : No such device (it may have been disconnected) [-4]
2023-05-05T20:18:14.498746Z WARNING fwupd[4758]: 20:18:14.498 FuEngine             failed to probe device usb:02:00:03:04:02: USB error on device 03f0:5488 : No such device (it may have been disconnected) [-4]
```

The only log we got from the update attempt:

```
2023-05-05T20:25:31.371508Z ERROR chrome[8213:8213]: [fwupd_client.cc(369)] Firmware install failed with error: org.freedesktop.DBus.Error.NoReply
```

The install call from Chrome is set with infinite timeout: https://source.chromium.org/chromium/chromium/src/+/main:chromeos/ash/components/dbus/fwupd/fwupd_client.cc;l=166

hughsie, 2024-11-11:
I believe we fixed this ChromeOS side, please re-open if incorrect.

closed 2024-11-11 by hughsie