fwupd/fwupd #5794: Upgradeable hardwared detected, then it returns No supported devices found

StorkyStorky, 2023-05-07:
**Describe the bug**
"No supported devices found" even though the following is reported as being updateable to version 20
Devices with the latest available firmware version:
 • Intel Management Engine
 • MZVLB512HAJQ-000L7
 • System Firmware
20L9001HUS
│
└─Thunderbolt host controller:
  │   Device ID:          ff6228e47c17e843c58949d7d4058054f3e4bc7c
  │   Summary:            Unmatched performance for high-speed I/O
  │   Current version:    14.00
  │   Vendor:             Lenovo (THUNDERBOLT:0x0109, TBT:0x0109)
  │   GUIDs:              72021ca5-cd76-5b37-85ca-d39e7184bf77 ← THUNDERBOLT\VEN_0109&DEV_1703
  │                       d0b85dea-a21f-5d1b-a693-55a3cf1bca5e ← THUNDERBOLT\VEN_0109&DEV_1703&REV_00
  │                       5bed8b5c-e5c2-547b-8806-6603930184ee ← TBT-01091703-native
  │                       f69aca31-6f9e-5b23-93ec-9683b02d66eb ← TBT-01091703-native-controller0-0
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Signed Payload

**Steps to Reproduce**
fwupdmgr refresh
fwupdmgr get-updates
fwupdmgr update
fwupdmgr --verbose update
(fwupdmgr:10736): GLib-DEBUG: 10:27:07.649: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:10736): GLib-GIO-DEBUG: 10:27:07.666: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.666: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.666: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:10736): dconf-DEBUG: 10:27:07.667: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:10736): GLib-GIO-DEBUG: 10:27:07.670: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:10736): Fwupd-DEBUG: 10:27:07.673: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ]Devices with no available firmware updates: 
 • UEFI Device Firmware
 • Unifying Receiver
Devices with the latest available firmware version:
 • Intel Management Engine
(fwupdmgr:10736): FuMain-DEBUG: 10:27:07.695: current version is 184.93.4323: 184.93.4323=same, 184.92.4222=older, 184.86.3909=older
 • MZVLB512HAJQ-000L7
(fwupdmgr:10736): FuMain-DEBUG: 10:27:07.700: current version is 5L2QEXA7: 5L2QEXA7=same
 • System Firmware
(fwupdmgr:10736): FuMain-DEBUG: 10:27:07.711: current version is 0.1.34: 0.1.28=older, 0.1.27=older, 0.1.21=older, 0.1.34=same, 0.1.33=older, 0.1.32=older
Decompressing…           [ -                                     ]╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Thunderbolt host controller from 14.00 to 20.00?                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ N27TF20W update PD                                                           ║
║                                                                              ║
║ • Patch bundle upgrade to F407.06.29                                         ║
║                                                                              ║
║ Bug Fixes Key 	Summary                                                       ║
║                                                                              ║
║ TPS65982-14138 	988CE/SNK was occasionally entering an incorrect state when  ║
║ the far-end/SRC prevented the SNK from doing a transmission by holding the   ║
║ CC line at SinkTxNG state, thus preventing the device from resending         ║
[... 248 lines cut ...]
      Bootloader Version: BOT01.04_B0016
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
      Device Flags:       • Updatable
                          • Signed Payload
    
root@pop-os:~# 

</details>

**Additional questions**

- Operating system and version: 
Linux pop-os 6.2.6-76060206-generic #202303130630~1681329778~22.04~d824cd4 SMP PREEMPT_DYNAMIC Wed A x86_64 x86_64 x86_64 GNU/Linux

DISTRIB_ID=Pop
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Pop!_OS 22.04 LTS"

- Have you tried rebooting? Yes
- Is this a regression? I didn't find a similar issue when I searched, but I do make mistakes.

hughsie, 2023-05-07:
Can you try something newer? 1.8.0 is *old*. 1.8.14 was released a few months ago.

thorrrr, 2023-05-08:
Same issue on a XPS 15 9560  will not update whatever i try

hughsie, 2023-05-08:
If you download the .cab archive manually, does `fwupdmgr install foo.cab` work? If not, does `sudo fwupdtool install foo.cab --verbose --verbose` work? If not, can you upload the logs for the latter please.

hughsie, 2023-05-08:
And I also think Dell has done something dumb in the last 9560 upload -- I've emailed them and asked for more details.

thorrrr, 2023-05-08:
> If you download the .cab archive manually, does `fwupdmgr install foo.cab` work? If not, does `sudo fwupdtool install foo.cab --verbose --verbose` work? If not, can you upload the logs for the latter please.

foo.cab nothing found the latter runs but does not fix the issue 
PLease can you tell me location of the logs you need ?

hughsie, 2023-05-08:
Not foo.cab, the name of the downlosded .cab file please.

flipflop97, 2023-05-08:
@hughsie Not working for me unfortunately:

```
$ sudo fwupdtool install fd53933b96c117f21370821e7fc89b30d84176c48fbc616ac533249c2c40d45c-Dell_Precision_5520_System_BIOS_Ver.1.32.0.cab --verbose --verbose

11:43:41.172 FuDebug              verbose to debug (on console 0)
Bezig met laden...       [ -                                     ]Bezig met laden...       [ -                                     ]11:43:41.173 FuMain               locked /run/lock/fwupdtool
11:43:41.175 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
11:43:41.177 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
11:43:41.178 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
11:43:41.178 FuCommon             reading /etc/fwupd/daemon.conf with 2563 bytes
11:43:41.178 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
11:43:41.178 FuConfig             using autodetected max archive size 4,1 GB
11:43:41.178 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for 'gio-vfs'
11:43:41.179 FuRemoteList         loading /usr/share/fwupd/metainfo
11:43:41.179 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
11:43:41.179 XbSilo               file: 2e418662-8683-2734-8333-88f6d4090f2f, current:2e418662-8683-2734-8333-88f6d4090f2f, cached: (null)
11:43:41.179 XbSilo               loading silo with file contents
11:43:41.179 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs.conf.pacnew
11:43:41.179 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
11:43:41.179 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs-testing.conf.pacnew
11:43:41.179 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
11:43:41.179 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
11:43:41.179 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
11:43:41.179 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
11:43:41.179 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
11:43:41.179 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
11:43:41.179 FuRemoteList         ordering lvfs-testing=lvfs+1
11:43:41.179 FuRemoteList         ignoring unfound remote fwupd
11:43:41.179 FuRemoteList         enabled remotes: dell-esrt, lvfs
11:43:41.180 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
11:43:41.180 Jcat                 loaded 1 certificates
11:43:41.180 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
11:43:41.180 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
11:43:41.180 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
11:43:41.180 Jcat                 loaded 1 certificates
11:43:41.180 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
11:43:41.180 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
11:43:41.180 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
11:43:41.180 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
11:43:41.180 FuEngine             client certificate now exists: nothing to do
Bezig met laden...: 1%
11:43:41.180 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
11:43:41.180 FuHistory            got schema version of 8
Bezig met laden...: 2%
11:43:41.185 FuPlugin             load(/usr/lib/fwupd-1.8.14/libfu_plugin_modem_manager.so)
11:43:41.187 FuContext            added udev subsystem watch of platform
11:43:41.187 FuContext            added udev subsystem watch of pci
11:43:41.187 FuContext            added udev subsystem watch of block
11:43:41.187 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
11:43:41.187 FuPluginDell         Using libsmbios 2.4
11:43:41.187 FuContext            added udev subsystem watch of i2c
11:43:41.187 FuContext            added udev subsystem watch of i2c-dev
11:43:41.187 FuContext            added udev subsystem watch of hidraw
11:43:41.187 FuContext            added udev subsystem watch of gpio
11:43:41.187 FuContext            added udev subsystem watch of mei
11:43:41.187 FuContext            added udev subsystem watch of iommu
11:43:41.187 FuContext            added udev subsystem watch of video4linux
11:43:41.187 FuContext            added udev subsystem watch of msr
11:43:41.187 FuContext            added udev subsystem watch of mtd
11:43:41.187 FuContext            added udev subsystem watch of nvme
11:43:41.188 FuContext            added udev subsystem watch of drm
11:43:41.188 FuContext            added udev subsystem watch of drm_dp_aux_dev
11:43:41.188 FuContext            added udev subsystem watch of serio
11:43:41.188 FuPluginTest         init
11:43:41.189 FuContext            added udev subsystem watch of thunderbolt
11:43:41.189 FuContext            added udev subsystem watch of tpm
11:43:41.189 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
11:43:41.189 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
[... cut ...]
  Guid:                 d4a1ecaa-d1b1-5569-a4d1-e8ad3539b038 <- I2C\NAME_dummy ?
  Flags:                none
  PhysicalId:           DEVPATH=/devices/pci0000:00/0000:00:1f.4/i2c-6/6-0036
  BackendId:            /sys/devices/pci0000:00/0000:00:1f.4/i2c-6/6-0036
  AcquiesceDelay:       2500
  BusNumber:            0x6

11:43:42.439 FuEngine             /sys/devices/pci0000:00/0000:00:1f.4/i2c-6/6-0037 added FuI2cDevice:
  Flags:                none
  BackendId:            /sys/devices/pci0000:00/0000:00:1f.4/i2c-6/6-0037
  AcquiesceDelay:       2500
  BusNumber:

closed 2023-05-09 by hughsie

peterablehmann, 2023-12-31:
I've the same Problem on a Lenovo T580.
In version 1.9.9 of fwupd.
```
Laden …                  [****                                   ]20:06:18.402 FuDevice             already registered private FuMmDevice flag with value: use-branch:0x2
Laden …                  [*******************                    ]20:06:19.227 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]
Laden …                  [************************************** ]
Geräte mit keinen verfügbaren Firmware-Aktualisierungen: 
 • TPM
 • UEFI Device Firmware
 • UEFI Device Firmware
Geräte mit der neuesten verfügbaren Firmware-Version:
 • Intel Management Engine
 • MZVLB512HAJQ-000L7
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Thunderbolt host controller von 14.00 auf 20.00 aktualisieren?               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ N27TF20W update PD                                                           ║
║                                                                              ║
║ • Patch bundle upgrade to F407.06.29                                         ║
║                                                                              ║
║ Bug Fixes Key         Summary                                                       ║
║                                                                              ║
║ TPS65982-14138        988CE/SNK was occasionally entering an incorrect state when  ║
║ the far-end/SRC prevented the SNK from doing a transmission by holding the   ║
║ CC line at SinkTxNG state, thus preventing the device from resending         ║
║ pending PD commands on the CC line.                                          ║
║                                                                              ║
║ TPS65982-14485        988CE/SNK was occasionally setting the Overcurrent flag in   ║
║ register 0x5F (Data Status Register) when FRS was enabled and the far end    ║
║ was a source capable dongle/dock, due to a spurious interrupt upon           ║
║ disconnect.                                                                  ║
║                                                                              ║
║ TPS65982-4314         VCONN acceptable range for cables is 4.75V-5.5V               ║
║                                                                              ║
║ Thunderbolt host controller und alle angeschlossenen Geräte sind während     ║
║ der Aktualisierung möglicherweise nicht nutzbar.                             ║
╚══════════════════════════════════════════════════════════════════════════════╝
Operation durchführen? [Y|n]: y
Entpacken …              [ -                                     ]
No supported devices found
 • UEFI dbx
```

hughsie, 2024-01-02:
@peterablehmann that's `fwupdtool`, not `fwupdmgr` -- can you try with the latter please.

peterablehmann, 2024-01-02:
Same thing

superm1, 2024-01-02:
@peterablehmann please open a new issue, and attach the output for `sudo fwupdtool update --verbose --verbose`

peterablehmann, 2024-01-04:
Setting the verbose flag fixed the issue. I'm not sure why.