fwupd/fwupd #5795: Dell Precision 7560,Precision 7760 - checksum invalid

skipperTux, 2023-05-08:
**Describe the bug**
Upgrade System Firmware from 1.20.0 to 1.21.1 not possible on Fedora Linux 37 (Workstation Edition) on a Dell Inc. Precision 7560 because of a `checksum invalid` error message.

```shell
$ fwupdmgr update
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • USB2.0 Hub
 • Unifying Receiver
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19DCS
 • PM9A1 NVMe Samsung 2048GB
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.20.0 to 1.21.1?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • This release contains security updates as disclosed in the Dell            ║
║ Security Advisorie                                                           ║
║ • Firmware updates to address security vulnerabilities including (Common     ║
║ Vulnerabilities and Exposures - CVE)                                         ║
║ • Fixed the issue where the system cannot boot.                              ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ Precision 7560 must remain plugged into a power source for the duration of   ║
║ the update to avoid damage.                                                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y       
Downloading…             [************************************** ]
checksum invalid, expected 0a7b5a9fa567bda9554bef2cee407a80cb026bec56d981abaa9ac689d75cfc8c got a6e3bd7909ba552a63a9c1e04dd196f2647ba7e1659545fdb9c22e96f0925b95
```

**Steps to Reproduce**
Allow Firmware upgrades in System BIOS. Then do

```shell
$ fwupdmgr refresh
$ fwupdmgr get-updates
$ fwupdmgr update
```

**Expected behavior**
After Downloading the firmware 1.21.1, the system shall reboot and do the system firmware upgrade.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.14
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.14
runtime   org.kernel                    6.2.14-200.fc37.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `dnf`

<details>
[... 281 lines cut ...]
  │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
  │   Current version:    0x00000040
  │   Vendor:             Intel
  │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
  │                       30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
  │                       a6bd4ca5-75a6-5796-b564-66b5cab1b11b ← CPUID\PRO_0&FAM_06&MOD_8D
  │                       d9dd5e77-df17-5bab-b5ec-22827598bfed ← CPUID\PRO_0&FAM_06&MOD_8D&STP_1
  │   Device Flags:       • Internal device
  │ 
  └─USB4 host controller:
        Device ID:        e31d005c05adb800405a184422de8cbe4d566e2e
        Summary:          Unmatched performance for high-speed I/O
        GUID:             e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
        Device Flags:     • Internal device
                          • System requires external power source
                          • Signed Payload
```

</details>

**Additional questions**

- Operating system and version:  Fedora Linux 37 (Workstation Edition)
- Have you tried rebooting? Yes
- Is this a regression? Kind of, had this in the past.

hughsie, 2023-05-08:
If you do `fwupdmgr refresh --force` does the `fwupdmgr update` message go away? If you download the file using wget or curl (use fwupdmgr update --verbose to see the URL) does it contain anything like a proxy login HTML page or something?

skipperTux, 2023-05-08:
Running `fwupdmgr refresh --force` does not make the message go away (I did this before, but missed to document it in my report). Running `fwupdmgr update --verbose` gives the following URL

```shell
$ fwupdmgr update --verbose
[snip]
Perform operation? [Y|n]: y
Fwupd-INFO: 10:39:09.786: downloading https://fwupd.org/downloads/a6e3bd7909ba552a63a9c1e04dd196f2647ba7e1659545fdb9c22e96f0925b95-Precision_7X60_1.21.1.cab
(fwupdmgr:5005): Fwupd-DEBUG: 10:39:09.786: Emitting ::status-changed() [downloading]
[snip]
load progress: 100%
(fwupdmgr:5005): Fwupd-DEBUG: 10:39:12.823: Emitting ::status-changed() [idle]
Fwupd-INFO: 10:39:12.823: status-code was 200

checksum invalid, expected 0a7b5a9fa567bda9554bef2cee407a80cb026bec56d981abaa9ac689d75cfc8c got a6e3bd7909ba552a63a9c1e04dd196f2647ba7e1659545fdb9c22e96f0925b95
```

The invalid checksum comes from the download URL, no proxy login HTML is in place. It is the same checksum that can be found [LVFS: Dell Precision 7560,Precision 7760](https://fwupd.org/lvfs/devices/com.dell.uefib9870160.firmware) page.

hughsie, 2023-05-08:
Mea culpa; https://gitlab.com/fwupd/lvfs-website/-/commit/f98ebc4da3801b8a2cf49c159dc6219c789eaecd

Just deploying now.

hughsie, 2023-05-08:
Okay, can you try `fwupdmgr refresh --force` please.

skipperTux, 2023-05-08:
Yes, works now. System Firmware Upgrade done. Thank you for the fast response and fix, very appreciated. Case closed.

closed 2023-05-08 by skipperTux