fwupd/fwupd #5800: Retimer firmware upgrade through EC

Xin-ANX, 2023-05-10:
**Is your feature request related to a problem? Please describe.**
We have USB4/DP retimer chip (ANX7452) which bridged between CPU and USB Type-C port on Chromium OS, there is only a I2C bus connected between EC and retimer chip (ANX7452). For the retimer chip (ANX7452) flash access(firmware upgrade), APP cannot directly send I2C command, it must send the I2C request to EC first, then EC transmit it to retimer chip (ANX7452).

**Describe the solution you'd like**
Based on the currently fwupd, there is no interface to send I2C command over EC. There are 2 method to implement:
1. Develop a independent retimer update tool and fwupd plugin, once retimer firmware upgrade request received, lauch fwupd plugin to call this independent retimer update tool.
2. Integrate I2C access over EC feature in fwupd library which can be called by fwupd plugin.

**Describe alternatives you've considered**

**Additional context**

prashantmalani, 2023-05-10:
Thanks for writing this up Xin.

I would add that the other thing to consider here is that the i2c address is not fixed; it will likely be a per-board setting.. This is not an issue if the plugin will call and independent tool. However, if we want to run the host commands to the EC (via ioctl) directly from the plugin, there will need to be a way to supply the i2c address to the plugin through some kind of config/parameter.

hughsie, 2023-05-10:
> it must send the I2C request to EC first, then EC transmit it to retimer chip

Do you have other I2C devices connected to the EC that you want to enumerate and/or update?

> This is not an issue if the plugin will call and independent tool

That's not something that can happen, and lots of vendors have asked. It's probably easier for you, but impossible for us to maintain. See https://lvfs.readthedocs.io/en/latest/custom-plugin.html for a lot more rationale.

> there will need to be a way to supply the i2c address to the plugin through some kind of config/parameter

Is there a way to query the EC to discover child devices? That's much better than hardcoding per-platform.

prashantmalani, 2023-05-10:
Hi Richard,

Thanks for looking at this request.

> Is there a way to query the EC to discover child devices? That's much better than hardcoding per-platform.

I don't believe so. FWIU, there is no host command for it, but more pertinently:
- i2c scanning can be potentially dangerous , since there is no format which all devices follow , so the device can potentially be left in a bad state.

superm1, 2023-05-10:
For this design I think it would be better to encode the I2C address in DT or ACPI and let the kernel layer completely abstract it from userspace.  Then userspace can:
1. call an ioctl to discover if retimer is present
2. call an ioctl to discover retimer FW version
3. call an ioctl to upgrade retimer FW

This would mean the fwupd plugin is much more simple and the larger amount of the work lives in the kernel driver (I guess `cros_ec`).  It also will mean that userspace doesn't have open access to the I2C interface, but rather these narrowed down uses.

Xin-ANX, 2023-05-11:
Hi superm1, OK, I can develop a plugin module which can access retimer through ioctl based on "cros_ec" device. But I need 2 extra paramters input: USBC port id and retimer i2c slave address. As these 2 parameters depends on board/platform, can you suggest where can put these 2 parameter and fwupd can easy get them?

superm1, 2023-05-11:
How about it work's like this:

* R IOCTL to discover all `cros_ec` retimers.  The returned structure can be a an integer with number of ioctls and information to disambiguate which is which (port ID?).
* fwupd when it calls this IOCTL will use it to build X number of fwupd devices and use the port ID for building a GUID.

* R/W IOCTL to discover firmware version for a given port ID.
* fwupd will call this IOCTL with an argument of the port ID and the returned structure will have a GUINT32 firmware version.

* W IOCTL for staging firmware update payload
* fwupd will allocate a memory buffer with payload and stage it in this location.  It will call IOCTL and then cros_ec will start firmware flash

* sysfs file to show firmware upgrade status, use `sysfs_notify` to indicate status changes.  `cros_ec` can choose intervals to report status to this file.
* fwupd can wait for this file to change to update progress bar and for when flashing is complete.

With that design the i2c slave address and corresponding port Id are stored in DT/ACPI, cros_ec parses them.  When it gets IOCTL call for enumerating them it can return a structure with that info or an error if it doesn't exist.

Xin-ANX, 2023-05-12:
Hi superm1, OK, thanks for your advice, I'll try to do it.

closed 2024-11-11 by hughsie