fwupd/fwupd #5804: Explore using/creating snap `tracks`

superm1, 2023-05-10:
[Snap tracks](https://ubuntu.com/blog/controlling-snap-releases-with-channels-tracks-and-branches-part-2) are used for any software supporting parallel major releases.

As fwupd releases 1.7.x, 1.8.x, 1.9.x releases it might make sense to also create/use tracks for matching the `1_8_X` branch, `main` branch etc.

This can help mitigate risk for anyone wanting to match the behavior of a branch.

The unfortunate thing is that to open up a new track it means that a [store request needs](https://snapcraft.io/docs/channels?_ga=2.9760368.470174452.1683727100-1988020464.1683605976) to be opened up for each new track. 
This would mean every time we _branch_ to a new release a new store request needs to also be opened too.

@valentindavid @CragW @hughsie @hugh712 thoughts?

hugh712, 2023-05-11:
If the tracks were created, any plan to retire the old tracks in the future?

superm1, 2023-05-11:
The tracks would reflect the same as upstream. So if no changes to 1.8.x branch any more no more content in the track.

hugh712, 2023-05-11:
then I think we can also leverage on the tracks to isolate regression issue,
+1

superm1, 2023-12-21:
I've opened a request for making a 1.9.x track: https://forum.snapcraft.io/t/create-track-for-fwupd/38220

closed 2024-01-15 by superm1