fwupd/fwupd #5811: vid/pid for CE-LINK01 TBT4 dock (cypress)

alebrati, 2023-05-11:
(see attached file), had a talk with Benson on this.
Dock: CE-LINK TB4-Dock01
issue: The PD (cypress) is not detected over fwupd. as per my record, the vid/pid are :
  "vid=2095"
  "pid=4D01"

fwupd version used : 1.8.12
ChromeOS: R113-15393.46.0
platform: Redrix

fwupd output :

localhost /home # fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.11
compile   org.freedesktop.gusb          0.4.5
compile   com.hughsie.libjcat           0.1.8
runtime   org.kernel                    5.10.175-22068-g4b67de084816
runtime   org.freedesktop.gusb          0.4.5
compile   org.freedesktop.fwupd         1.8.12

localhost /home # fwupdmgr get-devices
Google Redrix
?
??ELAN2703:00 04F3:323B:
? ?   Device ID:          1f532bc465a37e0384a117d5f2da9c04a79b5de5
? ?   Summary:            Touchpad
? ?   Current version:    0x000c
? ?   Bootloader Version: 0x0003
? ?   Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
? ?   GUIDs:              0a6e154e-7e6e-5b7e-ae3c-06bc247df555 ? HIDRAW\VEN_04F3&DEV_323B
? ?                       e3af00d0-57bf-5651-88c0-7314727f721a ? HIDRAW\VEN_04F3&DEV_323B&REV_00
? ?                       bfae2c46-3ffc-53eb-a994-b6c37c27b4b8 ? HIDRAW\VEN_04F3&DEV_323B&MOD_0130
? ?                       4f9fe0ae-f26e-5c1d-8a4e-a7506fb02f96 ? ELANTP\ICTYPE_13
? ?                       4184508c-a984-5a3c-893c-6bae98f2fcf4 ? ELANTP\ICTYPE_13&MOD_0130
? ?                       ba73bfad-c59c-5d19-82ae-c730143b5979 ? ELANTP\ICTYPE_13&MOD_0130&DRIVER_HID
? ?   Device Flags:       ? Internal device
? ?                       ? Updatable
? ?
? ??HapticPad EEPROM:
?       Device ID:        ae348b09411b8010d974d966a6434e4f69332ad7
?       Current version:  22040802
?       Bootloader Version:0x0001
?       Vendor:           ELAN Microelectronics (HIDRAW:0x04F3)
?       GUIDs:            58279a02-cb82-5c04-a87b-8b7302187077 ? HIDRAW\VEN_04F3&DEV_323B&DRIVERIC_0002&MOD_0130
?                         4f9fe0ae-f26e-5c1d-8a4e-a7506fb02f96 ? ELANTP\ICTYPE_13
?                         3a4b6035-912d-5c64-a1d2-4bcba1686e16 ? ELANTP\ICTYPE_13&DRIVERIC_0002&MOD_0130
?                         35d08e43-460d-57b6-ba3c-ec671a4cdc9d ? ELANTP\ICTYPE_13&DRIVERIC_0002&MOD_0130&DRIVER_HID
?       Device Flags:     ? Internal device
?                         ? Updatable
?
??Display controller:
?     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
?     Current version:    0c
?     Vendor:             PCI:0x8086
?     GUIDs:              86b8e8a0-4c9c-5607-96a7-04073caa6063 ? PCI\VEN_8086&DEV_46B3
?                         af3ee7cb-cacc-57a0-8b4d-0f516ed2eba0 ? PCI\VEN_8086&DEV_46B3&REV_0C
?                         d8f3a7d0-9f81-5ec6-bb0e-a295b84aa6b0 ? PCI\VEN_8086&DEV_46B3&SUBSYS_808646B3
?                         348fe5b5-7c38-57bf-890c-75d0c507d269 ? PCI\VEN_8086&DEV_46B3&SUBSYS_808646B3&REV_0C
?                         0223f94c-a22f-5abc-9e50-dffff268313a ? PCI\VEN_8086&DEV_46B3&REV_00
?                         7b5c6ed9-13cd-5a7f-972a-c4ab9233bd83 ? PCI\VEN_8086&DEV_46B3&SUBSYS_808646B3&REV_00
?     Device Flags:       ? Internal device
?                         ? Cryptographic hash verification is available
?
??Integrated Webcam?:
?     Device ID:          08d460be0f1f9f128413f816022a6439e0078018
?     Summary:            Fake webcam
?     Current version:    1.2.2
?     Minimum Version:    1.2.0
?     Bootloader Version: 0.1.2
?     Vendor:             ACME Corp. (USB:0x046D)
?     GUID:               b585990a-003e-5270-89d5-3705a17f9a43
?     Device Flags:       ? Updatable
?                         ? System requires external power source
?                         ? Supported on remote server
?                         ? Cryptographic hash verification is available
?                         ? Unsigned Payload
?
??USB3.0 Hub:
?     Device ID:          cf294bf55b333004beb7c41f952c1838c23e1f4a
?     Current version:    33.0
?     Vendor:             Intel Corporation. (USB:0x8087)
?     GUIDs:              b3f8b518-ebdd-5969-9b44-6b096eff5b9b ? USB\VID_8087&PID_0B40
?                         27f4ed60-ed53-52a7-ae28-ea87d9a23302 ? USB\VID_8087&PID_0B40&REV_1234
?                         c6e7a14f-f13c-55e5-99de-d59461db1d27 ? USB\VID_8087&PID_0B40&HUB_0000
?                         4d26a9b0-49b1-5e13-8aa1-a8e8908287c6 ? TBT-00824d01
?     Device Flags:       ? Updatable
?                         ? Signed Payload
?
??USB4 Retimer:
      Device ID:          16d35c510618116e7899ae45e62605572872e713
      Summary:            A physical layer protocol-aware, software-transparent extension device that forms two separate electrical link segments
      Current version:    10.00
      Vendor:             Intel (THUNDERBOLT:0x8087)
      GUIDs:              e6f2e4ac-f76a-5fc4-b36a-6f9cb548689e ? THUNDERBOLT\VEN_8087&DEV_15EE
                          f00257c8-0e09-5a32-bbb3-f49748ae6cc5 ? THUNDERBOLT\VEN_8087&DEV_15EE&REV_00
                          6d71ac25-b368-5e7b-ac62-32df365fa640 ? TBT-808715ee-retimer0-0:1.1
      Device Flags:       ? Internal device
                          ? Updatable
                          ? Device stages [updates]([lsusb_verbose.log](https://github.com/fwupd/fwupd/files/11451865/lsusb_verbose.log))
[dmesg.log](https://github.com/fwupd/fwupd/files/11451866/dmesg.log)

alebrati, 2023-05-11:
attaching lsusb --verbose as well
[lsusb_verbose.log](https://github.com/fwupd/fwupd/files/11451919/lsusb_verbose.log)

alebrati, 2023-05-15:
Hi Richard,
do we have any estimation on which fwupd version this will be in ?

alebrati, 2023-05-15:
i think thats what i need in order to detect it via the chromebook.
then cab files should be prepared by the vendor and uploaded to lvfs

hughsie, 2023-05-15:
Is it DMC or HPI? e.g. the info we need is something like:

	# Anker Thunderbolt4 Mini Dock
	[USB\VID_291A&PID_8398]
	Plugin = ccgx
	GType = FuCcgxDmcDevice
	Summary = Dock Management Controller Device
	CcgxDmcTriggerCode = 0x01
	RemoveDelay = 732000

alebrati, 2023-05-15:
honestly i dont know . thats a question to the vendor? i have his mail kai.luo@ce-link.com.

closed 2024-11-11 by hughsie