fwupd/fwupd #5812: "Failed to download, server response was 412" when updating Firmware for the Logitech Unifying Receiver

mattst88, 2023-05-11:
**Describe the bug**
Attempting to update the `Firmware for the Logitech Unifying Receiver` from lvfs-testing produces a `Failed to download, server response was 412` error message.

**Steps to Reproduce**
```shell
fwupdmgr enable-remote lvfs-testing
fwupdtool update 3801e1f7043b9abdf1f873bf0bf0b945402e4238
````

**Expected behavior**
Firmware should download and flash to device.

**fwupd version information**

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.10
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.10
runtime   org.kernel                    6.1.12-gentoo-dist
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

On Gentoo, `emerge sys-apps/fwupd`:

````
[ebuild   R    ] sys-apps/fwupd-1.8.10::gentoo  USE="archive bluetooth gnutls gusb logitech lzma man nvme policykit sqlite systemd tpm uefi -amt -bash-completion -cbor -dell (-elogind) -fastboot -flashrom -gtk-doc -introspection -minimal -modemmanager -spi -synaptics -test" PYTHON_SINGLE_TARGET="python3_11 -python3_10" 0 KiB
````

**fwupd device information**

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Framework Laptop
│
├─Unknown Device:
│     Device ID:          63223cc2d1b0b8d4c45ef8cedcb1f45315d3ff20
│     GUID:               ca5c154c-a46a-59d8-bf77-0ca40522cc85 ← GPIO\ID_INT34C5:00
│   
├─11th Gen Intel Core™ i7-1165G7 @ 2.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000a6
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│                         1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
│     Device Flags:       • Internal device
│   
├─Fingerprint Sensor:
│     Device ID:          4295296d98b3ba38c72f6baa33d24f03a1d428f6
│     Summary:            Match-On-Chip fingerprint sensor
│     Current version:    01000248
│     Vendor:             Goodix (USB:0x27C6)
│     Install Duration:   10 seconds
│     Serial Number:      UID648BA478_XXXX_MOC_B0
│     GUIDs:              1e8c8470-a49c-571a-82fd-19c9fa32b8c3 ← USB\VID_27C6&PID_609C
│                         34def4c7-9461-5a32-a945-5dde0ca57d88 ← USB\VID_27C6&PID_609C&REV_0100
│     Device Flags:       • Updatable
│                         • Device can recover flash failures
│                         • Signed Payload
│   
├─SSD 960 PRO 1TB:
[... 157 lines cut ...]
````
# fwupdtool update 3801e1f7043b9abdf1f873bf0bf0b945402e4238 
Loading…                 [-                                      ]
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Unifying Receiver from RQR12.09_B0030 to RQR12.11_B0032?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This release addresses an encrypted keystroke injection vulnerability sent   ║
║ by pointing devices.The vulnerability is complex to replicate and would      ║
║ require a hacker to be physically close to a target.                         ║
║                                                                              ║
║ Unifying Receiver and all connected devices may not be usable while          ║
║ updating.                                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Writing…                 [***************************************]
Downloading…             [***************************************]
Failed to download, server response was 412
````

[fwupdtool update 3801e1f7043b9abdf1f873bf0bf0b945402e4238](https://github.com/fwupd/fwupd/files/11453335/log.txt)
[fwupdtool update --verbose 3801e1f7043b9abdf1f873bf0bf0b945402e4238](https://github.com/fwupd/fwupd/files/11453363/log-verbose.txt)
[fwupdtool get-updates 3801e1f7043b9abdf1f873bf0bf0b945402e4238](https://github.com/fwupd/fwupd/files/11453356/get-updates.txt)

`wget https://fwupd.org/downloads/bd7aa77ee66f98527df74fae678ced05992dc1f9f5ca4e3aac5f21ce339ea5f5-Logitech-Unifying-RQR12.11_B0032.cab` works.

hughsie, 2023-05-11:
What's your public IP address please?

hughsie, 2023-05-11:
Ohh no, ignore me -- 429 is when we've detected abuse.

hughsie, 2023-05-11:
What does `fwupdtool update 3801e1f7043b9abdf1f873bf0bf0b945402e4238 --verbose --verbose` say?

mattst88, 2023-05-11:
[fwupdtool update 3801e1f7043b9abdf1f873bf0bf0b945402e4238 --verbose --verbose](https://github.com/fwupd/fwupd/files/11453518/log-verbose.txt)

hughsie, 2023-05-11:
Out of interest, why fwupdtool and not fwupdmgr?

mattst88, 2023-05-11:
Only because I can never remember which one is the one I should use :)

mattst88, 2023-05-11:
Using `fwupdmgr` worked!

[fwupdmgr update 3801e1f7043b9abdf1f873bf0bf0b945402e4238 --verbose --verbose](https://github.com/fwupd/fwupd/files/11453565/log-verbose.txt)

I did not expect that. Sorry if this has now prevented further debugging /o\

hughsie, 2023-05-11:
I fear it might have done, as fwupdtool works fine for me.  It's weird it was just 412 -- we normally have an error message with more details.

mattst88, 2023-05-11:
If there's anything else I can do, let me know. Can I downgrade the firmware and try to reproduce the issue again?

If there's not an avenue for further debugging, feel free to close the issue.

hughsie, 2023-05-11:
I'm trying to work out what user agent was sent -- so if you can downgrade and strace the fwupdtool output that might give us some clues

closed 2024-11-11 by hughsie