fwupd/fwupd #5813: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached

jedis00, 2023-05-11:
When I login to my GNOME desktop environment on Rocky Linux 9.1, I am getting the below error in a notification window when it tries to check for updates. I am running in an unprivileged Proxmox LXC container that otherwise seems to work fine.

Any ideas on how to troubleshoot this or proceed forward?

"Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached"
```
[me@box~]$ sudo /usr/libexec/fwupd/fwupd
[sudo] password for me: 
15:28:26:0991 FuPluginUefiCapsule  SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specification is unsupported, but /sys/firmware/efi exists: offset bigger than size 9
15:28:26:0995 FuPluginUefiCapsule  cannot find default ESP: no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b: no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
15:28:27:0032 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:15.0: failed to add device using on pci_bcr: could not read BCR: failed to read from port 0x00dc: No such file or directory
15:28:27:0034 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:15.1: failed to add device using on pci_bcr: could not read BCR: failed to read from port 0x00dc: No such file or directory
15:28:27:0036 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:16.0: failed to add device using on pci_mei: could not read HFS1: failed to read from port 0x0040: No such file or directory
15:28:27:0039 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:19.0: failed to add device using on pci_bcr: could not read BCR: failed to read from port 0x00dc: No such file or directory
15:28:27:0040 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:19.1: failed to add device using on pci_bcr: could not read BCR: failed to read from port 0x00dc: No such file or directory
15:28:27:0047 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:1e.3: failed to add device using on pci_bcr: could not read BCR: failed to read from port 0x00dc: No such file or directory
15:28:27:0049 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:1f.0: failed to add device using on pci_bcr: could not read BCR: failed to read from port 0x00dc: No such file or directory
15:28:27:0056 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:1f.5: failed to add device using on pci_bcr: could not read BCR: failed to read from port 0x00dc: No such file or directory
15:28:27:0060 FuEngine             failed to add device /sys/devices/LNXSYSTM:00/LNXSYBUS:00/MSFT0101:00/tpm/tpm0: failed to initialize TPM library
15:28:27:0075 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:06.0/0000:01:00.0/nvme/nvme0: failed to open /dev/nvme0: No such file or directory
15:28:27:0117 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/drm_dp_aux0: failed to add device using on synaptics_mst: failed to open /dev/drm_dp_aux0: No such file or directory
15:28:27:0117 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-2/drm_dp_aux1: failed to add device using on synaptics_mst: failed to open /dev/drm_dp_aux1: No such file or directory
```

jedis00, 2023-05-11:
Verbose:

```
[me@box ~]$ sudo /usr/libexec/fwupd/fwupd --verbose
15:31:04:0424 FuDebug              Verbose debugging enabled (on console 1)
15:31:04:0464 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
15:31:04:0464 FuCommon             reading /etc/fwupd/daemon.conf with 1930 bytes
15:31:04:0464 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
15:31:04:0464 FuConfig             using autodetected max archive size 18.4 EB
15:31:04:0464 FuRemoteList         loading /usr/share/fwupd/metainfo
15:31:04:0464 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
15:31:04:0464 XbSilo               file: 11b334a1-0bae-12ea-dc35-55cfcf2dac18, current:11b334a1-0bae-12ea-dc35-55cfcf2dac18, cached: (null)
15:31:04:0464 XbSilo               loading silo with file contents
15:31:04:0464 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
15:31:04:0464 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
15:31:04:0464 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
15:31:04:0464 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
15:31:04:0464 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
15:31:04:0464 FuRemoteList         ordering lvfs-testing=lvfs+1
15:31:04:0464 FuRemoteList         ignoring unfound remote fwupd
15:31:04:0464 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
15:31:04:0464 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:31:04:0464 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
15:31:04:0464 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
15:31:04:0464 Jcat                 loaded 1 certificates
15:31:04:0464 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:31:04:0464 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
15:31:04:0464 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
15:31:04:0465 Jcat                 loaded 1 certificates
15:31:04:0465 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
15:31:04:0465 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
15:31:04:0465 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
15:31:04:0468 FuEngine             client certificate exists and working
15:31:04:0468 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
15:31:04:0468 FuHistory            got schema version of 7
15:31:04:0468 FuIdle               setting timeout to 7200s
15:31:04:0468 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
15:31:04:0470 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
15:31:04:0470 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
15:31:04:0470 XbSilo               file: 933455b4-7909-6335-516d-acca16a3cd70, current:933455b4-7909-6335-516d-acca16a3cd70, cached: (null)
15:31:04:0470 XbSilo               loading silo with file contents
15:31:04:0471 FuSmbios             ignoring Failed to open file “/sys/firmware/dmi/tables/smbios_entry_point”: Permission denied
15:31:04:0471 FuSmbios             trying to read /sys/class/dmi/id
15:31:04:0471 FuSmbios             unable to read SMBIOS data from /sys/class/dmi/id/product_serial: Failed to open file “/sys/class/dmi/id/product_serial”: Permission denied
15:31:04:0471 FuSmbios             unable to read SMBIOS data from /sys/class/dmi/id/board_serial: Failed to open file “/sys/class/dmi/id/board_serial”: Permission denied
15:31:04:0471 FuSmbios             unable to read SMBIOS data from /sys/class/dmi/id/chassis_serial: Failed to open file “/sys/class/dmi/id/chassis_serial”: Permission denied
15:31:04:0471 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
15:31:04:0471 FuHwids              not loading HwId overrides from /var/lib/fwupd/daemon.conf
15:31:04:0471 FuHwids              smbios property Manufacturer=AZW
15:31:04:0471 FuHwids              smbios property EnclosureKind=4
15:31:04:0471 FuHwids              smbios property Family=MiniPC
15:31:04:0471 FuHwids              smbios property ProductName=SEi
15:31:04:0471 FuHwids              smbios property ProductSku=ADLP
15:31:04:0471 FuHwids              smbios property BiosVendor=American Megatrends International, LLC.
15:31:04:0471 FuHwids              smbios property BiosVersion=DB20A014
15:31:04:0471 FuHwids              ignoring BiosMajorRelease: offset bigger than size 9
15:31:04:0471 FuHwids              ignoring BiosMinorRelease: offset bigger than size 9
15:31:04:0471 FuHwids              ignoring FirmwareMajorRelease: offset bigger than size 9
15:31:04:0471 FuHwids              ignoring FirmwareMinorRelease: offset bigger than size 9
15:31:04:0471 FuHwids              smbios property BaseboardManufacturer=AZW
15:31:04:0471 FuHwids              smbios property BaseboardProduct=SEi
15:31:04:0471 FuHwids              HardwareID-0 is not available, not available as 'BiosMajorRelease' unknown
15:31:04:0471 FuHwids              HardwareID-1 is not available, not available as 'BiosMajorRelease' unknown
[... cut ...]
15:31:04:0674 GLib-GIO             _g_io_module_get_default: Found default implementation dbus (GMemoryMonitorDBus) for ‘gio-memory-monitor’
15:31:04:0675 FuMain               Daemon ready for requests (locale en_US.UTF-8)
15:31:04:0678 FuMain               acquired name: org.freedesktop.fwupd
15:31:09:0947 FuPluginThunderbolt  failed to check usb4 port path: failed to find /sys/devices/pci0000:00/0000:00:0d.2/domain0/0-0/usb4_port1/offline
15:31:09:0947 FuPluginThunderbolt  failed to check usb4 port path: failed to find /sys/devices/pci0000:00/0000:00:0d.3/domain1/1-0/usb4_port1/offline
```

hughsie, 2023-05-11:
> I am running in an unprivileged Proxmox LXC container that otherwise seems to work fine

Does it work when **not** using Proxmox?

jedis00, 2023-05-11:
> 

Proxmox is the hypervisor -- don't really have a way to test without it.

hughsie, 2023-05-11:
so it starts super quick; when you get a failure, does systemctl give an error about the service?

jedis00, 2023-05-11:
Is there a way to disable it so the error doesn't popup on login to GNOME desktop env? I still want it to detect and prompt to install regular package updates, but it can skip updating firmware related things.

hughsie, 2023-05-11:
Off hand, no idea, sorry.

superm1, 2023-05-11:
If you're not using it just mask the service that starts it or uninstall it.

riffer33, 2023-06-21:
I get basically the same error running the snap in Solus Gnome:

fwupdmgr refresh
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

hughsie, 2023-06-21:
@riffer33 what does `sudo /usr/libexec/fwupd/fwupd --verbose` say?

superm1, 2023-06-21:
For the snap, please use candidate channel.

riffer33, 2023-06-21:
Hi.  This is the candidate channel.  The stable won't install on Solus similarly to issue #5122.

sudo: /usr/libexec/fwupd/fwupd: command not found

me@me ~ $ fwupdmgr get-devices --verbose
(/snap/fwupd/4959/bin/fwupdmgr:22364): GLib-DEBUG: 10:17:34.943: setenv()/putenv() are not thread-safe and should not be used after threads are created
(process:22387): GLib-DEBUG: 10:17:34.946: Failed to set thread scheduler attributes: Operation not permitted
(/snap/fwupd/4959/bin/fwupdmgr:22364): GLib-GIO-DEBUG: 10:17:34.964: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
(/snap/fwupd/4959/bin/fwupdmgr:22364): GLib-GIO-DEBUG: 10:17:34.964: _g_io_module_get_default: Found default implementation keyfile (GKeyfileSettingsBackend) for ‘gsettings-backend’
(/snap/fwupd/4959/bin/fwupdmgr:22364): GLib-GIO-DEBUG: 10:17:34.964: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(/snap/fwupd/4959/bin/fwupdmgr:22364): GLib-DEBUG: 10:17:34.964: Failed to set thread scheduler attributes: Operation not permitted
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

I'm a bit of a newb here.  I did learn about --verbose today though :)

superm1, 2023-10-02:
This should be fixed by the current snap in the stable channel.  If you have problems still, please re-open with new details.