fwupd/fwupd #5815: Implement Microsoft guidance on blocking vulnerable Windows boot managers from fwupd

superm1, 2023-05-12:
Microsoft has published the following guidance for blocking vulnerable Windows boot managers:
https://support.microsoft.com/en-us/topic/kb5027455-guidance-for-blocking-vulnerable-windows-boot-managers-522bb851-0a61-44ad-aa94-ad11119c5e91

They explicitly suggest that systems that don't dual boot should apply mitigations immediately.
> For Secure Boot enabled systems that are only booting non-Windows operating systems
> For systems that are only starting non-Windows operating systems and will never start Windows, these mitigations can be applied to the system immediately.

The mitigations are basically creating a set of UEFI variables and applying the DBX revocations.  From the fwupd side, here is my proposal:

1. We add support for fwupdx64.efi to be able to create these variables. 
2. To avoid bumping the fwupd EFI binary interface we can assign a special "caspule" GUID for this behavior. 
3. We should encode in the EFI binary whether it supports this feature (so fwupd daemon can know whether to do it).
4. We modify the fwupd dbx plugin to have a "dbx sub device" that is able to trigger a capsule update with the special GUID.
5. The dbx plugin already examines for vulnerable bootloaders on the ESP filesystem, so no extra checks are needed for creating these variables.
6. When the variables are created or present the subdevice will have version 1.  When they're missing the subdevice will have version 
7. This subdevice would take a dummy capsule from LVFS with no content.

superm1, 2023-05-12:
The tough thing I think about this though is this point: 
>  and will never start Windows

Which means we need to have something in the release notes that makes it clear that effectively all Windows boot media from before May 2023 stops working with secure boot enabled once this update is installed.

Wack0, 2023-05-13:
As discoverer of the initial baton drop bug (among others) and as a researcher of the Windows boot environment:

CiPolicy was never intended to revoke bootmgr itself. Expect at least another round of revocation later (if it's even possible with lack of dbx space).

If a system will never run Windows, then there is no reason for it to have the Windows producton signing cert (Windows Production PCA 2011) in db.

superm1, 2023-05-13:
> If a system will never run Windows, then there is no reason for it to have the Windows producton signing cert (Windows Production PCA 2011) in db.

I don't disagree but the reality however is that all UEFI machines ship with this cert in db.  But as you point out for completeness these mitigations are ONLY needed if that cert was found in db, so this should probably be checked as part of deciding whether to apply the mitigations.

hughsie, 2024-11-11:
I think this is solved with the hughsie/SkuSiPolicy -- please reopen if this is not the case.

closed 2025-05-04 by hughsie