fwupd/fwupd #5829: vid/pid for Belkin Components Thunderbolt 4 5-in-1 Core Hub dock

alebrati, 2023-05-17:
(see attached file)
Dock: Belkin Components Thunderbolt 4 5-in-1 Core Hub dock

issue:
The PD (cypress) is not detected over fwupd. as per my record, the vid/pid are :
"vid=050D"
"pid=006E"

fwupd version used : 1.8.12
OS: R113-15393.46.0
platform: Redrix

fwupd version used : 1.8.12
OS: R113-15393.46.0
platform: Redrix

fwupd output :

localhost /home # fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.11
compile   org.freedesktop.gusb          0.4.5
compile   com.hughsie.libjcat           0.1.8
runtime   org.kernel                    5.10.175-22068-g4b67de084816
runtime   org.freedesktop.gusb          0.4.5
compile   org.freedesktop.fwupd         1.8.12
[get_devices.log](https://github.com/fwupd/fwupd/files/11500731/get_devices.log)
[belkin_lsusb_verbos.log](https://github.com/fwupd/fwupd/files/11500732/belkin_lsusb_verbos.log)
[dmesg.log](https://github.com/fwupd/fwupd/files/11500733/dmesg.log)

IfxBrent, 2023-05-24:
I don't have any info about this model. 
But, as I know, TBT4 reference has 2 kinds of DMC design, CY7C65219(Regular DMC) and CYUSB4357(HX3PD).

hughsie, 2023-05-26:
> TBT4 reference has 2 kinds of DMC design, CY7C65219(Regular DMC) and CYUSB4357(HX3PD).

Super useful, thanks. Both of those are covered in the ccgx-dmc plugin.

alebrati, 2023-06-12:
tested on Richard branch and dmc is detected now.
```
root@yair-935QDB:/home/yair# fwupdtool get-devices
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
SAMSUNG ELECTRONICS CO., LTD. 935QDB
│
├─11th Gen Intel Core™ i7-1160G7 @ 1.20GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000a6
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│                         1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
│     Device Flags:       • Internal device
│
├─720p HD Camera:
│     Device ID:          65a54fb6ce182f0e75edf0e43047d547a0d61f0e
│     Current version:    0.11
│     Vendor:             Generic (USB:0x2B7E)
│     Serial Number:      0001
│     GUIDs:              56938296-0b35-59e3-9b6a-a897952a27c9 ← USB\VID_2B7E&PID_0171
│                         6ca9b3e9-e758-5ba6-b8f1-28f810777394 ← USB\VID_2B7E&PID_0171&REV_0011
│     Device Flags:       • Updatable
│
├─Display controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    01
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              5e77f508-559e-5926-b30c-dc90a8e539f8 ← PCI\VEN_8086&DEV_9A40
│                         6800005f-7666-575c-90f6-b03cd89c62d4 ← PCI\VEN_8086&DEV_9A40&REV_01
│                         dcb903f9-612a-5bbe-a56f-07abd9222112 ← PCI\VEN_8086&DEV_9A40&SUBSYS_144DC1A4
│                         a32a0d68-581a-5300-a24a-c548a3d746fc ← PCI\VEN_8086&DEV_9A40&SUBSYS_144DC1A4&REV_01
│                         7a23a6a7-e4b3-52b4-86a8-83d9ae0cecd5 ← PCI\VEN_8086&DEV_9A40&REV_00
│                         f294f923-7cba-5223-a39f-4f9253750bfc ← PCI\VEN_8086&DEV_9A40&SUBSYS_144DC1A4&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─MZ9LQ512HBLU-00B:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    FXM7AK1Q
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      S6HVNE0R101993
│     GUIDs:              8b86707a-1412-5389-8f66-725224d28879 ← NVME\VEN_144D&DEV_A809
│                         e8ae98bc-bc37-56b1-9680-8d64884a2792 ← NVME\VEN_144D&DEV_A809&REV_00
│                         71c19ff0-0bc0-5c85-aa6f-ba47fcc33eee ← NVME\VEN_144D&DEV_A809&SUBSYS_144DA801
│                         ff40c884-8b84-53e0-9ad0-82c87436a89a ← NVME\VEN_144D&DEV_A809&SUBSYS_144DA801&REV_00
│                         ae9684af-902e-5f15-9baf-7995570a1307 ← SAMSUNG MZ9LQ512HBLU-00B
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    8
│ │   Minimum Version:    8
│ │   Vendor:             SAMSUNG ELECTRONICS CO., LTD. (DMI:American Megatrends International, LLC.)
│ │   Update State:       Success
│ │   GUIDs:              922dbe27-1bf4-41c4-b111-3cf1e4005552
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
[... 67 lines cut ...]
│
├─UEFI Device Firmware:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    35193089
│     Vendor:             DMI:American Megatrends International, LLC.
│     Update State:       Success
│     GUID:               3b87537f-91b2-4fbb-95e3-fe8e41148b6e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
└─USB3.0 Hub:
      Device ID:          cf294bf55b333004beb7c41f952c1838c23e1f4a
      Current version:    41.81
      Vendor:             Intel Corporation. (USB:0x8087)
      GUIDs:              b3f8b518-ebdd-5969-9b44-6b096eff5b9b ← USB\VID_8087&PID_0B40
                          27f4ed60-ed53-52a7-ae28-ea87d9a23302 ← USB\VID_8087&PID_0B40&REV_1234
                          c6e7a14f-f13c-55e5-99de-d59461db1d27 ← USB\VID_8087&PID_0B40&HUB_0000
                          7b0409e5-df2c-5e48-856a-be4dac3e901f ← TBT-0013006e
      Device Flags:       • Updatable
                          • Signed Payload
```

alebrati, 2023-06-12:
[
[log2.txt](https://github.com/fwupd/fwupd/files/11720530/log2.txt)
](url)

hughsie, 2023-06-12:
Does it work if you're using the locally-built daemon and fwupmgr out of interest? the fwupdtool debug program is in-process and the hotplug events can be handled differently.

alebrati, 2023-06-12:
I tried this now, pls let me know if thats what you meant:

```
root@yair-935QDB:/home/yair# fwupdmgr --allow-reinstall install pd_dmc_belkin.cab
WARNING: This package has not been validated, it may not work properly.
Writing…                 [                                       ]
writing firmware not supported by device
root@yair-935QDB:/home/yair# fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.2
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.9.2
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.19.0-43-generic
runtime   org.freedesktop.fwupd         1.9.2
```

alebrati, 2023-06-12:
[
[fpdlog.log](https://github.com/fwupd/fwupd/files/11721684/fpdlog.log)
](url)

hughsie, 2023-06-12:
Ahh, no -- you need to run the daemon with `-v` to see anything useful, and `-v -v` to see all the juicy debugging.

closed 2023-12-01 by fdanis-oss