fwupd/fwupd #5849: LVFS Repository Checksum Mismatch

adampankow, 2023-05-22:
**Describe the bug**
While attempting to sync the LVFS repo, to prevent rate-limiting issues, a checksum mismatch error is encountered on line 1003 of the `PULP_MANIFEST`. The error prevents successfully syncing the repo.

**Steps to Reproduce**
```bash
$ wget https://cdn.fwupd.org/downloads/PULP_MANIFEST
$ head -1003 PULP_MANIFEST | tail -1
1f60443595b349d2238289ac7729d93209a8d64a-O7050-1121_Pro-1.wu.cab,b0d28ef3b43eb45ecf931fb474610aa841f3bba0d8d014d7e95d58128474b0b3,9245670
```

**Expected behavior**
Checksum in filename matches checksum in LVFS `PULP_MANIFEST` for all line entries

hughsie, 2023-05-22:
I don't get the connection with rate limiting -- can you describe in more detail what you're doing please?

adampankow, 2023-05-22:
> I don't get the connection with rate limiting -- can you describe in more detail what you're doing please?

Sure, I'm roughly trying to follow this guide. Using Pulp to mirror the LVFS repo, as our enterprise ends up hitting the 250 downloads per day limit, resulting in all `fwupd` clients erroring out. I am using Foreman's Content Library functionality to read the `PULP_MANIFEST` file and sync any changed firmware from the LVFS repo nightly.

hughsie, 2023-05-22:
I can't reproduce:

	wget https://fwupd.org/downloads/1f60443595b349d2238289ac7729d93209a8d64a-O7050-1121_Pro-1.wu.cab
	--2023-05-22 22:44:36--  https://fwupd.org/downloads/1f60443595b349d2238289ac7729d93209a8d64a-O7050-1121_Pro-1.wu.cab
	Resolving fwupd.org (fwupd.org)... 35.82.243.163, 34.211.166.9, 2600:1f14:414:5602::35b6, ...
	Connecting to fwupd.org (fwupd.org)|35.82.243.163|:443... connected.
	HTTP request sent, awaiting response... 302 FOUND
	Location: https://cdn.fwupd.org/downloads/1f60443595b349d2238289ac7729d93209a8d64a-O7050-1121_Pro-1.wu.cab?ts=1561110757 [following]
	--2023-05-22 22:44:38--  https://cdn.fwupd.org/downloads/1f60443595b349d2238289ac7729d93209a8d64a-O7050-1121_Pro-1.wu.cab?ts=1561110757
	Resolving cdn.fwupd.org (cdn.fwupd.org)... 151.101.62.49, 2a04:4e42:f::561
	Connecting to cdn.fwupd.org (cdn.fwupd.org)|151.101.62.49|:443... connected.
	HTTP request sent, awaiting response... 200 OK
	Length: 9245670 (8.8M) [application/vnd.ms-cab-compressed]
	Saving to: ‘1f60443595b349d2238289ac7729d93209a8d64a-O7050-1121_Pro-1.wu.cab’
	1f60443595b349d2238289ac7729d93209a8d64 100%[==============================================================================>]   8.82M  3.01MB/s    in 2.9s    
	2023-05-22 22:44:43 (3.01 MB/s) - ‘1f60443595b349d2238289ac7729d93209a8d64a-O7050-1121_Pro-1.wu.cab’ saved [9245670/9245670]

	$ sha256sum 1f60443595b349d2238289ac7729d93209a8d64a-O7050-1121_Pro-1.wu.cab 
	b0d28ef3b43eb45ecf931fb474610aa841f3bba0d8d014d7e95d58128474b0b3  1f60443595b349d2238289ac7729d93209a8d64a-O7050-1121_Pro-1.wu.cab

	$ wc -c 1f60443595b349d2238289ac7729d93209a8d64a-O7050-1121_Pro-1.wu.cab 
	9245670 1f60443595b349d2238289ac7729d93209a8d64a-O7050-1121_Pro-1.wu.cab

Got any more details on how to reproduce using Foreman's Content Library?

hughsie, 2023-05-22:
> as our enterprise ends up hitting the 250 downloads per day limit

This was a bit unexpected too; you've got more than 250 machines requesting the same firmware file from one single IP? I can certainly add an allowlist entry for your IP as it's legitimate, but you might be more interested in either using satellite to mirror lvfs content or the "downstream lvfs" thing that we're working on for a couple of Red Hat customers.

hughsie, 2023-05-23:
> I do understand that there are egress costs 

It's what makes up most our AWS bill.

> and I'd like to help the Linux Foundation in keeping those costs more reasonable

Sounds like a plan. I'm just syncing my machine and trying to work out what the problem is.

adampankow, 2023-05-23:
I'm inclined to close the issue for now, then get our Foreman to latest, to see if perhaps it could be an issue with the version of Pulp bundled with our present version of Foreman. It seems like others had reported checksum issues with other repositories to the Foreman project and it was believed to be a Pulp issue in those instances. I wouldn't want to have you chasing down an issue that might prove to not actually be an LVFS repo issue.

hughsie, 2023-05-23:
I've just run https://github.com/pulp/pulp-manifest on the secondary mirror and compared the output with the upstream PULP_MANIFEST. No different checksums -- but some duplicate files which might be upsetting Foreman perhaps. I'll fix that now.

The difference in firmware-testing.xml.gz is probably because the CDN and server have a different opinion on the latest content -- which unfortunately is an artefact of using the CDN.

hughsie, 2023-05-23:
I think the kind of people running an internal mirror would also be fairly paranoid about what firmware stats are being shared.

superm1, 2023-05-23:
> I think the kind of people running an internal mirror would also be fairly paranoid about what firmware stats are being shared.

Yeah it would have to be opt in, but some people would be open to it, no?

hughsie, 2023-05-24:
> Thank you for doing that. I will see if the nightly sync succeeds

I've merged https://gitlab.com/fwupd/lvfs-website/-/commit/191d9d0f9fddea931b3d908a4faa379932a25712 -- deployed it and now regenerated the stable remote.

Note; if you're a Red Hat customer (and I suspect you might be!) feel free to drop stuff like this in Red Hat bugzilla too. My firmware work at Red Hat is little known and it would be great to increase visibility a bit if we have customers using this for large-deployments.

adampankow, 2023-05-24:
Unfortunately that didn't seem to resolve my issue, but I'm going to chalk it up to an issue with our Foreman Katello release's  bundled Pulp for now. My mirror using `pulp-sync.py` will be sufficient until we get our Foreman Katello instance updated to the current release and I can try mirroring again.

We unfortunately don't have an enterprise Red Hat subscription, but I do have a personal one, and will keep in mind reporting via Bugzilla in the future, to ensure you get credit where it's due.

Thank you again for your help troubleshooting!