fwupd/fwupd #5869: Building the best UEFI GUID database using LVFS

NikolajSchlej, 2023-06-01:
Right now there are several efforts tor build a database of GUIDs related to UEFI and its implementations:
- [UEFITool](https://github.com/LongSoft/UEFITool/blob/new_engine/common/guids.csv)
- [Binarly](https://github.com/binarly-io/guiddb)

Such database is useful for researching images that don't include UI sections, but based on the same sources than other images that do include such sections.

The most recent version of UEFIExtract ([A66](https://github.com/LongSoft/UEFITool/releases/tag/A66)) added automatic dumping of all detected GUID-name pairs into ${filename}.guids.csv when launched with default set of parameters. I know that UEFIExtract is already getting applied to LVFS uploads, so we can build a database of all GUID-name pairs known to LVFS once, and then add entries to it when a new file gets uploaded. The resulting database will likely be miles ahead of both DBs linked above.

NikolajSchlej, 2023-06-01:
CC @hughsie @matrosov @yeggor

NikolajSchlej, 2023-06-01:
More GUIDs: https://github.com/abazhaniuk/efi-whitelist

hughsie, 2023-06-04:
> More GUIDs: https://github.com/abazhaniuk/efi-whitelist

Ohh wow, not a lot of updates in the last 5 years there! Do you think they're still useful?

hughsie, 2023-06-04:
> Right now there are several efforts tor build a database of GUIDs related to UEFI and its implementations:
>    [UEFITool](https://github.com/LongSoft/UEFITool/blob/new_engine/common/guids.csv)

Would it be acceptable to use the guids.csv data under the CC0-1.0 licence?

NikolajSchlej, 2023-06-04:
> Ohh wow, not a lot of updates in the last 5 years there! Do you think they're still useful?
Might still be, especially if covering images not covered by LVFS yet.
> Would it be acceptable to use the guids.csv data under the CC0-1.0 licence?
Sure thing.

hughsie, 2023-06-05:
https://gitlab.com/fwupd/lvfs-website/-/commit/7757355abef97b420e6adb950423049386045c37 is a good start. I'll deploy that and then let the cron job chew it over during the night.

NikolajSchlej, 2023-06-06:
Thanks a ton, will update my guids.csv now to remove most of those differences (because `UsbRtSmm` is a better name than `USBRT` after all). The growth of the DB is expected and welcome.

yeggor, 2023-06-06:
> can I absorb the data from guiddb and make it available under a CC0-1.0 licence

@hughsie of course you can

hughsie, 2023-06-06:
guiddb also imported:

[image]

yeggor, 2023-06-06:
> @yeggor you should be able to edit/add entries with https://fwupd.org/lvfs/shards/ and there's a anonymous download available from https://fwupd.org/lvfs/shards/export/csv -- the latter will take an hour or so to update.

thank you!

hughsie, 2023-06-06:
@yeggor do you need me to provide a guiddb-compatible export endpoint or is the CSV good for you too?

yeggor, 2023-06-06:
CSV looks fine, thank you. I will write a script for`guids.json` updates

hughsie, 2023-06-08:
Okay; lets close for now and we can re-open if Alex gives us the nod for the older GUIDs.

closed 2023-06-08 by hughsie