fwupd/fwupd #5870: Consistenet verison format for Lenovo UEFI devices

joakim-tjernlund, 2023-06-02:
**Is your feature request related to a problem? Please describe.**
Unknown Lenovo UEFI ESRT devices has a flat version number while known ones has a x.y.z version number.
Installing non LVFS updates confuses fwupmgr w.r.t how to compare version numbers.

**Describe the solution you'd like**
Add a general rule:
If UEFI ESRT and DMI:LENOVO -> assume x.y.z version format.

superm1, 2023-06-02:
> **Is your feature request related to a problem? Please describe.**
> Unknown Lenovo UEFI ESRT devices has a flat version number while known ones has a x.y.z version number.
> Installing non LVFS updates confuses fwupmgr w.r.t how to compare version numbers.

Can you elaborate more on what is happening now?
Eventually when update is published there shouldn't things get sorted?

> 
> **Describe the solution you'd like**
> Add a general rule:
> If UEFI ESRT and DMI:LENOVO -> assume x.y.z version format.
> 
> 

The problem with this idea is that if lenovo ever wanted to change formats they wouldn't be able to.

joakim-tjernlund, 2023-06-02:
> Can you elaborate more on what is happening now? Eventually when update is published there shouldn't things get sorted?

This was a while ago so memory is not great but you can install CABs manually, see
 https://download.lenovo.com/pccbbs/mobiles/n3bul10w.txt
and that then later confuse fwupdmgr when it is comparing version numbers. I cannot
remember the details, sorry. 

Eventually can be a very long time too.
> 
> > **Describe the solution you'd like**
> > Add a general rule:
> > If UEFI ESRT and DMI:LENOVO -> assume x.y.z version format.
> 
> The problem with this idea is that if lenovo ever wanted to change formats they wouldn't be able to.

Why not? Now there is a default of flat numbers and that works to change into x.y.z
The rule would act like a default route, last resort when no specific entry exists.

hughsie, 2023-06-02:
Right, but even "Lenovo == triplet" isn't accurate in all cases. If we choose the wrong default the vercmp can fail in all sorts of terrible (and security relevant) ways.

joakim-tjernlund, 2023-06-02:
No worse than now as the flat numbers as always wrong when comparing, right?
Once a device is known you change this to the format specified for the device.

Also, would be nice if unknown devices are hidden in get-devices, they are of no use and only use screen space for no good reason. Many a verbose flag to show everything.

hughsie, 2023-06-02:
> would be nice if unknown devices are hidden in get-devices

You mean hide the uefi devices with no version format unless specifying --show-all?

joakim-tjernlund, 2023-06-02:
> > would be nice if unknown devices are hidden in get-devices
> 
> You mean hide the uefi devices with no version format unless specifying --show-all?

Yes, something like that. ATM there is no information in these, no name and no version that make sense.
Only useful to devs I  think.

superm1, 2023-06-02:
> > > would be nice if unknown devices are hidden in get-devices
> > 
> > You mean hide the uefi devices with no version format unless specifying --show-all?
> 
> Yes, something like that. ATM there is no information in these, no name and no version that make sense.
> Only useful to devs I  think.

I disagree. If you cross reference windows in device manager the exact same esrt devices show up with same information.

If one of these versions changed some day when not flashed with fwupd (malware?) wouldn't you want to be able to know?

joakim-tjernlund, 2023-06-02:
> I disagree. If you cross reference windows in device manager the exact same esrt devices show up with same information.

And no useful version information too?

> 
> If one of these versions changed some day when not flashed with fwupd (malware?) wouldn't you want to be able to know?

How would I know? It is just anonymous data ATM. One can you --show-all too
Add in version that can be matched to Lenovo versions on their www site and there is some use.

hughsie, 2023-06-04:
Can you try with 1.9.x please? Installing a trusted local .cab file it should "absorb" the new verfmt prior to showing the details.

joakim-tjernlund, 2023-06-07:
> Can you try with 1.9.x please? Installing a trusted local .cab file it should "absorb" the new verfmt prior to showing the details.

I will once Gentoo gets 1.9.x. However I don't have any local .cab files ATM that we need so it may take some time.

It does not solve the general problems with global default flat version numbers though.

hughsie, 2023-06-08:
> It does not solve the general problems with global default flat version numbers though.

We used to do what you suggested, but a few years ago we went back to "flat" version numbers due to complexity and real-world bugs. I don't think this is something we should do, sorry.

closed 2023-06-08 by hughsie

joakim-tjernlund, 2023-06-09:
Would you consider hiding these from get-devices?

superm1, 2023-06-09:
> Would you consider hiding these from get-devices?

I still don't see any argument for why.

Hiding the devices without a version format would mean if you tried to check devices they're missing until they're supported by LVFS or you run a refresh cycle.

That means all UEFI devices don't show up on live media either. They're harmless to keep in.